openwrt/packages #26026: frr: fail to build AM_COND_IF: no such condition "LUAJIT"

THLIVSQAZ, 2025-02-21:
build errors:
```
make[2]: Entering directory '/mnt/openwrtZone/opmaster/feeds/packages/net/frr'
. /mnt/openwrtZone/opmaster/include/shell.sh; /mnt/openwrtZone/opmaster/staging_dir/host/bin/libdeflate-gzip -dc /mnt/openwrtZone/opmaster/dl/frr-2025-01-29.tar.gz | tar -C /mnt/openwrtZone/opmaster/build_dir/hostpkg/frr-05f9d44340755e5ca612e4852b710f4037254e7b/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /mnt/openwrtZone/opmaster/build_dir/hostpkg/frr-05f9d44340755e5ca612e4852b710f4037254e7b

Applying /mnt/openwrtZone/opmaster/feeds/packages/net/frr/patches/098-fix_mips_libyang.patch using plaintext: 
patching file lib/northbound.h

Applying /mnt/openwrtZone/opmaster/feeds/packages/net/frr/patches/996-qpb_header_fix.patch using plaintext: 
patching file qpb/qpb_allocator.h

Applying /mnt/openwrtZone/opmaster/feeds/packages/net/frr/patches/997-reverse_python_test.patch using plaintext: 
patching file configure.ac
touch /mnt/openwrtZone/opmaster/build_dir/hostpkg/frr-05f9d44340755e5ca612e4852b710f4037254e7b/.preparedd643c11fcf937a0617438ef181cba969_6664517399ebbbc92a37c5bb081b5c53
(cd /mnt/openwrtZone/opmaster/build_dir/hostpkg/frr-05f9d44340755e5ca612e4852b710f4037254e7b; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf ./autom4te.cache; [ -e ./config.rpath ] || ln -s /mnt/openwrtZone/opmaster/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/mnt/openwrtZone/opmaster/staging_dir/host/bin/autom4te AUTOCONF=/mnt/openwrtZone/opmaster/staging_dir/host/bin/autoconf AUTOMAKE=/mnt/openwrtZone/opmaster/staging_dir/host/bin/automake ACLOCAL=/mnt/openwrtZone/opmaster/staging_dir/host/bin/aclocal AUTOHEADER=/mnt/openwrtZone/opmaster/staging_dir/host/bin/autoheader LIBTOOLIZE=/mnt/openwrtZone/opmaster/staging_dir/host/bin/libtoolize LIBTOOL=/mnt/openwrtZone/opmaster/staging_dir/host/bin/libtool M4=/mnt/openwrtZone/opmaster/staging_dir/host/bin/m4 AUTOPOINT=true GTKDOCIZE=true LIBTOOLIZE='/mnt/openwrtZone/opmaster/staging_dir/host/bin/libtoolize --install' /mnt/openwrtZone/opmaster/staging_dir/host/bin/autoreconf -v -f -i -B /mnt/openwrtZone/opmaster/staging_dir/host/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /mnt/openwrtZone/opmaster/staging_dir/host/bin/aclocal -I m4 -I . --force -I m4 -Wall,no-override
configure.ac:378: error: AM_COND_IF: no such condition "LUAJIT"
/mnt/openwrtZone/opmaster/staging_dir/host/share/aclocal-1.17/cond-if.m4:23: AM_COND_IF is expanded from...
/mnt/openwrtZone/opmaster/staging_dir/host/share/aclocal/ax_lua.m4:199: AX_PROG_LUA is expanded from...
configure.ac:378: the top level
autom4te: error: /mnt/openwrtZone/opmaster/staging_dir/host/bin/m4 failed with exit status: 1
aclocal.real: error: /mnt/openwrtZone/opmaster/staging_dir/host/bin/autom4te failed with exit status: 1
autoreconf: error: /mnt/openwrtZone/opmaster/staging_dir/host/bin/aclocal failed with exit status: 1
(cd /mnt/openwrtZone/opmaster/build_dir/hostpkg/frr-05f9d44340755e5ca612e4852b710f4037254e7b/; if [ -x configure ]; then cp -fpR /mnt/openwrtZone/opmaster/scripts/config.{guess,sub} /mnt/openwrtZone/opmaster/build_dir/hostpkg/frr-05f9d44340755e5ca612e4852b710f4037254e7b// && CC="/mnt/openwrtZone/opmaster/staging_dir/host/bin/gcc" CFLAGS="-O2 -I/mnt/openwrtZone/opmaster/staging_dir/host/include -I/mnt/openwrtZone/opmaster/staging_dir/hostpkg/include -I/mnt/openwrtZone/opmaster/staging_dir/target-i386_pentium4_musl/host/include" CXX="/mnt/openwrtZone/opmaster/staging_dir/host/bin/g++" CPPFLAGS="-I/mnt/openwrtZone/opmaster/staging_dir/host/include -I/mnt/openwrtZone/opmaster/staging_dir/hostpkg/include -I/mnt/openwrtZone/opmaster/staging_dir/target-i386_pentium4_musl/host/include -I/mnt/openwrtZone/opmaster/staging_dir/host/include/libelf" CXXFLAGS="-O2 -I/mnt/openwrtZone/opmaster/staging_dir/host/include -I/mnt/openwrtZone/opmaster/staging_dir/hostpkg/include -I/mnt/openwrtZone/opmaster/staging_dir/target-i386_pentium4_musl/host/include" LDFLAGS="-L/mnt/openwrtZone/opmaster/staging_dir/host/lib -L/mnt/openwrtZone/opmaster/staging_dir/hostpkg/lib -L/mnt/openwrtZone/opmaster/staging_dir/target-i386_pentium4_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/mnt/openwrtZone/opmaster/staging_dir/hostpkg --exec-prefix=/mnt/openwrtZone/opmaster/staging_dir/hostpkg --sysconfdir=/mnt/openwrtZone/opmaster/staging_dir/hostpkg/etc --localstatedir=/mnt/openwrtZone/opmaster/staging_dir/hostpkg/var --sbindir=/mnt/openwrtZone/opmaster/staging_dir/hostpkg/bin --enable-clippy-only ; fi )
/mnt/openwrtZone/opmaster/staging_dir/host/bin/sed -i -e 's/$(MAKE) $(AM_MAKEFLAGS) install-am/# $(MAKE) $(AM_MAKEFLAGS) install-am/' /mnt/openwrtZone/opmaster/build_dir/hostpkg/frr-05f9d44340755e5ca612e4852b710f4037254e7b/Makefile.in
/mnt/openwrtZone/opmaster/staging_dir/host/bin/sed: can't read /mnt/openwrtZone/opmaster/build_dir/hostpkg/frr-05f9d44340755e5ca612e4852b710f4037254e7b/Makefile.in: No such file or directory
make[2]: *** [Makefile:231: /mnt/openwrtZone/opmaster/build_dir/hostpkg/frr-05f9d44340755e5ca612e4852b710f4037254e7b/.configured] Error 2
make[2]: Leaving directory '/mnt/openwrtZone/opmaster/feeds/packages/net/frr'
time: package/feeds/packages/frr/host-compile#2.91#0.38#4.25
    ERROR: package/feeds/packages/frr [host] failed to build.
make[1]: *** [package/Makefile:187: package/feeds/packages/frr/host/compile] Error 1
make[1]: Leaving directory '/mnt/openwrtZone/opmaster'
make: *** [/mnt/openwrtZone/opmaster/include/toplevel.mk:233: package/frr/compile] Error 2
```

OpenWrt version:
master 70586f574d93bf5b48f878a3c2a264b0249830c1

Config:
```
./scripts/diffconfig.sh 
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_generic=y
CONFIG_TARGET_x86_generic_DEVICE_generic=y
CONFIG_FRR_INTERNAL=y
CONFIG_FRR_NO_SNMP=y
CONFIG_PACKAGE_frr=y
CONFIG_PACKAGE_frr-babeld=y
CONFIG_PACKAGE_frr-bfdd=y
CONFIG_PACKAGE_frr-bgpd=y
CONFIG_PACKAGE_frr-eigrpd=y
CONFIG_PACKAGE_frr-fabricd=y
CONFIG_PACKAGE_frr-isisd=y
CONFIG_PACKAGE_frr-ldpd=y
CONFIG_PACKAGE_frr-nhrpd=y
CONFIG_PACKAGE_frr-ospf6d=y
CONFIG_PACKAGE_frr-ospfd=y
CONFIG_PACKAGE_frr-pathd=y
CONFIG_PACKAGE_frr-pbrd=y
CONFIG_PACKAGE_frr-pim6d=y
CONFIG_PACKAGE_frr-pimd=y
CONFIG_PACKAGE_frr-ripd=y
CONFIG_PACKAGE_frr-ripngd=y
CONFIG_PACKAGE_frr-staticd=y
CONFIG_PACKAGE_frr-vrrpd=y
CONFIG_PACKAGE_frr-watchfrr=y
CONFIG_PACKAGE_frr-zebra=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcares=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libprotobuf-c=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libyang=y
CONFIG_PACKAGE_more=y
CONFIG_PACKAGE_terminfo=y
```

cpswan, 2025-02-21:
@THLIVSQAZ seems this is another example of builds failing due to an issue with autoconf that's being tracked at https://github.com/openwrt/openwrt/issues/18059

dmascord, 2025-02-24:
This doesn't seem to be resolved by the revert of autoconf and automake, it is still failing to build on my local system.  I have tried a distclean and rebuild, but still no joy

dmascord, 2025-02-24:
Looks like reverting to 2023.02.20 of autoconf-archive resolves this issue - https://forums.gentoo.org/viewtopic-t-1172092.html?sid=72c50dc990f183b2bf17121ad1481a52
```
diff --git a/tools/autoconf-archive/Makefile b/tools/autoconf-archive/Makefile
index a0c75c1dca..a4ab423779 100644
--- a/tools/autoconf-archive/Makefile
+++ b/tools/autoconf-archive/Makefile
@@ -1,11 +1,11 @@
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=autoconf-archive
-PKG_VERSION:=2024.10.16
+PKG_VERSION:=2023.02.20
 
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
 PKG_SOURCE_URL:=@GNU/autoconf-archive
-PKG_HASH:=7bcd5d001916f3a50ed7436f4f700e3d2b1bade3ed803219c592d62502a57363
+PKG_HASH:=71d4048479ae28f1f5794619c3d72df9c01df49b1c628ef85fde37596dc31a33
 
 include $(INCLUDE_DIR)/host-build.mk
```

lucize, 2025-02-25:
https://github.com/openwrt/openwrt/commit/e1acc7a67b92fa964710afe0a3ae2dad5c1f00c7

THLIVSQAZ, 2025-02-27:
Now can be compiled successfully, Thanks

closed 2025-02-27 by THLIVSQAZ