openwrt/packages #26027: ERROR: package/feeds/packages/rust [host] failed to build

woni928, 2025-02-21:
openwrt 23.05
help

[image]

brada4, 2025-02-21:
Any chance you link your CI job as particular screenshot does not contain ultimate error message.

woni928, 2025-02-22:
> 您是否有可能链接您的 CI 作业，因为特定的屏幕截图不包含最终的错误消息。

thanks

[10_Compile the firmware.txt](https://github.com/user-attachments/files/18920440/10_Compile.the.firmware.txt)

redfrog999, 2025-02-23:
> > 您是否有可能链接您的 CI 作业，因为特定的屏幕截图不包含最终的错误消息。
> 
> thanks 同样的代码切换为24.10编译就正常。
> 
> [10_Compile the firmware.txt](https://github.com/user-attachments/files/18920454/10_Compile.the.firmware.txt)

+1 这几天23.05就是跑不过，一直卡rust，不知道哪个大佬修复下。

Jejz168, 2025-02-23:
> > 您是否有可能链接您的 CI 作业，因为特定的屏幕截图不包含最终的错误消息。
> 
> thanks 同样的代码切换为24.10编译就正常。
> 
> [10_Compile the firmware.txt](https://github.com/user-attachments/files/18920454/10_Compile.the.firmware.txt)

CONFIG_PACKAGE_shadowsocks-rust-sslocal=y
改为 is not set

redfrog999, 2025-02-23:
> > > 您是否有可能链接您的 CI 作业，因为特定的屏幕截图不包含最终的错误消息。
> > 
> > 
> > thanks 同样的代码切换为24.10编译就正常。
> > [10_Compile the firmware.txt](https://github.com/user-attachments/files/18920454/10_Compile.the.firmware.txt)
> 
> CONFIG_PACKAGE_shadowsocks-rust-sslocal=y 改为 is not set

没用还是同样的报错。

redfrog999, 2025-02-23:
> > > 您是否有可能链接您的 CI 作业，因为特定的屏幕截图不包含最终的错误消息。
> > 
> > 
> > thanks 同样的代码切换为24.10编译就正常。
> > [10_Compile the firmware.txt](https://github.com/user-attachments/files/18920454/10_Compile.the.firmware.txt)
> 
> CONFIG_PACKAGE_shadowsocks-rust-sslocal=y 改为 is not set

倒是没报错了，但是passwall需要这个组件，硬路由的小空间后来补装还有点麻烦。

Jejz168, 2025-02-23:
> > > > 您是否有可能链接您的 CI 作业，因为特定的屏幕截图不包含最终的错误消息。
> > > 
> > > 
> > > thanks 同样的代码切换为24.10编译就正常。
> > > [10_Compile the firmware.txt](https://github.com/user-attachments/files/18920454/10_Compile.the.firmware.txt)
> > 
> > CONFIG_PACKAGE_shadowsocks-rust-sslocal=y 改为 is not set
> 
> 倒是没报错了，但是passwall需要这个组件，硬路由的小空间后来补装还有点麻烦。

你还是多看看别人编译的再说吧

redfrog999, 2025-02-23:
> > > > > 您是否有可能链接您的 CI 作业，因为特定的屏幕截图不包含最终的错误消息。
> > > > 
> > > > 
> > > > thanks 同样的代码切换为24.10编译就正常。
> > > > [10_Compile the firmware.txt](https://github.com/user-attachments/files/18920454/10_Compile.the.firmware.txt)
> > > 
> > > 
> > > CONFIG_PACKAGE_shadowsocks-rust-sslocal=y 改为 is not set
> > 
> > 
> > 倒是没报错了，但是passwall需要这个组件，硬路由的小空间后来补装还有点麻烦。
> 
> 你还是多看看别人编译的再说吧

你这个根本没卵用，最后依然报同样的错。现在暂时切换24.10，等待上游修复。

woni928, 2025-02-23:
> 你这个根本没卵用，最后依然报同样的错。现在暂时切换24.10，等待上游修复。

兄弟，有解决办法吗？

BKPepe, 2025-05-20:
I need to close this issue as it is not clear for me:
- which OpenWrt version are you using (if you are official sources [git sources, downloaded tagged releases, ...] or not)
- you need to post log not screenshot
- you dont specify how I can reproduce it

Most likely temporary hiccup of rust server, thus closing this as we can not fix it on our end.

closed 2025-05-20 by BKPepe