openwrt/packages #26076: [question] tangd not starting

eskultety, 2025-03-01:
Hi, I couldn't find any OpenWrt docs on how to configure tang. The default service installation doesn't seem to start the process.
```
# /etc/init.d/tang enable
# echo $?
0
# /etc/init.d/tang enabled
# echo $?
0
# /etc/init.d/tang start
# echo $?
0
# /etc/init.d/tang status
active with no instances
# ubus call service list | grep -A10 tang
        "tang": {

        },
        "ubus": {
                "instances": {
                        "instance1": {
                                "running": true,
                                "pid": 1081,
                                "command": [
                                        "/sbin/ubusd"
                                ],
```
To rule out permissions, I set `/usr/share/tang/db` to `777`, didn't help. `logread` didn't provide me with anything and I don't know how to debug `procd`. Can you please help me with what I might be doing wrong? I can reproduce this both on an ath79 bare metal device as well as in an x86_64 test VM.

q-b, 2025-03-24:
Same problem here.

q-b, 2025-04-04:
Ok dumb testing:
```
# cat /etc/init.d/tang 
```
```
#!/bin/sh /etc/rc.common

START=50
USE_PROCD=1

start_service() {
	KEYS=$(find /usr/share/tang/db -name "*.jw*" -maxdepth 1 2>/dev/null | wc -l)
	if [ -z "${KEYS}" ] || [ "${KEYS}" = "0" ]; then # if db is empty generate new key pair
		mkdir -p /usr/share/tang/db
		/usr/sbin/tangd-keygen /usr/share/tang/db
		chown -R tang /usr/share/tang/db
	fi

	config_load "tang"
	config_get_bool enabled "config" "enabled" 0
	config_get port "config" "port" "9090"

echo "step 1 ${enabled}" > /tmp/debug_tang

	[ "${enabled}" = "1" ] || return

echo "step 2" >> /tmp/debug_tang

	procd_open_instance
	procd_set_param command /usr/sbin/tangd -p "${port}" -l /usr/share/tang/db
	procd_set_param respawn
	procd_set_param user tang
	procd_close_instance
}
```
```
# service tang enable
# service tang start
# cat /tmp/debug_tang 
step 1 0
# service tang enabled
# service tang start
# cat /tmp/debug_tang 
step 1 0
```
Something is off with `enable/enabled`, I checked with `/etc/init.d/tang xxx` as well, same result

q-b, 2025-04-08:
Got it working, need to find time to submit a PR.
Please note that the key creation part has not been tested.

_/etc/config/tang_
```
config tang 'config'
	option port '9090'
```

_/etc/init.d/tang_
```
#!/bin/sh /etc/rc.common

START=50
USE_PROCD=1

start_service() {
	KEYS=$(find /usr/share/tang/db -name "*.jw*" -maxdepth 1 2>/dev/null | wc -l)
	if [ -z "${KEYS}" ] || [ "${KEYS}" = "0" ]; then # if db is empty generate new key pair
		mkdir -p /usr/share/tang/db
		/usr/sbin/tangd-keygen /usr/share/tang/db
		chown -R tang /usr/share/tang/db
	fi

	config_load "tang"
	config_get port "config" "port" "9090"

	procd_open_instance
	procd_set_param command /usr/sbin/tangd -p "${port}" -l /usr/share/tang/db
	procd_set_param respawn
	procd_set_param user tang
	procd_set_param stderr 1
	procd_set_param stdout 1
	procd_close_instance
}
```

_permissions_
```
chown root:root /usr/share/tang
chmod 755 /usr/share/tang

chown -R tang:tang /usr/share/tang/db
chmod 700 /usr/share/tang/db
chmod 600 /usr/share/tang/db/*
```

rugk, 2025-09-16:
@q-b  Do you maybe want to do the hurdle of signing of some PR with "your real name"? Then you can use my code in https://github.com/openwrt/packages/pull/27401

Alternatively only use the one line to only log STDERR entries.

I guess you wanted to contribute/do a PR before.