openwrt/packages #26086: num-threads max 2

devianceluka, 2025-03-03:
https://github.com/openwrt/packages/blob/7ce67fb7dc08ebd541dc36eccb0f68c0f8c7f48b/net/unbound/files/unbound.sh#L712

Why does a system specify what would be the _sane_ limit _for_ the user?

`option num-threads` is where we specify how much threads  _we_ as users want the system to use. As it is, **the only** possible scenarios (over-ruled by the system/**UCI init.d script**) are 1 (single core) and 2.

2 was _multi-core_ maybe 6 years ago, but not in 2025. In my opinion, even 6 years ago _any_ over-ruled by the system limit was dumb.

brada4, 2025-03-04:
Try the shell maths yourself:
https://unbound.docs.nlnetlabs.nl/en/latest/topics/core/performance.html
It has to fit in RAM too.

devianceluka, 2025-03-04:
The problem is not doing the math. The problem is that UCI is accepting (_translating_) only:
1) `option num-threads 1`
2) `option num-threads 2`

If I want anything over `2` there is no other choice but to use `option manual_conf 1` and translate _everything else also_ myself.

brada4, 2025-03-04:
Some of memory is allocated per-thread, some per-pool.

EricLuehrsen, 2025-03-09:
The other options need to match up with number of threads. No more than two threads should share a "slab" of DNS cache. Adding slabs makes Unbound grow quickly. Some other tuning options also get entangled. UCI is about simplicity. If you  want to use Unbound at this level, then it might be time to switch to manual mode and edit the Unbound configuration directly.

devianceluka, 2025-03-09:
Understood. The only option was to manually edit.

But still. I understand its hard to calculate and waste time to make everything simple when simple isnt anywhere to start with... This limit(ation) is maybe **too** simple (my personal opinion). Its like either red pill or blue pill and **nothing else** for something that **can** get quite complex.

Should I close this?

brada4, 2025-03-10:
I kind of dropped ref to official documentation mentioning options vs memories if anyone is goid at shell maths...

EricLuehrsen, 2025-03-15:
Its not about "hard to calculate" but rather how much complexity and size is appropriate for a typically small OpenWrt device. At what point is the configuration more like a DHCP/DNS enterprise appliance? Unbound cache management leaves bloat. It does not prune on TTL. It requeries to TTL. It keeps everything in cache until its full. The it makes space based on record age in cache. Its the consequences from similar quirks that is challenging.

devianceluka, 2025-03-15:
Since when is unbound considered a better choice for a grandma with an old and small OpenWrt device?

devianceluka, 2025-03-15:
For anyone higher up in natural selection hierarchy reading this, if you want to close this, feel free to close it. I solved it manually and moved on. My opinion still stands that this should be changed in UCI.

EricLuehrsen, 2025-05-02:
Again its not like that. If you use Unbound through UCI, then you expect it to just work. Unbound and its sibling NSD are less bloated than Bind, but they still have bloat. When Unbound uses more threads and slabs, you need to babysit it. You need to adjust settings for you machine. If we add UCI options, then we also have responsibility to maybe have some chron job that reports unbound status. It should also make recommendations. I could strawman some of that, but I wouldn't have time to test it. I have had Unbound on TPLink 4300 and Archer (mips, 1 core) and WRT3200ACM (arm, 2 core). It really does well in these devices.

closed 2025-11-15 by dibdot