openwrt/packages #26096: x86: imagebuilder fails to build an image when package luci-ssl-nginx is included.

davidandreoletti, 2025-03-05:
**Problem:**

With imagebuilder `snapshots` [0] and the command line below:
- building an image **excluding package "luci-ssl-nginx"** succeeds. 
- building an image **including package "luci-ssl-nginx"** fails. 

```
 make V=s -j 6 image PROFILE=generic 'PACKAGES=luci-base luci-ssl-nginx luci-mod-admin-full luci-theme-bootstrap kmod-usb-storage kmod-usb-ohci kmod-usb-uhci debootstrap debian-archive-keyring gpgv2 ip-full rsync iperf3 htop vim-full vim-runtime ethtool pciutils lsblk blkid diffutils openssh-server openssh-sftp-server wget curl ca-certificates ddns-scripts luci-app-ddns etherwake luci-app-wol luci-proto-wireguard wireguard-tools tmux collectd-mod-thermal collectd-mod-cpu collectd-mod-disk collectd-mod-memory collectd collectd-mod-rrdtool luci-app-statistics jq yq adguardhome luci-app-sqm openssh-client autossh nut-web-cgi nut-upslog nut-upsc nut-upsmon nut-server nut-common nut block-mount e2fsprogs kmod-fs-ext4 kmod-usb-storage kmod-usb-storage-uas kmod-usb2 kmod-usb3 usbutils coreutils-dd coreutils-sha256sum findutils-find eject gzip nginx mosquitto shairport-sync-openssl mosquitto owntone shadow-useradd mpd-full ffmpeg flock mpc mosquitto-client procps-ng-pgrep procps-ng-kill wget wget-ssl qemu-x86_64-softmmu qemu-aarch64-softmmu qemu-ga qemu-bridge-helper qemu-img kmod-tun kmod-vhost-net telnet-bsd xz xz-utils curl coreutils-truncate cloudflared luci-app-cloudflared tailscale tailscaled iptables-nft kmod-ipt-conntrack kmod-ipt-conntrack-extra kmod-ipt-conntrack-label kmod-nft-nat kmod-ipt-nat nut-driver-usbhid-ups kmod-bluetooth bluez-daemon bluez-libs bluez-utils bluez-utils-extra bluez-tools bluez-utils-btmon python3 kmod-input-uinput sbc gst1-mod-bluez pulseaudio-daemon-avahi pulseaudio-profiles pulseaudio-tools alsa-utils dbus dbus-utils kmod-sound-dummy sudo sed patch shadow-groupadd shadow-usermod madplay libblkid' FILES=/var/tmp/openwrt/build/image/davida-router-hw1-dev/generic/files EXTRA_IMAGE_NAME=davida-router-hw1-dev BIN_DIR=/mnt/share_openwrt/image/releases/image/davida-router-hw1-dev/snapshots

Generate local signing keys...
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
Checking 'true'... ok.
Checking 'false'... ok.
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'perl-ipc-cmd'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... updated.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... updated.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... updated.
Checking 'egrep'... updated.
Checking 'getopt'... ok.
Checking 'realpath'... ok.
Checking 'stat'... ok.
Checking 'gzip'... updated.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'install'... ok.
Checking 'perl'... ok.
Checking 'python'... updated.
Checking 'python3'... updated.
Checking 'python3-distutils'... ok.
Checking 'python3-stdlib'... ok.
Checking 'file'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.
Building images for x86 - Generic x86/64
Packages: luci-base luci-ssl-nginx luci-mod-admin-full luci-theme-bootstrap kmod-usb-storage kmod-usb-ohci kmod-usb-uhci debootstrap debian-archive-keyring gpgv2 ip-full rsync iperf3 htop vim-full vim-runtime ethtool pciutils lsblk blkid diffutils openssh-server openssh-sftp-server wget curl ca-certificates ddns-scripts luci-app-ddns etherwake luci-app-wol luci-proto-wireguard wireguard-tools tmux collectd-mod-thermal collectd-mod-cpu collectd-mod-disk collectd-mod-memory collectd collectd-mod-rrdtool luci-app-statistics jq yq adguardhome luci-app-sqm openssh-client autossh nut-web-cgi nut-upslog nut-upsc nut-upsmon nut-server nut-common nut block-mount e2fsprogs kmod-fs-ext4 kmod-usb-storage kmod-usb-storage-uas kmod-usb2 kmod-usb3 usbutils coreutils-dd coreutils-sha256sum findutils-find eject gzip nginx mosquitto shairport-sync-openssl mosquitto owntone shadow-useradd mpd-full ffmpeg flock mpc mosquitto-client procps-ng-pgrep procps-ng-kill wget wget-ssl qemu-x86_64-softmmu qemu-aarch64-softmmu qemu-ga qemu-bridge-helper qemu-img kmod-tun kmod-vhost-net telnet-bsd xz xz-utils curl coreutils-truncate cloudflared luci-app-cloudflared tailscale tailscaled iptables-nft kmod-ipt-conntrack kmod-ipt-conntrack-extra kmod-ipt-conntrack-label kmod-nft-nat kmod-ipt-nat nut-driver-usbhid-ups kmod-bluetooth bluez-daemon bluez-libs bluez-utils bluez-utils-extra bluez-tools bluez-utils-btmon python3 kmod-input-uinput sbc gst1-mod-bluez pulseaudio-daemon-avahi pulseaudio-profiles pulseaudio-tools alsa-utils dbus dbus-utils kmod-sound-dummy sudo sed patch shadow-groupadd shadow-usermod madplay libblkid apk-mbedtls base-files ca-bundle dnsmasq dropbear e2fsprogs firewall4 fstools grub2-bios-setup kernel kmod-amazon-ena kmod-amd-xgbe kmod-bnx2 kmod-button-hotplug kmod-drm-i915 kmod-dwmac-intel kmod-e1000 kmod-e1000e kmod-forcedeth kmod-fs-vfat kmod-igb kmod-igc kmod-ixgbe kmod-nft-offload kmod-r8169 kmod-tg3 libc libgcc libustream-mbedtls logd mkf2fs mtd netifd nftables odhcp6c odhcpd-ipv6only partx-utils ppp ppp-mod-pppoe procd-ujail uci uclient-fetch urandom-seed urngd

WARNING: opening /var/tmp/openwrt/build/image/davida-router-hw1-dev/generic/image/openwrt-imagebuilder/packages/packages.adb: No such file or directory
OK: 0 MiB in 0 packages
Package list missing or not up-to-date, generating it.

Building package index...

Installing packages...
ERROR: unable to select packages:
  nginx (no such package):
    required by: world[nginx] luci-nginx-25.063.09777~4c692b8[nginx]
  nginx-mod-luci (no such package):
    required by: luci-nginx-25.063.09777~4c692b8[nginx-mod-luci]
make[2]: *** [Makefile:226: package_install] Error 3
make[1]: *** [Makefile:157: _call_image] Error 2
make: *** [Makefile:332: image] Error 2
```

**Environment**

OpenWRT Target: x86/64
[image]

robimarko, 2025-03-05:
nginx building is currently broken, I just pushed the fix so it should get rebuilt in the next round:
https://github.com/openwrt/packages/pull/26099

robimarko, 2025-03-05:
Its constantly building as we have a lot of package times, so its just a matter of one rounding finishing

davidandreoletti, 2025-03-06:
Got it.

davidandreoletti, 2025-03-06:
FIxed.

closed 2025-03-06 by davidandreoletti