openwrt/packages #26107: Possibly vulnerable packages overview

aparcar, 2025-03-06:
Hi all, with some recent work we got better integration in repology.org, which kindly tracks our packages and even show outdated/vulnerable packages.

Looking at the overview, it's clear that quite some packages require and upgrade or should be dropped if no longer maintained within OpenWrt. I invite everyone to have a quick look at the table below:

https://repology.org/projects/?inrepo=openwrt_24_10_x86_64&vulnerable=1

feckert, 2025-03-07:
@aparcar I would suggest to pin this to the issue overiew page.

Neustradamus, 2025-04-21:
To follow this important ticket.

brada4, 2026-01-05:
nginx external modules sit unupdated for 3-5 years, and they are masked by commit hash so that nobody sees anything wrong. Found 3 CVE-s there, ball is yours.

brada4, 2026-01-05:
nginx-naxsi

brada4, 2026-02-13:
Softether stable https://www.softether.org/9-about/News/904-SEVPN202301 (why prerelease is even there)

Yang-Wei-Ting, 2026-03-06:
https://github.com/openwrt/openwrt/blob/0c0b00422ed173820c2c7e57b160150fae70931e/package/Makefile#L143

```make
		$(STAGING_DIR_HOST)/bin/apk adbdump --format json packages.adb | \
			$(SCRIPT_DIR)/make-index-json.py -f apk -a "$(ARCH_PACKAGES)" - > index.json; \
```

We may need to provide a format that Repology can consume.

geldot, 2026-08-17:
> Hi, I'm trying to add support for OpenWrt 25.12 to Repology: [repology/repology-updater#1559](https://github.com/repology/repology-updater/pull/1559)
> 
> However, it looks like the `Packages.manifest` file is no longer available for 25.12. https://downloads.openwrt.org/releases/packages-25.12/x86_64/base/Packages.manifest 404
> 
> Is this related to the recent switch to APK? If so, does `packages.adb` now serve the same purpose that `Packages.manifest` used to?

Repology say they [require](https://github.com/repology/repology-updater/issues/1624#issuecomment-5181323353) the parser to be in rust, but the apk v3 binary format is too messy to develop a rust implementation just for this purpose, so the binary index will not help.

Hence the only hope is if @aparcar completes https://github.com/repology/repology-updater/issues/1473 (rich JSON manifest files) and provides a parser to Repology, and publishing these manifests both for releases and for nightly snapshots.

Right now identifying what's outdated or vulnerable is guesswork.

Yang-Wei-Ting, 2026-08-18:
> Right now identifying what's outdated or vulnerable is guesswork.

We still have [uscan](https://sdwalker.github.io/uscan/index.html)

sanderpqr, 2026-11-07:
I've opened a PR to re-create a package manifest that Repology could use. Mentioning it here for some visibility.

https://github.com/openwrt/openwrt/pull/26425