openwrt/packages #26119: QNAME minimisation is NOT enabled on your resolver

cmonty14, 2025-03-10:
Hi,
I have added this line to _/etc/unbound/unbound_srv.conf_:
```
## Privacy
# DNS Query Name Minimisation to Improve Privacy
qname-minimisation: yes
```
However, verification fails:
```
❯ dig @172.16.1.2 txt qnamemintest.internet.nl +short
a.b.qnamemin-test.internet.nl.
"NO - QNAME minimisation is NOT enabled on your resolver :("
```

Can you please advise how to fix this?

THX

brada4, 2025-03-13:
Yoi can &se /etc/config/unbound.
https://github.com/openwrt/packages/blob/214a8c47dc29b65f8052d1858e10d5244abad833/net/unbound/files/unbound.uci#L20

EricLuehrsen, 2025-03-15:
UCI already write this so your added line would be a redundant conflicting option. I am not sure which wins.