openwrt/packages #26121: unbound: service is not listening on localhost 127.0.0.1

cmonty14, 2025-03-11:
Hi,
I have configured [Unbound + odhcpd](https://github.com/openwrt/packages/tree/master/net/unbound/files#unbound-and-odhcpd), and now I want to enhance the setup with AdGuard Home as my primary DNS server.
In this setup _Unbound_ should work as recursive DNS server, means all DNS requests from a client are sent to AdGuard Home, and if the requested domain is not filtered out, it will be send to _Unbound_.
For this setup I would configure _Unbound_ to listen on `127.0.0.1:5335` and configure in AdGuard Home >Upstream-DNS-Server with this value: 127.0.0.1:5335.

However, using Unbound configured by UCI I get different interfaces:
```
root@router:~# netstat -tulpn | grep 53
tcp        0      0 172.16.10.2:53          0.0.0.0:*               LISTEN      3268/AdGuardHome
tcp        0      0 127.0.0.1:8953          0.0.0.0:*               LISTEN      6776/unbound
tcp        0      0 0.0.0.0:5335            0.0.0.0:*               LISTEN      6776/unbound
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      3268/AdGuardHome
tcp        0      0 172.16.1.2:53           0.0.0.0:*               LISTEN      3268/AdGuardHome
tcp        0      0 ::1:53                  :::*                    LISTEN      3268/AdGuardHome
tcp        0      0 ::1:8953                :::*                    LISTEN      6776/unbound
tcp        0      0 :::5335                 :::*                    LISTEN      6776/unbound
udp        0      0 172.16.10.2:53          0.0.0.0:*                           3268/AdGuardHome
udp        0      0 172.16.1.2:53           0.0.0.0:*                           3268/AdGuardHome
udp        0      0 127.0.0.1:53            0.0.0.0:*                           3268/AdGuardHome
udp        0      0 0.0.0.0:5335            0.0.0.0:*                           6776/unbound
udp        0      0 ::1:53                  :::*                                3268/AdGuardHome
udp        0      0 :::5335                 :::*                                6776/unbound

```
Can you please advise how to fix this?
THX

brada4, 2025-03-13:
AGH is listening on those ports?

EricLuehrsen, 2025-03-15:
Unbound is listening on all addreses (0.0.0.0) at 5335. You can add local-data zones or access control entries for how Unbound replies for each. Binding to a direct address can create restart tie-up with the physical interface.