openwrt/packages #26149: knot: broken knot.conf; template->max-journal-db-size + max-timer-db-size options are invalid as of 3.1

wrog, 2025-03-16:
after doing a fresh install of knot 3.1.4 on OpenWRT 24.10 (hardware is OpenWRT One, but I don't think this matters) using opkg, the following shows up in the log on reboot
```

Sat Mar 15 23:01:49 2025 daemon.err knot[3244]: error: config, file '/etc/knot/knot.conf', line 28, item max-'journal-db-size', value '20M' (invalid item)
Sat Mar 15 23:01:49 2025 daemon.crit knot[3244]: critical: failed to load configuration file '/etc/knot/knot.conf' (invalid item)
```

It turns out that (according to the [changelog](https://www.knot-dns.cz/docs/3.4/singlehtml/#upgrade-2-8-x-to-2-9-x) ) `max-journal-db-size` and `max-timer-db-size` were moved from the `template` section to the `database` section, the old names being kept around for backwards compatibility, but ignored with a deprecation notice as of 3.0 and completely removed as of 3.1.

The following change seems to fix things (or, at least, makes the errors go away even if it still [entirely expectedly] warns about empty zones [since I haven't put in any actual data yet])

```
modified   etc/knot/knot.conf
@@ -13,6 +13,8 @@ log:
 
 database:
     storage: "/var/lib/knot"
+    journal-db-max-size: 20M
+    timer-db-max-size: 5M
 
 remote:
 #  - id: secondary
@@ -25,8 +27,6 @@ template:
   - id: default
     storage: "/var/lib/knot"
     file: "%s.zone"
-    max-journal-db-size: 20M
-    max-timer-db-size: 5M
 
 zone:
 #    # Primary zone
```
I'd do a pull request but I have no idea how this file is being constructed in the build tree, but since the max-size lines are not present in the upstream version I'm assuming this is something that's being done specifically for OpenWRT (due to limited memory on some boxes?).

I also have this theory that the `knotd` service should start out disabled by default, since there will never be any point to running it until there's actual zone data available

BKPepe, 2025-03-16:
cc: @Payne-X6 and @salzmdan

salzmdan, 2025-03-16:
@wrog you are correct. We will fix the default config file (https://github.com/openwrt/packages/blob/master/net/knot/patches/02_knot.conf.patch).

PR #26172 (2025-04-10): knot: fix patch for configuration sample file
Maintainer: Daniel Salzman / @salzmdan
Compile tested: ARM Cortex-A9 vfpv3, Turris Omnia, TurrisOS 8.0.0 (hbl)
Run tested: ARM Cortex-A9 vfpv3, Turris Omnia, TurrisOS 8.0.0 (hbl)

Description:
- Fixes #26149
commit 087bc671 (2025-05-22): knot: fix patch for configuration sample

Fixes #26149

(cherry picked from commit 49e5ac2beb33d96c4ea06d740ebfd4c08998a068)