openwrt/packages #26169: HTTPS-DNS-Proxy intermittent disconnection on E8450

wolfdael, 2025-03-19:
Hi @stangri , thank you for your work. I have frequent disconnections on my E8450 and  can you please tell me if it's the package which has been recently updated or something to do with my configuration? Looking forward to your response.

```
root@OpenWrt:~# ubus call system board
{
        "kernel": "6.6.73",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Linksys E8450 (UBI)",
        "board_name": "linksys,e8450-ubi",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "mediatek/mt7622",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
root@OpenWrt:~# curl -V
curl 8.10.1 (aarch64-openwrt-linux-gnu) libcurl/8.10.1 mbedTLS/3.6.2 nghttp2/1.63.0
Release-Date: 2024-09-18
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
root@OpenWrt:~# dnsmasq --version
Dnsmasq version 2.90  Copyright (c) 2000-2024 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
root@OpenWrt:~# https-dns-proxy -V
2023.12.26-r4
root@OpenWrt:~# uci export dhcp
package dhcp

config dnsmasq
        option domainneeded '1'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option localservice '1'
        option ednspacket_max '1232'
        list rebind_domain 'plex.direct'
        option noresolv '1'
        list server '127.0.0.1#6053'
        list server '127.0.0.1#443'
        option doh_backup_noresolv '1'
        list doh_backup_server '127.0.0.1#6053'
        list doh_server '127.0.0.1#443'

config dhcp 'lan'
        option interface 'lan'
        option leasetime '12h'
        option dhcpv4 'server'
        option start '20'
        option limit '250'
        option ra 'server'
        option ra_slaac '0'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'
        option dhcpv6 'server'

[... 276 lines cut ...]
                                                "https-dns-proxy"
                                        ],
                                        [
                                                "run_script",
                                                "/etc/init.d/https-dns-proxy",
                                                "reload",
                                                "on_config_change"
                                        ]
                                ],
                                1000
                        ]
                ]
        }
}
root@OpenWrt:~# nslookup google.com 127.0.0.1:5053
nslookup: write to '127.0.0.1:5053': Connection refused
;; connection timed out; no servers could be reached

root@OpenWrt:~# nslookup google.com 127.0.0.1:5053
nslookup: write to '127.0.0.1:5053': Connection refused
;; connection timed out; no servers could be reached

root@OpenWrt:~#

```

stangri, 2025-03-19:
> I have frequent disconnections on my E8450

Can you please elaborate on that? Disconnections of what and from what? What are the symptoms?

> list server '127.0.0.1#6053'
>         list server '127.0.0.1#443'
>         option doh_backup_noresolv '1'
>         list doh_backup_server '127.0.0.1#6053'
>         list doh_server '127.0.0.1#443'

The doh_backup_server being the same as the currently active server is definitely something that shouldn't happen and an issue with the package if you can walk me thru on how you got to this, I'd appreciate that, it would help me fix the problem.

wolfdael, 2025-03-19:
This happened when I changed port from 443 to 6053 in the https-dns-proxy. Now I have read the docs and set the port to 5053.

```
        option doh_backup_noresolv '-1'
	option noresolv '1'
	list doh_backup_server ''
	list server '127.0.0.1#5053'
	list doh_server '127.0.0.1#5053'
```

This is what I get when it is running. Is this right or need to be edited? 
 And by disconnection, I mean there is no internet or there is an exclamation mark on the android Wi-Fi symbol. Only way to fix this is to restart or stop/start in https-dns-proxy luci to get the internet working.
I have been periodically restarting to get it working, but only now I realized the listen port should be 5053 and above. I probably had disconnections because I was using 443 mentioned on mullvad doh page.

So can you please tell me if the above dnsmasq setting is right for what I have posted above? I want to start fresh and see if the package still disconnects and report back here with logs.

wolfdael, 2025-03-23:
@stangri ,  this is from the syslog and I have also replied above to your prior questions. Something happened during this timeframe and I lost internet connection. All I did to fix it was to go into luci https dns proxy and click restart and got my connection back.

Can you please tell me what is causing this? Thank you.

```
Sun Mar 23 12:43:34 2025 daemon.err hostapd: nl80211: kernel reports: key addition failed
Sun Mar 23 12:43:34 2025 daemon.info hostapd: wl0-ap1: STA b2:f9:57:dc:89:bb IEEE 802.11: associated (aid 2)
Sun Mar 23 12:43:34 2025 daemon.notice hostapd: wl0-ap1: AP-STA-CONNECTED b2:f9:57:dc:89:bb auth_alg=ft
Sun Mar 23 12:43:34 2025 daemon.notice hostapd: wl1-ap1: Prune association for b2:f9:57:dc:89:bb
Sun Mar 23 12:43:34 2025 daemon.notice hostapd: wl1-ap1: AP-STA-DISCONNECTED b2:f9:57:dc:89:bb
Sun Mar 23 12:44:04 2025 daemon.info hostapd: wl1-ap1: STA b2:f9:57:dc:89:bb IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Mar 23 12:45:00 2025 cron.err crond[9510]: USER root pid 18394 cmd scp /tmp/dhcp.leases root@192.168.86.245:/tmp/dhcp.leases
Sun Mar 23 12:48:37 2025 daemon.err hostapd: nl80211: kernel reports: key addition failed
Sun Mar 23 12:48:37 2025 daemon.info hostapd: wl1-ap1: STA 24:29:34:98:fb:8e IEEE 802.11: associated (aid 1)
Sun Mar 23 12:48:37 2025 daemon.notice hostapd: wl1-ap1: AP-STA-CONNECTED 24:29:34:98:fb:8e auth_alg=ft
Sun Mar 23 12:50:00 2025 cron.err crond[9510]: USER root pid 18398 cmd scp /tmp/dhcp.leases root@192.168.86.245:/tmp/dhcp.leases
Sun Mar 23 12:53:05 2025 daemon.notice hostapd: wl0-ap1: AP-STA-DISCONNECTED b2:f9:57:dc:89:bb
Sun Mar 23 12:53:05 2025 daemon.err hostapd: nl80211: kernel reports: key addition failed
Sun Mar 23 12:53:05 2025 daemon.info hostapd: wl0-ap1: STA b2:f9:57:dc:89:bb IEEE 802.11: associated (aid 2)
Sun Mar 23 12:53:05 2025 daemon.notice hostapd: wl0-ap1: AP-STA-CONNECTED b2:f9:57:dc:89:bb auth_alg=ft
Sun Mar 23 12:54:06 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan.4) 192.168.87.216 3c:0a:f3:be:26:8b
Sun Mar 23 12:54:06 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan.4) 192.168.87.216 3c:0a:f3:be:26:8b BrotherPrinter
Sun Mar 23 12:55:00 2025 cron.err crond[9510]: USER root pid 18411 cmd scp /tmp/dhcp.leases root@192.168.86.245:/tmp/dhcp.leases
Sun Mar 23 12:59:19 2025 daemon.err uhttpd[2060]: [info] luci: accepted login on / for root from 192.168.86.145
Sun Mar 23 13:00:00 2025 cron.err crond[9510]: USER root pid 19193 cmd scp /tmp/dhcp.leases root@192.168.86.245:/tmp/dhcp.leases
Sun Mar 23 13:00:40 2025 daemon.notice hostapd: wl0-ap1: AP-STA-DISCONNECTED b2:f9:57:dc:89:bb
Sun Mar 23 13:00:40 2025 daemon.info hostapd: wl0-ap1: STA b2:f9:57:dc:89:bb IEEE 802.11: disassociated due to inactivity
Sun Mar 23 13:00:41 2025 daemon.info hostapd: wl0-ap1: STA b2:f9:57:dc:89:bb IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Mar 23 13:02:37 2025 daemon.err uhttpd[2060]: [info] luci: accepted login on / for root from 192.168.86.145
Sun Mar 23 13:02:45 2025 user.notice https-dns-proxy [20344]: Starting https-dns-proxy 2023.12.26-r4 instances ✓
Sun Mar 23 13:02:45 2025 user.notice https-dns-proxy [20344]: Setting trigger for wan ✓
Sun Mar 23 13:03:41 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan.4) 192.168.87.247 84:3e:1d:8f:75:ba
Sun Mar 23 13:03:41 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan.4) 192.168.87.247 84:3e:1d:8f:75:ba RevanTV
Sun Mar 23 13:05:00 2025 cron.err crond[9510]: USER root pid 20702 cmd scp /tmp/dhcp.leases root@192.168.86.245:/tmp/dhcp.leases

```

wolfdael, 2025-03-31:
[image]

@stangri I have disconnected and reconnected Wi-Fi on my phone, laptop and nothing works until I go into luci https-dns-proxy and click restart. As soon as I click restart, internet is restored.

wolfdael, 2025-03-31:
UPDATED SETTINGS:

```
root@OpenWrt:~# ubus call system board
{
        "kernel": "6.6.73",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Linksys E8450 (UBI)",
        "board_name": "linksys,e8450-ubi",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "mediatek/mt7622",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
root@OpenWrt:~# curl -V
curl 8.10.1 (aarch64-openwrt-linux-gnu) libcurl/8.10.1 mbedTLS/3.6.2 nghttp2/1.63.0
Release-Date: 2024-09-18
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
root@OpenWrt:~# dnsmasq --version
Dnsmasq version 2.90  Copyright (c) 2000-2024 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
root@OpenWrt:~# https-dns-proxy -V
2023.12.26-r4
root@OpenWrt:~# uci export dhcp
package dhcp

config dnsmasq
        option domainneeded '1'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option localservice '1'
        option ednspacket_max '1232'
        list rebind_domain 'plex.direct'
        list server '127.0.0.1#5053'
        option doh_backup_noresolv '-1'
        option noresolv '1'
        list doh_backup_server '127.0.0.1#5053'
        list doh_server '127.0.0.1#5053'

config dhcp 'lan'
        option interface 'lan'
        option leasetime '12h'
        option dhcpv4 'server'
        option start '20'
        option limit '250'
        option ra 'server'
        option ra_slaac '0'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'
        option dhcpv6 'server'

config dhcp 'wan'
[... 279 lines cut ...]
                                                "run_script",
                                                "/etc/init.d/https-dns-proxy",
                                                "reload",
                                                "on_config_change"
                                        ]
                                ],
                                1000
                        ]
                ]
        }
}
root@OpenWrt:~# nslookup google.com 127.0.0.1:5053
Server:         127.0.0.1:5053
Address:        127.0.0.1:5053

Non-authoritative answer:
Name:   google.com
Address: 172.217.169.14

Non-authoritative answer:
Name:   google.com
Address: 2a00:1450:4009:817::200e

root@OpenWrt:~#
```

egc112, 2025-03-31:
Stangri knows the package much better than I do but a few things you might need to look into:
```
config dhcp 'lan'
        option interface 'lan'
        option leasetime '12h'
        option dhcpv4 'server'
        option start '20'       <<<<<<<
        option limit '250'      <<<<<<<  unless you have a non standard subnetmask `start+ limit` must not be more than 254
```

The bootstrap servers do not seem publicly available:
```
option bootstrap_dns '194.242.2.4,2a07:e340::4'
```

Maybe just use https-dnbs-proxy with its default settings tot test if it works better, so also without VPN etc.

wolfdael, 2025-03-31:
@egc112 

> You start DHCP at 20 and have a limit of 250 which together makes 270 but there are only 254 address on your subnet so lower option limit to 230

I have permanant addresses after 230 and my dumb AP is at 245. I believe it will break some things if I just change it. Is there something else I can do?

> The bootstrap servers do not seem publicly available

They are publicly available here - https://mullvad.net/en/help/dns-over-https-and-dns-over-tls
and I don't use a VPN on openwrt

egc112, 2025-03-31:
See for yourself
```
PS C:\Users\egc> nslookup openwrt.org 194.242.2.4
Server:  UnKnown
Address:  194.242.2.4

DNS request timed out.
    timeout was 2 seconds.
*** Request to UnKnown timed-out
```

> I have permanant addresses after 230 and my dumb AP is at 245. I believe it will break some things if I just change it. Is there something else I can do?

I have tried to explain it twice to you and unfortunately do not know how else to explain it it but there are only 254 DHCP address on your /24 subnet and not 20 + 250 = 270.
For me it is even good practice to have my static addresses outside the DHCP range but that is not really necessary.

wolfdael, 2025-04-01:
@egc112 

I've changed it to 230. 

```
config dhcp 'lan'
        option interface 'lan'
        option leasetime '12h'
        option dhcpv4 'server'
        option start '20'
        option limit '230'
        option ra 'server'
        option ra_slaac '0'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'
        option dhcpv6 'server'
```

stangri, 2025-04-01:
> [](https://github.com/egc112)
> 
> I've changed it to 230.

Do you still experience issues? If so, for troubleshooting, can you try disabling IPv6 on your PPPoE connection?

egc112, 2025-04-01:
> [@egc112](https://github.com/egc112)
>
> I've changed it to 230.
>
> ```
> config dhcp 'lan'
>         option interface 'lan'
>         option leasetime '12h'
>         option dhcpv4 'server'
>         option start [...]

Excellent now the other more important error in your setup, the lack of a working bootstrap server.
The bootstrap server is used to resolve the name of the DoT server,  that has to be a working publicly available DNS server and in my testing 194.242.2.4 does not work.
**Replace your current bootstrap server with e.g. 9.9.9.9**
Reboot and test again.

wolfdael, 2025-04-08:
```
Tue Apr  8 08:00:00 2025 cron.err crond[1992]: USER root pid 25747 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:03:08 2025 daemon.notice hostapd: wl0-ap1: AP-STA-DISCONNECTED 9e:e1:61:83:d7:f2
Tue Apr  8 08:03:08 2025 daemon.err hostapd: nl80211: kernel reports: key addition failed
Tue Apr  8 08:03:08 2025 daemon.err hostapd: nl80211: kernel reports: key addition failed
Tue Apr  8 08:05:00 2025 cron.err crond[1992]: USER root pid 25767 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:06:22 2025 daemon.info hostapd: wl0-ap1: STA 9e:e1:61:83:d7:f2 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Apr  8 08:10:00 2025 cron.err crond[1992]: USER root pid 25771 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:15:00 2025 cron.err crond[1992]: USER root pid 25783 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:20:00 2025 cron.err crond[1992]: USER root pid 25787 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:25:00 2025 cron.err crond[1992]: USER root pid 25808 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:30:00 2025 cron.err crond[1992]: USER root pid 25813 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:35:00 2025 cron.err crond[1992]: USER root pid 25825 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:37:39 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan.4) 192.168.87.222 9e:e1:61:83:d7:f2
Tue Apr  8 08:37:39 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan.4) 192.168.87.222 9e:e1:61:83:d7:f2 Galaxy-A55-5G
Tue Apr  8 08:37:40 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan.99) 192.168.86.231 84:b8:b8:54:55:c8
Tue Apr  8 08:37:40 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan.99) 192.168.86.231 84:b8:b8:54:55:c8 LenovoSmartClock
Tue Apr  8 08:37:49 2025 daemon.err hostapd: nl80211: kernel reports: key addition failed
Tue Apr  8 08:37:49 2025 daemon.info hostapd: wl0-ap1: STA 9e:e1:61:83:d7:f2 IEEE 802.11: associated (aid 2)
Tue Apr  8 08:37:49 2025 daemon.notice hostapd: wl0-ap1: AP-STA-CONNECTED 9e:e1:61:83:d7:f2 auth_alg=ft
Tue Apr  8 08:40:00 2025 cron.err crond[1992]: USER root pid 25830 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:41:33 2025 daemon.notice hostapd: wl0-ap1: AP-STA-DISCONNECTED 9e:e1:61:83:d7:f2
Tue Apr  8 08:41:33 2025 daemon.err hostapd: nl80211: kernel reports: key addition failed
Tue Apr  8 08:41:33 2025 daemon.info hostapd: wl0-ap1: STA 9e:e1:61:83:d7:f2 IEEE 802.11: associated (aid 2)
Tue Apr  8 08:41:33 2025 daemon.notice hostapd: wl0-ap1: AP-STA-CONNECTED 9e:e1:61:83:d7:f2 auth_alg=ft
Tue Apr  8 08:45:00 2025 cron.err crond[1992]: USER root pid 25852 cmd scp /tmp/dhcp.leases root@192.168.86.175:/tmp/dhcp.leases
Tue Apr  8 08:47:27 2025 daemon.notice hostapd: wl0-ap1: AP-STA-DISCONNECTED b2:f9:57:dc:89:bb
Tue Apr  8 08:48:34 2025 daemon.info hostapd: wl0-ap1: STA b2:f9:57:dc:89:bb IEEE 802.11: authenticated
Tue Apr  8 08:48:34 2025 daemon.info hostapd: wl0-ap1: STA b2:f9:57:dc:89:bb IEEE 802.11: associated (aid 1)
Tue Apr  8 08:48:34 2025 daemon.notice hostapd: wl0-ap1: AP-STA-CONNECTED b2:f9:57:dc:89:bb auth_alg=open
Tue Apr  8 08:48:34 2025 daemon.info hostapd: wl0-ap1: STA b2:f9:57:dc:89:bb WPA: pairwise key handshake completed (RSN)
Tue Apr  8 08:48:34 2025 daemon.notice hostapd: wl0-ap1: EAPOL-4WAY-HS-COMPLETED b2:f9:57:dc:89:bb
Tue Apr  8 08:48:34 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan.4) b2:f9:57:dc:89:bb
Tue Apr  8 08:48:34 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan.4) 192.168.87.130 b2:f9:57:dc:89:bb
Tue Apr  8 08:48:34 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan.4) 192.168.87.130 b2:f9:57:dc:89:bb
Tue Apr  8 08:48:34 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan.4) 192.168.87.130 b2:f9:57:dc:89:bb UVA
Tue Apr  8 08:48:34 2025 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Tue Apr  8 08:48:39 2025 daemon.notice hostapd: wl1-ap1: AP-STA-DISCONNECTED f2:af:83:68:29:48
Tue Apr  8 08:48:40 2025 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Tue Apr  8 08:48:57 2025 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Tue Apr  8 08:49:08 2025 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Tue Apr  8 08:49:11 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan.4) f2:af:83:68:29:48
Tue Apr  8 08:49:11 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan.4) 192.168.87.211 f2:af:83:68:29:48
Tue Apr  8 08:49:11 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan.4) 192.168.87.211 f2:af:83:68:29:48
Tue Apr  8 08:49:11 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan.4) 192.168.87.211 f2:af:83:68:29:48 Pixel-7a
Tue Apr  8 08:49:17 2025 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Tue Apr  8 08:49:51 2025 daemon.err hostapd: nl80211: kernel reports: key addition failed
Tue Apr  8 08:49:51 2025 daemon.info hostapd: wl1-ap1: STA f2:af:83:68:29:48 IEEE 802.11: associated (aid 1)
Tue Apr  8 08:49:51 2025 daemon.notice hostapd: wl1-ap1: AP-STA-CONNECTED f2:af:83:68:29:48 auth_alg=ft
Tue Apr  8 08:49:52 2025 daemon.notice hostapd: wl1-ap1: AP-STA-DISCONNECTED f2:af:83:68:29:48
[... cut ...]
I changed my DoH resolvers to ControlD with 9.9.9.9 in bootstrap dns like adviced. I changed the resolver because I used https://dnsspeedtest.online/ and mullvad was unavailable for my region. 
However, even after ControlD DoH and 9.9.9.9 as bootstrap dns I got a internet disconnection in the time range syslog provided above. 

And I am sorry, I will be unable to disable ipv6 even for troubleshooting because I am under ipv4 CGNAT and all my vital services which are exposed to the internet, run through ipv6 reverse proxy. 
However, I am happy to fully invest my time , monitor and provide any logs necessary. Thank you!

stangri, 2025-04-08:
@wolfdael thank you for capturing the log. I'm quite busy with my paying job until the early May, but I'll try to go over the log and see if anything pops up right after.

wolfdael, 2025-04-10:
@stangri Can you please tell me where and how to enable this option?

stangri, 2025-04-10:
@wolfdael is there anything which is unclear from the documentation linked in a previous reply?

wolfdael, 2025-04-11:
@stangri 

""Configuration contains the general (named) “main” config section where you can configure which dnsmasq settings the service will automatically affect and the typed (unnamed) https-dns-proxy instance settings.""

Does this explicitly state where to do place the procd_trigger_wan6 or atleast in the section below procd_trigger_wan6? It's implicitly trying to tell something. 
If I were an OpenWrt or a network engineer I'd know if it's the main option or proxy option and where this is located in the file system.
It's not that I didn't read the docs, it's a genuine confusion.

stangri, 2025-04-14:
> It's not that I didn't read the docs, it's a genuine confusion.

You're right, that intro sentence could definitely be improved.

For now I've updated the sample config in https://docs.openwrt.melmac.net/https-dns-proxy/#ConfigurationSettings section, so in a few hours when HTML is rebuilt you'll see all the options (including `procd_trigger_wan6`) in the sample so hopefully it will be easy to edit your config.