openwrt/packages #26170: Experiencing "Verify it's you" Errors with Google Services Using OpenWrt's mwan3

shrikant2002, 2025-03-19:
# #

[image]

Hello everyone,

I'm encountering an issue where a specific client on my network is frequently prompted with Google's "Verify it's you" security checks, and I suspect it might be related to my network configuration. Here's an overview of my setup:

Router Firmware: OpenWrt 22.03.7
Multi-WAN Management: Using mwan3 for load balancing
WAN Interfaces: Two active connections labeled as wan and wanb
Issue Details:

The client device with the IP address 192.168.100.164 (referred to as kedar_desk) frequently encounters "Verify it's you" prompts, especially when accessing YouTube Studio.

Troubleshooting Steps Taken:

Increased Sticky Timeout: Adjusted the sticky timeout to 3600 seconds to maintain session persistence, but the issue persists.

Assigned Specific Policy: Applied a wan_only policy to kedar_desk to ensure all its traffic routes through a single WAN interface, yet the problem continues.

Reviewed System Logs: Checked system logs for errors related to this issue but found none.

Verified mwan3 Status: Confirmed that mwan3 is functioning correctly, with all interfaces showing as online.

mwan3 Configuration:

Below is the relevant portion of my mwan3 configuration:

config rule 'kedar_desk'
	option family 'ipv4'
	option proto 'all'
	option src_ip '192.168.100.164/32'
	option sticky '1'
	option use_policy 'wan_wanb_fail'

Seeking Advice On:

Session Persistence: Despite setting a sticky timeout and assigning a specific policy, the client still encounters verification prompts. Are there additional configurations within mwan3 that could enhance session persistence for this client?

Alternative Solutions: Has anyone experienced similar issues with specific clients and Google services in a multi-WAN setup? If so, what solutions or workarounds have been effective?

Any insights or recommendations would be greatly appreciated. Thank you in advance for your assistance!
  
  
Here is my mwan3 config

root@Load-Balancer2:\~# cat /etc/config/mwan3

config globals 'globals'

option mmx\_mask '0x3F00'

option logging '1'

option loglevel 'info'

list rt\_table\_lookup '220'

  

config interface 'wan'

option enabled '1'

option family 'ipv4'

option initial\_state 'online'

option track\_method 'ping'

option count '1'

option size '56'

option max\_ttl '60'

option timeout '4'

option failure\_interval '5'

option recovery\_interval '5'

list flush\_conntrack 'ifup'

list flush\_conntrack 'ifdown'

option down '3'

option up '3'

list track\_ip '8.8.8.8'

list track\_ip '1.1.1.1'

option reliability '1'

option interval '5'

  

config interface 'wanb'

option family 'ipv4'

option reliability '1'

option initial\_state 'online'

option track\_method 'ping'

option count '1'

option size '56'

option max\_ttl '60'

option timeout '4'

option failure\_interval '5'

option recovery\_interval '5'

list flush\_conntrack 'ifup'

list flush\_conntrack 'ifdown'

option enabled '1'

option down '3'

option up '3'

list track\_ip '8.8.4.4'

list track\_ip '1.0.0.1'

option interval '5'

  

config policy 'wan\_only'

option last\_resort 'unreachable'

list use\_member 'wan\_m1\_w1'

  

config policy 'wanb\_only'

option last\_resort 'unreachable'

list use\_member 'wanb\_m1\_w1'

  

config policy 'balanced'

option last\_resort 'unreachable'

list use\_member 'wan\_m1\_w1'

list use\_member 'wanb\_m1\_w2'

  

config policy 'wan\_wanb'

option last\_resort 'unreachable'

list use\_member 'wan\_m1\_w2'

list use\_member 'wanb\_m1\_w1'

  

config policy 'wanb\_wan'

option last\_resort 'unreachable'

list use\_member 'wanb\_m1\_w2'

list use\_member 'wan\_m1\_w1'

  

config rule 'kedar\_desk'

option family 'ipv4'

option proto 'all'

option src\_ip '192.168.100.164/32'

option sticky '1'

option use\_policy 'wan\_wanb\_fail'

  

config rule 'default\_rule\_v4'

option dest\_ip '0.0.0.0/0'

option use\_policy 'balanced'

option family 'ipv4'

option proto 'all'

option sticky '0'

  

config rule 'https'

option sticky '1'

option proto 'tcp'

option family 'ipv4'

option dest\_port '53,443'

option use\_policy 'wan\_wanb\_fail'

  

config member 'wan\_m1\_w1'

option interface 'wan'

option metric '1'

option weight '1'

  

config member 'wanb\_m1\_w2'

option interface 'wanb'

option metric '1'

option weight '2'

  

config member 'wan\_m1\_w2'

option interface 'wan'

option metric '1'

option weight '2'

  

config member 'wanb\_m1\_w1'

option interface 'wanb'

option metric '1'

option weight '1'

  

config member 'wanb\_m2\_w1'

option interface 'wanb'

option metric '2'

option weight '1'

  

config member 'wan\_m2\_w1'

option interface 'wan'

option metric '2'

option weight '1'

  

config member 'wanb\_m2\_w2'

option interface 'wanb'

option metric '2'

option weight '2'

  

config member 'wan\_m2\_w2'

option interface 'wan'

option metric '2'

option weight '2'

  

config policy 'wan\_wanb\_fail'

option last\_resort 'unreachable'

list use\_member 'wan\_m1\_w1'

list use\_member 'wanb\_m2\_w2'

***********

mwan3 status

Interface status:
 interface wan is online 01h:11m:57s, uptime 17h:23m:44s and tracking is active
 interface wanb is online 01h:11m:58s, uptime 13h:23m:28s and tracking is active

Current ipv4 policies:
balanced:
 wanb (66%)
 wan (33%)
wan_only:
 wan (100%)
wan_wanb:
 wanb (33%)
 wan (66%)
wan_wanb_fail:
 wan (100%)
wanb_only:
 wanb (100%)
wanb_wan:
 wan (33%)
 wanb (66%)

Current ipv6 policies:
balanced:
 unreachable
wan_only:
 unreachable
wan_wanb:
 unreachable
wan_wanb_fail:
 unreachable
wanb_only:
 unreachable
wanb_wan:
 unreachable

Directly connected ipv4 networks:
127.255.255.255
127.0.0.0/8
224.0.0.0/3
127.0.0.1
172.67.1.176
202.134.149.67
127.0.0.0
172.12.54.2
172.22.111.34
192.168.100.0
192.168.100.255
192.168.100.1
192.168.100.0/24

Directly connected ipv6 networks:
fe80::/64
fe80::d315:f105:f4:57e
fe80::5c7:e89b:79ff:ca8f
fe80::f0:924a
fe80::b13b:bb21:b534:c955

Active ipv4 user rules:
  917  553K S kedar_desk  all  --  *      *       192.168.100.164      0.0.0.0/0            
10896 3512K - balanced  all  --  *      *       0.0.0.0/0            0.0.0.0/0            
    0     0 S https  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 53,443 

Active ipv6 user rules:

brada4, 2025-03-20:
Try supported version before continuing here, and go(gle errors are emitted by ... not very hard to guess ... goigle, so ask them.

shrikant2002, 2025-03-20:
> Try supported version before continuing here, and google errors are emitted by ... not very hard to guess ... goigle, so ask them.

Isn't version 22.03.7 supported with mwan3? I'm asking here because it only happens while using this network with mwan3 with 2 wans.

brada4, 2025-03-20:
That OpenWrt version is no longer supported as in being developed. But versions will not impact your relation to google services