openwrt/packages #26175: coreutils fails to build using gcc14

graysky2, 2025-03-19:
Recently, building coreutils ends in errors
```
lib/randint.c: In function 'randint_all_new':
lib/randint.c:87:43: error: 'nullptr' undeclared (first use in this function); did you mean 'nullptr_t'?
   87 |   return (source ? randint_new (source) : nullptr);
      |                                           ^~~~~~~
      |                                           nullptr_t
lib/randint.c:87:43: note: each undeclared identifier is reported only once for each function it appears in
```

Full build log: https://gist.github.com/graysky2/3ea412677130361bcbe2b5adf8396f7c

hardfalcon, 2025-03-19:
Possibly relevant information:
https://www.gnu.org/software/gnulib/manual/html_node/nullptr.html
> Portability problems fixed by Gnulib:
> 
> - Some platforms lack nullptr: For C: GCC 12, Clang 15, and other pre-2023 C compilers. For C++: pre-2011 C++ compilers.
> - Some platforms incorrectly warn when nullptr is a sentinel argument: GCC 13.2 and 14.0. 
> 
> Portability problems not fixed by Gnulib: 
> […]
> - GCC 14 defines nullptr_t even when <stddef.h> is not included. This bug should be fixed in GCC 15.

hardfalcon, 2025-03-20:
Additional remark: coreutils 9.5 builds just fine even with GCC 14, so this seems to be an issue that only affects coreutils 9.6. I'm currently using the following patch (applied to the root directory of my openwrt buildroot) to build my images with coreutils 9.5:

```patch
diff --git a/feeds/packages/utils/coreutils/Makefile b/feeds/packages/utils/coreutils/Makefile
index 8e6bf8b097f6..5f98c6f8ee3c 100644
--- a/feeds/packages/utils/coreutils/Makefile
+++ b/feeds/packages/utils/coreutils/Makefile
@@ -8,12 +8,12 @@
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=coreutils
-PKG_VERSION:=9.6
-PKG_RELEASE:=2
+PKG_VERSION:=9.5
+PKG_RELEASE:=0
 
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
 PKG_SOURCE_URL:=@GNU/coreutils
-PKG_HASH:=7a0124327b398fd9eb1a6abde583389821422c744ffa10734b24f557610d3283
+PKG_HASH:=cd328edeac92f6a665de9f323c93b712af1858bc2e0d88f3f7100469470a1b8a
 
 PKG_MAINTAINER:=Jo-Philipp Wich <jo@mein.io>
 PKG_LICENSE:=GPL-3.0-or-later
@@ -95,7 +95,7 @@ endef
 define GenPlugin
  define Package/$(1)
    $(call Package/coreutils/Default)
-   DEPENDS:=+coreutils $(DEPENDS_$(2))
+   DEPENDS:=coreutils $(DEPENDS_$(2))
    TITLE:=Utility $(2) from the GNU core utilities
    ALTERNATIVES:=$(ALTS_$(2))
  endef
diff --git a/feeds/packages/utils/coreutils/patches/001-no_docs_man_tests.patch b/feeds/packages/utils/coreutils/patches/001-no_docs_man_tests.patch
index e64ceab99254..e82578bac74d 100644
--- a/feeds/packages/utils/coreutils/patches/001-no_docs_man_tests.patch
+++ b/feeds/packages/utils/coreutils/patches/001-no_docs_man_tests.patch
@@ -10,7 +10,7 @@
  EXTRA_DIST =				\
    .mailmap				\
 @@ -211,6 +211,3 @@ AM_CPPFLAGS = -Ilib -I$(top_srcdir)/lib
- include $(top_srcdir)/gl/local.mk
+ 
  include $(top_srcdir)/lib/local.mk
  include $(top_srcdir)/src/local.mk
 -include $(top_srcdir)/doc/local.mk
@@ -18,7 +18,7 @@
 -include $(top_srcdir)/tests/local.mk
 --- a/Makefile.in
 +++ b/Makefile.in
-@@ -4389,11 +4389,7 @@ RECURSIVE_TARGETS = all-recursive check-
+@@ -4149,11 +4149,7 @@ RECURSIVE_TARGETS = all-recursive check-
  	install-ps-recursive install-recursive installcheck-recursive \
  	installdirs-recursive pdf-recursive ps-recursive \
  	tags-recursive uninstall-recursive
@@ -31,22 +31,22 @@
  am__vpath_adj_setup = srcdirstrip=`echo "$(srcdir)" | sed 's|.|.|g'`;
  am__vpath_adj = case $$p in \
      $(srcdir)/*) f=`echo "$$p" | sed "s|^$$srcdirstrip/||"`;; \
-@@ -4643,11 +4639,10 @@ am__DIST_COMMON = $(doc_coreutils_TEXINF
+@@ -4403,10 +4399,10 @@ am__DIST_COMMON = $(doc_coreutils_TEXINF
  	$(top_srcdir)/build-aux/missing \
  	$(top_srcdir)/build-aux/test-driver \
  	$(top_srcdir)/build-aux/texinfo.tex \
 -	$(top_srcdir)/build-aux/ylwrap $(top_srcdir)/doc/local.mk \
+-	$(top_srcdir)/lib/alloca.c $(top_srcdir)/lib/config.hin \
+-	$(top_srcdir)/lib/local.mk $(top_srcdir)/man/local.mk \
+-	$(top_srcdir)/src/local.mk $(top_srcdir)/src/single-binary.mk \
 +	$(top_srcdir)/build-aux/ylwrap \
- 	$(top_srcdir)/gl/local.mk $(top_srcdir)/lib/alloca.c \
--	$(top_srcdir)/lib/config.hin $(top_srcdir)/lib/local.mk \
[... 35 lines cut ...]
-@@ -9132,7 +9127,7 @@ all: $(BUILT_SOURCES)
+@@ -8437,7 +8433,7 @@ all: $(BUILT_SOURCES)
  .SUFFIXES: .1 .c .dvi .log .o .obj .pl .pl$(EXEEXT) .ps .sh .sh$(EXEEXT) .trs .x .xpl .xpl$(EXEEXT) .y
  am--refresh: Makefile
  	@:
--$(srcdir)/Makefile.in:  $(srcdir)/Makefile.am $(top_srcdir)/gl/local.mk $(top_srcdir)/lib/local.mk $(srcdir)/lib/gnulib.mk $(top_srcdir)/src/local.mk $(srcdir)/src/cu-progs.mk $(top_srcdir)/src/single-binary.mk $(top_srcdir)/doc/local.mk $(top_srcdir)/man/local.mk $(top_srcdir)/tests/local.mk $(am__configure_deps)
-+$(srcdir)/Makefile.in:  $(srcdir)/Makefile.am $(top_srcdir)/gl/local.mk $(top_srcdir)/lib/local.mk $(srcdir)/lib/gnulib.mk $(top_srcdir)/src/local.mk $(srcdir)/src/cu-progs.mk $(top_srcdir)/src/single-binary.mk $(am__configure_deps)
+-$(srcdir)/Makefile.in:  $(srcdir)/Makefile.am $(top_srcdir)/lib/local.mk $(srcdir)/lib/gnulib.mk $(top_srcdir)/src/local.mk $(srcdir)/src/cu-progs.mk $(top_srcdir)/src/single-binary.mk $(top_srcdir)/doc/local.mk $(top_srcdir)/man/local.mk $(top_srcdir)/tests/local.mk $(am__configure_deps)
++$(srcdir)/Makefile.in:  $(srcdir)/Makefile.am $(top_srcdir)/lib/local.mk $(srcdir)/lib/gnulib.mk $(top_srcdir)/src/local.mk $(srcdir)/src/cu-progs.mk $(top_srcdir)/src/single-binary.mk $(am__configure_deps)
  	@for dep in $?; do \
  	  case '$(am__configure_deps)' in \
  	    *$$dep*) \
-@@ -9154,7 +9149,7 @@ Makefile: $(srcdir)/Makefile.in $(top_bu
+@@ -8459,7 +8455,7 @@ Makefile: $(srcdir)/Makefile.in $(top_bu
  	    echo ' cd $(top_builddir) && $(SHELL) ./config.status $@ $(am__maybe_remake_depfiles)'; \
  	    cd $(top_builddir) && $(SHELL) ./config.status $@ $(am__maybe_remake_depfiles);; \
  	esac;
--$(top_srcdir)/gl/local.mk $(top_srcdir)/lib/local.mk $(srcdir)/lib/gnulib.mk $(top_srcdir)/src/local.mk $(srcdir)/src/cu-progs.mk $(top_srcdir)/src/single-binary.mk $(top_srcdir)/doc/local.mk $(top_srcdir)/man/local.mk $(top_srcdir)/tests/local.mk $(am__empty):
-+$(top_srcdir)/gl/local.mk $(top_srcdir)/lib/local.mk $(srcdir)/lib/gnulib.mk $(top_srcdir)/src/local.mk $(srcdir)/src/cu-progs.mk $(top_srcdir)/src/single-binary.mk $(am__empty):
+-$(top_srcdir)/lib/local.mk $(srcdir)/lib/gnulib.mk $(top_srcdir)/src/local.mk $(srcdir)/src/cu-progs.mk $(top_srcdir)/src/single-binary.mk $(top_srcdir)/doc/local.mk $(top_srcdir)/man/local.mk $(top_srcdir)/tests/local.mk $(am__empty):
++$(top_srcdir)/lib/local.mk $(srcdir)/lib/gnulib.mk $(top_srcdir)/src/local.mk $(srcdir)/src/cu-progs.mk $(top_srcdir)/src/single-binary.mk $(am__empty):
  
  $(top_builddir)/config.status: $(top_srcdir)/configure $(CONFIG_STATUS_DEPENDENCIES)
  	$(SHELL) ./config.status --recheck
```

graysky2, 2025-03-20:
Does it make sense to revert the 9.6 update and replace it with the 9.5 update?

hardfalcon, 2025-03-20:
> Does it make sense to revert the 9.6 update and replace it with the 9.5 update?

That's not my call to make, but I guess knowing the issue was introduced somewhere between coreutils versions 9.5 and 9.6 at least makes bisecting the origin of this issue a little easier.

hardfalcon, 2025-03-22:
Adding `PKG_FIXUP:=autoreconf` solves this issue for me, so I've opened https://github.com/openwrt/packages/pull/26195

hnyman, 2025-03-22:
I merged the gcc14-oriented fix proposed on #26195 , as it compiled ok in the CI testing for all tested targets (and for my own DL-WRX36 build, with gcc 13), but now it seems to fail in buildbot (with gcc 13). E.g. 
https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a7_vfpv4/packages/coreutils/compile.txt

```
/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a7+vfpv4_gcc-13.3.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: lib/libcoreutils.a(libcoreutils_a-unicodeio.o): in function `unicode_to_mb':
unicodeio.c:(.text+0x148): undefined reference to `u8_uctomb_aux'
/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a7+vfpv4_gcc-13.3.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: lib/libcoreutils.a(libcoreutils_a-iconv.o): in function `rpl_iconv':
iconv.c:(.text+0x438): undefined reference to `u8_uctomb_aux'
/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a7+vfpv4_gcc-13.3.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: iconv.c:(.text+0x47c): undefined reference to `u8_mbtoucr'
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:10524: src/printf] Error 1
```

Apparently the autoreconf does not play well in our build env.

graysky2, 2025-03-22:
Can it be applied conditionally only against gcc 14?

hnyman, 2025-03-22:
> Also, is there tracker detailing broken packages with gcc 14? 

Not to my knowledge. Buildbot (and of course most private builds) default to the default gcc 13, so 14 is currently probably used by just rather small population.
@neheb  did a major effort in 2024 to fix lots of packages for gcc14, but like this coreutils version bump shows, there will be new breakages.

I will wait a bit to see if coreutils now fails for all targets, and if so, then I revert the autoreconf commit.

There was also an alternative fix proposal in the upgrade commit discussion:
https://github.com/openwrt/packages/commit/2d3f68cc8c165b1fa01308f566394cbd7d06766f#commitcomment-153860241

> using -std=c17 also allows compilation to pass,

Possibly that would be less invasive than the autoreconf.

graysky2, 2025-03-22:
> Can it be applied conditionally only against gcc 14?

@hnyman - would something like this work:

```
ifeq ($(CONFIG_GCC_VERSION_14),y)
  PKG_FIXUP:=autoreconf
endif
```

hnyman, 2025-03-22:
Thanks, that #26200 may work as a temporary solution, but likely not a permanent solution.

I suspect that the same compilation problem that autoreconf causes now for buildbot with gcc13 will appear again when OpenWrt default (and buildbot) moves to gcc14 and autoreconf gets again activated. (As buildbot builds all packages, the various configure scripts detect more features than usual private builds. That is the likely reason why it now seems to work for private builds, but not in buildbot.)

anomeome, 2025-03-22:
Still an issue extent on x86 build
```
make[6]: Entering directory '/home/kc/c7v2/source/build_dir/target-x86_64_musl/coreutils-9.6'
ccache x86_64-openwrt-linux-musl-gcc   -march=tremont -mtune=tremont -O3 -pipe -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/home/kc/c7v2/source/build_dir/target-x86_64_musl/coreutils-9.6=coreutils-9.6 -Wformat -Werror=format-security -DPIC -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -Wl,--as-needed -L/home/kc/c7v2/source/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/usr/lib -L/home/kc/c7v2/source/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/lib -fuse-ld=bfd -DPIC -fpic -specs=/home/kc/c7v2/source/include/hardened-ld-pie.specs -znow -zrelro  -o src/printf src/printf.o src/libver.a lib/libcoreutils.a   lib/libcoreutils.a  
/home/kc/c7v2/source/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/14.2.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: lib/libcoreutils.a(libcoreutils_a-unicodeio.o): in function `unicode_to_mb':
unicodeio.c:(.text+0x1e1): undefined reference to `u8_uctomb_aux'
/home/kc/c7v2/source/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/14.2.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: lib/libcoreutils.a(libcoreutils_a-iconv.o): in function `rpl_iconv':
iconv.c:(.text+0x43e): undefined reference to `u8_mbtoucr'
/home/kc/c7v2/source/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/14.2.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: iconv.c:(.text+0x565): undefined reference to `u8_uctomb_aux'
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:10530: src/printf] Error 1
make[6]: Leaving directory '/home/kc/c7v2/source/build_dir/target-x86_64_musl/coreutils-9.6'
make[5]: *** [Makefile:21714: all-recursive] Error 1
make[5]: Leaving directory '/home/kc/c7v2/source/build_dir/target-x86_64_musl/coreutils-9.6'
make[4]: *** [Makefile:7919: all] Error 2
make[4]: Leaving directory '/home/kc/c7v2/source/build_dir/target-x86_64_musl/coreutils-9.6'
make[3]: *** [Makefile:157: /home/kc/c7v2/source/build_dir/target-x86_64_musl/coreutils-9.6/.built] Error 2
make[3]: Leaving directory '/home/kc/c7v2/source/feeds/packages/utils/coreutils'
time: package/feeds/packages/coreutils/compile#0.36#0.07#0.42
    ERROR: package/feeds/packages/coreutils failed to build.
make[2]: *** [package/Makefile:189: package/feeds/packages/coreutils/compile] Error 1
make[2]: Leaving directory '/home/kc/c7v2/source'
make[1]: *** [package/Makefile:183: /home/kc/c7v2/source/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2

```

schm0, 2025-03-23:
> Still an issue extent on x86 build
>
> ```
> make[6]: Entering directory '/home/kc/c7v2/source/build_dir/target-x86_64_musl/coreutils-9.6'
> ccache x86_64-openwrt-linux-musl-gcc   -march=tremont [...]

Disabling stripping fixes it but it increases the final image quiet a bit...
Is it possible to disable stripping on per package basis?

hnyman, 2025-03-23:
> Still an issue extent on x86 build

Your error with gcc14 with autoreconf looks quite similar that buildbot got with gcc13 when autoreconf was enabled for gcc13.

> Disabling stripping fixes it

Interesting.  Might failing at stripping be related to LTO (link-time optimization)?

> Is it possible to disable stripping on per package basis?

To my knowledge, yes. 
Diverse examples below. One of those ways might help coreutils:

https://github.com/openwrt/packages/blob/2d66b6c8f2234dc486032d6c45836cc10537fdbd/lang/golang/golang/Makefile#L122-L124

https://github.com/openwrt/packages/blob/2d66b6c8f2234dc486032d6c45836cc10537fdbd/net/dnsdist/Makefile#L115-L118

https://github.com/openwrt/packages/blob/2d66b6c8f2234dc486032d6c45836cc10537fdbd/lang/python/micropython/Makefile#L74

https://github.com/openwrt/packages/blob/2d66b6c8f2234dc486032d6c45836cc10537fdbd/net/keepalived/Makefile#L202

schm0, 2025-03-23:
> > Still an issue extent on x86 build
> 
> Your error with gcc14 with autoreconf looks quite similar that buildbot got with gcc13 when autoreconf was enabled for gcc13.
> 
> > Disabling stripping fixes it
> 
> Interesting. Might failing at stripping be related to LTO (link-time optimization)?
> 
> > Is it possible to disable stripping on per package basis?
> 
> To my knowledge, yes. Diverse examples below. One of those ways might help coreutils:
> 
> [packages/lang/golang/golang/Makefile](https://github.com/openwrt/packages/blob/2d66b6c8f2234dc486032d6c45836cc10537fdbd/lang/golang/golang/Makefile#L122-L124)
> 
> Lines 122 to 124 in [2d66b6c](/openwrt/packages/commit/2d66b6c8f2234dc486032d6c45836cc10537fdbd)
>  # don't strip ELF executables in test data 
>  RSTRIP:=: 
>  STRIP:=: 
> 
> [packages/net/dnsdist/Makefile](https://github.com/openwrt/packages/blob/2d66b6c8f2234dc486032d6c45836cc10537fdbd/net/dnsdist/Makefile#L115-L118)
> 
> Lines 115 to 118 in [2d66b6c](/openwrt/packages/commit/2d66b6c8f2234dc486032d6c45836cc10537fdbd)
>  ifneq ($(CONFIG_DNSDIST_DEBUG_SYMBOLS),) 
>  	RSTRIP:=: 
>  	STRIP:=: 
>  endif 
> 
> [packages/lang/python/micropython/Makefile](https://github.com/openwrt/packages/blob/2d66b6c8f2234dc486032d6c45836cc10537fdbd/lang/python/micropython/Makefile#L74)
> 
> Line 74 in [2d66b6c](/openwrt/packages/commit/2d66b6c8f2234dc486032d6c45836cc10537fdbd)
>  MAKE_FLAGS += BUILD_VERBOSE=1 STRIP= 
> 
> [packages/net/keepalived/Makefile](https://github.com/openwrt/packages/blob/2d66b6c8f2234dc486032d6c45836cc10537fdbd/net/keepalived/Makefile#L202)
> 
> Line 202 in [2d66b6c](/openwrt/packages/commit/2d66b6c8f2234dc486032d6c45836cc10537fdbd)
>  MAKE_FLAGS += STRIP="/bin/true"

I can't get it to work.

Removing the autoreconf fixup and adding -std=c11 / -std=c17 seems to work though...

EdiTurn, 2025-03-24:
Still have errros with `CONFIG_NO_STRIP=y` and `CONFIG_USE_MOLD=y`. Compilation passes after disabling MOLD. 
```
mold: error: undefined symbol: u8_mbtoucr
>>> referenced by iconv.c
>>>               lib/libcoreutils.a(libcoreutils_a-iconv.o):(rpl_iconv)
mold: error: undefined symbol: u8_uctomb_aux
>>> referenced by unicodeio.c
>>>               lib/libcoreutils.a(libcoreutils_a-unicodeio.o):(unicode_to_mb)
>>> referenced by iconv.c
>>>               lib/libcoreutils.a(libcoreutils_a-iconv.o):(rpl_iconv)
```

ptpt52, 2025-04-30:
mark.

neheb, 2025-08-14:
@graysky2 this should be closed.