openwrt/packages #26180: xwrt sqm installation error

yousendsxx, 2025-03-20:
I'm trying to install luci-app-sqm module on xwrt, I'm getting this error, how do I solve it.

[image]

Information of the device :

[image]

brada4, 2025-03-20:
Try openwrt and report back if there is any problem.

yousendsxx, 2025-03-20:
openwrt does not work stably on my device, the internet is constantly cut off, I installed it because there is no problem with xwrt

brada4, 2025-03-20:
Your post kind of says opposite, either way only xwrt can help you with their build.

brada4, 2025-03-31:
Since you got the pointer to solution - no action need be taken here.
https://github.com/x-wrt/x-wrt/issues/447