openwrt/packages #26181: ddns-scripts: "/etc/init.d/ddns start" weird download stats showing (possible memory leak also)

AcidSlide, 2025-03-20:
Hi all,

I've been noticing this for more than a month with the latest `ddns-scripts` builds.  And it only usually happens after a reboot of the router. 

Steps to reproduce:

1. Reboot router (and wait for everything to start/run - i usually wait 5mins just to be sure)
2. Stop ddns with `/etc/init.d/ddns stop`
3. Start ddns with `/etc/init.d/ddns start`

Then the following happens...
```
# /etc/init.d/ddns start
root@PantherVision /tmp/usb/openwrt/scripts [#]#  060432  WARN : Service section disabled! - TERMINATE
 060432  WARN : Service section disabled! - TERMINATE
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1556    0  1384  100   172   1272    158  0:00:01  0:00:01 --:--:--  1431
100  1556    0  1384  100   172   1062    132  0:00:01  0:00:01 --:--:--  1304
  % T o t%a lT o t a l%   R e c%e iRveecde i%v eXdf e%r dX f eArvde r aAgvee rSapgeee dS p e eTdi m e  T i m eTime     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   353    0   181  100   172    152    144  0:00:01  0:00:01 --:--:--   297
100   349    0   177  100   172    148    144  0:00:01  0:00:01 --:--:--   293
  %   T%o tTaolt a l    %   R%e cReeicveeidv e%d  X%f eXrfde r dA v eArvaegrea gSep eSepde e d  T i mTei m e    T i mTei m e      T i mTei m eC u rCruernrte
n t
                                                              D l oDaldo a dU p lUopaldo a d  T o tTaolt a l  S p eSnpte n t    L e fLte f tS p eSepde
ed
0    0      0    0    0    0      0    0    0    0      0    0        0    0        0   -0- :----::----: ---- :----:100   255    0    20  100   235     14    166  0:00:01  0:00:01 --:--:--   180
100   251    0    20  100   231     14    162  0:00:01  0:00:01 --:--:--   177
```

Then after a few mins without doing anything while still in the command line (via ssh).. this happens..
```
root@openwrt /root [#]#   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1556    0  1384  100   172   1110    138  0:00:01  0:00:01 --:--:--  1372
100  1556    0  1384  100   172   1212    150  0:00:01  0:00:01 --:--:--  1364
    %%  TToottaall        %%  RReecceeiivveedd  %%  XXffeerrdd    AAvveerraaggee  SSppeeeedd      TTiimmee        TTiimmee          TTiimmee    CCuurrrreenntt

                                                          D l o a d     U pDllooaadd     UTpoltoaald      STpoetnatl       SLpeefntt    S p eLeedf
e e d0
     00         00         00         00       0    0        0  0          0  0- - : - - : -0-  ----::----::----  --100   353    0   181  100   172    162    154  0:00:01  0:00:01 --:--:--   316
100   349    0   177  100   172    152    147  0:00:01  0:00:01 --:--:--   300
```

I'm using PorkBun DDNS script via `ddns-scripts-porkbun`.  To be honest I'm not sure if the main ddns script is the issue or the porkbun ddns addon.

feckert, 2025-03-20:
I suspect this is the [api_call](https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/usr/lib/ddns/update_porkbun_v3.sh#L57) function of the ddns-scripts-porkbun script. This function is called several times.
https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/usr/lib/ddns/update_porkbun_v3.sh#L91
https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/usr/lib/ddns/update_porkbun_v3.sh#L106
https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/usr/lib/ddns/update_porkbun_v3.sh#L118
https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/usr/lib/ddns/update_porkbun_v3.sh#L137

Take a look at the other providers. What are they doing with the output?
Maybe we need to redirect the stderr or move it to /dev/null.

AcidSlide, 2025-03-20:
> Take a look at the other providers. What are they doing with the output?
> Maybe we need to redirect the stderr or move it to /dev/null.

Those where the first things I checked also but haven't had the time to do debugging on the calls.  I'll try to check how it's done by the other addon's.

Tagging original author @cahorn hopefully he can check it out also.

AcidSlide, 2025-03-22:
Hi @feckert, below is definitely part of the ddns script

On my test, i've got 2 out of 4 disabled DDNS, and the following shows and I need to press enter in the keyboard to show the prompt again.

```
# /etc/init.d/ddns stop && sleep 10 && /etc/init.d/ddns start
root@PantherVision /tmp/usb/openwrt/scripts [#]#  183307  WARN : Service section disabled! - TERMINATE
 183307  WARN : Service section disabled! - TERMINATE
```

Below is from syslog
```
Sat Mar 22 18:33:07 2025 user.warn ddns-scripts[32457]: backspace_acidslide_ipv6: Service section disabled! - TERMINATE
Sat Mar 22 18:33:07 2025 user.warn ddns-scripts[32455]: local_acidslide_ipv6: Service section disabled! - TERMINATE
Sat Mar 22 18:33:07 2025 user.warn ddns-scripts[32457]: backspace_acidslide_ipv6: PID '32457' exit WITH ERROR '1' at 2025-03-22 18:33
Sat Mar 22 18:33:07 2025 user.warn ddns-scripts[32455]: local_acidslide_ipv6: PID '32455' exit WITH ERROR '1' at 2025-03-22 18:33
```

AcidSlide, 2025-03-22:
And by the way @feckert and @cahorn , I fixed the issue with the download stats showing by fixing the `$CURL` call, below is the updated `api_call()` function

```
# Make Porkbun API call
# $1 - Porkbun API endpoint
# $2 - request JSON payload
function api_call() {
	local response url
	url="$__API/$1"
	write_log 7 "API endpoint URL: $url"
	write_log 7 "API request JSON payload: $2"
	response=$($CURL -s -X POST "$url" -H "Content-Type: application/json" --data "$2" -o $DATFILE 2>$ERRFILE)
	write_log 7 "API response JSON payload: $(cat $DATFILE)"
	echo "$(cat $DATFILE)"
}
```

Already tested it and works based on the logs.  Do you need me to create a PR for this?

closed 2025-04-20 by feckert