openwrt/packages #26194: aria2: not work in 24.10 X86_64

PantaFH, 2025-03-22:
Sat Mar 22 03:36:59 2025 daemon.err aria2c[2109]: Exception caught
Sat Mar 22 03:36:59 2025 daemon.info aria2c[2109]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:54 2025 daemon.info aria2c[2086]: jail: jail (2087) exited with exit: 1
Sat Mar 22 03:36:54 2025 daemon.err aria2c[2086]:
Sat Mar 22 03:36:54 2025 daemon.err aria2c[2086]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:54 2025 daemon.err aria2c[2086]: Exception caught
Sat Mar 22 03:36:54 2025 daemon.info aria2c[2086]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:49 2025 daemon.info aria2c[2015]: jail: jail (2075) exited with exit: 1
Sat Mar 22 03:36:49 2025 daemon.err aria2c[2015]:
Sat Mar 22 03:36:49 2025 daemon.err aria2c[2015]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:49 2025 daemon.err aria2c[2015]: Exception caught
Sat Mar 22 03:36:49 2025 daemon.info aria2c[2015]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:49 2025 daemon.info aria2: Please make sure user 'nobody' has write access to download dir: /mnt/nas/Download/Aria2
Sat Mar 22 03:36:49 2025 daemon.info aria2: Aria2 will run with user 'nobody'.
Sat Mar 22 03:36:18 2025 daemon.info procd: Instance aria2::aria2.main s in a crash loop 6 crashes, 0 seconds since last crash
Sat Mar 22 03:36:18 2025 daemon.info aria2c[1535]: jail: jail (1536) exited with exit: 1
Sat Mar 22 03:36:18 2025 daemon.err aria2c[1535]:
Sat Mar 22 03:36:18 2025 daemon.err aria2c[1535]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:18 2025 daemon.err aria2c[1535]: Exception caught
Sat Mar 22 03:36:18 2025 daemon.info aria2c[1535]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:13 2025 daemon.info aria2c[1523]: jail: jail (1524) exited with exit: 1
Sat Mar 22 03:36:13 2025 daemon.err aria2c[1523]:
Sat Mar 22 03:36:13 2025 daemon.err aria2c[1523]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:13 2025 daemon.err aria2c[1523]: Exception caught
Sat Mar 22 03:36:13 2025 daemon.info aria2c[1523]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:08 2025 daemon.info aria2c[1511]: jail: jail (1512) exited with exit: 1
Sat Mar 22 03:36:08 2025 daemon.err aria2c[1511]:
Sat Mar 22 03:36:08 2025 daemon.err aria2c[1511]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:08 2025 daemon.err aria2c[1511]: Exception caught
Sat Mar 22 03:36:08 2025 daemon.info aria2c[1511]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:03 2025 daemon.info aria2c[1499]: jail: jail (1500) exited with exit: 1

{
        "kernel": "6.6.73",
        "hostname": "EXWRT",
        "system": "Genuine Intel(R) CPU 0000 @ 2.40GHz",
        "model": "VMware, Inc. VMware20,1",
        "board_name": "vmware-inc-vmware20-1",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "x86/64",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}

PantaFH, 2025-03-22:
# Auto generated file, changes to this file will be lost.
dir=/mnt/nas/Download/Aria2
enable-rpc=true
rpc-allow-origin-all=true
rpc-listen-all=true
quiet=true
continue=true
input-file=/mnt/nas/data/aria2_data/aria2.session.main
save-session=/mnt/nas/data/aria2_data/aria2.session.main
log=/mnt/nas/data/aria2_data/aria.log
log-level=warn
rpc-secret=cmhkkue9k9f1f8yf45g674qr9bep9bel
check-certificate=false
enable-dht=true
dht-file-path=/mnt/nas/data/aria2_data/dht.dat.main
enable-dht6=true
dht-file-path6=/mnt/nas/data/aria2_data/dht6.dat.main
bt-enable-lpd=true
bt-save-metadata=true
disk-cache=512M
enable-peer-exchange=true
file-allocation=falloc
follow-torrent=true
max-concurrent-downloads=120
max-connection-per-server=32
max-overall-upload-limit=1
max-upload-limit=1
save-session-interval=30
split=32

PantaFH, 2025-03-22:
After much effort, I finally found a solution, and it worked, but I still don't know what happened.[https://github.com/aria2/aria2/issues/2152#issuecomment-1848164124](url)

amirhossein5592, 2025-07-07:
Please tell me how you fix it pleaseeeee

PantaFH, 2025-10-18:
OpenWrt 24.10
/etc/ini.d/aria2:
#!/bin/sh /etc/rc.common
# Copyright (C) 2016-2017 Hsing-wang Liao <kuoruan@gmail.com>
# Licensed to the public under the Apache License 2.0.

START=99
USE_PROCD=1

NAME=aria2
PROG=/usr/bin/aria2c

_info() {
	logger -p daemon.info -t "$NAME" "$*"
}

_err() {
	logger -p daemon.err -t "$NAME" "$*"
}

_make_dir() {
	local d
	for d in "$@"; do
		if [ ! -d "$d" ]; then
			mkdir -p "$d" 2>/dev/null || return 1
		fi
	done

	return 0
}

_create_file() {
	touch "$@" 2>/dev/null
}

_change_owner() {
	local u="$1"; shift

	local d
	for d in "$@"; do
		if [ -f "$d" ]; then
			chown "$u" "$d" 2>/dev/null || return 1
		elif [ -d "$d" ]; then
			chown -R "$u" "$d" 2>/dev/null || return 1
		fi
	done

	return 0
}

_change_file_mode() {
	local mod="$1"; shift
	chmod "$mod" "$@" 2>/dev/null
}

_reset_dir_mode() {
	local d
	for d in "$@"; do
		if [ -d "$d" ]; then
			find "$d" -type d -exec chmod 755 {} \; 2>/dev/null
			find "$d" -type f -exec chmod 644 {} \; 2>/dev/null
		fi
	done
}

append_options() {
	local o; local v
	for o in "$@"; do
		v="$(eval echo "\$$o")"
		[ -n "$v" ] && \
[... 265 lines cut ...]

	procd_set_param respawn
	procd_set_param stdout 1
	procd_set_param stderr 1

	procd_set_param file "$config_file"
	[ -n "$user" ] && \
		procd_set_param user "$user"

	procd_add_jail "$NAME.$section" log
	procd_add_jail_mount "/usr/lib/ossl-modules"
	procd_add_jail_mount "$ca_certificate" "$certificate" "$rpc_certificate" "$rpc_private_key"
	procd_add_jail_mount_rw "$dir" "$config_dir" "$log"
	procd_close_instance
}

service_triggers() {
	procd_add_reload_trigger "$NAME"
	procd_add_validation aria2_validate
}

start_service() {
	config_load "$NAME"
	config_foreach aria2_validate "aria2" aria2_start
}

closed 2025-10-18 by PantaFH