openwrt/packages #26228: adblock-fast: omitting allowed domains from config results in failure to process allow-list

alexmede, 2025-03-27:
Maintainer: @stangri 

Environment:
```
Model: GL.iNet GL-MT6000
Architecture: ARMv8 Processor rev 4
Target Platform: mediatek/filogic
Firmware Version: OpenWrt 24.10-SNAPSHOT r28565-de2718b2f3 / LuCI openwrt-24.10 branch 25.085.78111~e5381f5
Kernel Version: 6.6.83

adblock-fast - 1.1.3-r5
luci-app-adblock-fast - 1.1.2-r18
```

After upgrading from adblock-fast 1.1.1-5 to 1.1.3-r5 while retaining the same config, I received the service errors:
```
Failed to process allow-list!
Failed to process allow-list!
Errors encountered, please check the [README](https://docs.openwrt.melmac.net/adblock-fast/)!
```

Manual reloading of the adblock-fast service results in two failed steps:
```
Removing allowed domains from combined list[✗]
Allowing domains [✗]
```

The following is also printed excessively in the system log per blocked domain line:
```
daemon.err dnsmasq[1]: bad option at line n of /var/run/adblock-fast/dnsmasq.servers
```

This does not occur when at least one domain is listed as allowed in the config, for example:
```
list allowed_domain 'example.com'
```

Expected behavior is to be able to process all lists with no allowed domains.

stangri, 2025-04-01:
@alexmede thank you for reporting/diagnosing it. I'll try to look into this first or second weekend of April.

stangri, 2025-04-08:
@alexmede still on my radar, didn't have time to look into it properly.

closed 2025-04-09 by stangri