openwrt/packages #26231: python3-bcrypt missing in packages

TT-E1NFACH, 2025-03-29:
The package

    python3-bcrypt_4.0.1-1_mips_24kc

and

    python3-cryptography_41.0.7-1_mips_24kc

are not compiled correctly (https://downloads.openwrt.org/releases/faillogs-23.05/mipsel_24kc/packages/ , https://downloads.openwrt.org/releases/faillogs-24.10/mipsel_24kc/packages/python-bcrypt/ ) and so it is missing inside the packages. This counts for 23.05.x onwards to 24.10.x.

How to fix this? Cause i need python3-paramiko and these packages are dependencies.

PS:
Two weeks ago these packages were compiled correctly.

predators46, 2025-03-30:
you can update the version or downgrade and build it again

BKPepe, 2025-09-24:
I dont see any faillog anymore for OpenWrt 24.10 and as OpenWrt 23.05 is end of life, I am closing this.

closed 2025-09-24 by BKPepe