openwrt/packages #26232: lpac: MBIM devices not working with uqmi

presslab-us, 2025-03-29:
Maintainer: @blocktrron
Environment: ARM Cortex-A53, Banana Pi-R4, OpenWRT Snapshot 16th March 2025

Description:

### Problem
MBIM devices are not working with the UQMI backend.  There doesn't seem to be any way to add the `-m` command line parameter to `uqmi`.

I am using MBIM because it is 10x faster than QMI.  When I switch to QMI mode on my modem it is working but ideally I'd rather not have to go through that whole process when I switch eSIM.  Using something e.g. `uqmi -d /dev/cdc-wdm0 -m --uim-get-sim-state` I can see that `uqmi` is working with `-m`.

presslab-us, 2025-03-31:
I see that libmbim is natively supported by lpac now.  I have submitted a pull request for adding in support to compile this.

https://github.com/openwrt/packages/pull/26237

brada4, 2025-03-31:
mbkm and qmi are different ptotocols, all 5g modems support just one protocol at once. Others playnly "it does not work" . No bug, just ` fact of life.

closed 2025-04-03 by blocktrron