openwrt/packages #26255: [CVE-2024-50602, CVE-2024-8176] Please upgrade vulnerable Expat to 2.7.1

hartwork, 2025-04-02:
Hi! 👋 

OpenWrt seems to be at vulnerable Expat 2.6.3 in two places:
- https://github.com/openwrt/packages/blob/HEAD/libs/expat/Makefile
- https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=tools/expat/Makefile

That leaves vulnerabilities CVE-2024-50602 and CVE-2024-8176 unfixed. Please consider upgrading to Expat 2.7.1. Thank you! 🙏 

PS: If you are curious about the story behind CVE-2024-8176, blog post [Recursion kills: The story behind CVE-2024-8176 in libexpat](https://blog.hartwork.org/posts/expat-2-7-0-released/) could be of interest.

Best. Sebastian

GeorgeSapkin, 2025-04-04:
Looks like there are quite a few packages that depend on it directly, so it might not be trivial to test:

```bash
grep -r +libexpat .
./lang/luaexpat/Makefile:  DEPENDS:=+lua +libexpat
./lang/perl-xml-parser/Makefile:  DEPENDS:=perl +libexpat +perlbase-dynaloader +perlbase-essential
./libs/apr-util/Makefile:  DEPENDS:=+libapr +libexpat +libuuid $(ICONV_DEPENDS)
./libs/avahi/Makefile:  DEPENDS:=+libavahi-dbus-support +libexpat +librt +libdaemon
./libs/avahi/Makefile:  DEPENDS:=+libavahi-nodbus-support +libexpat +librt +libdaemon
./libs/libarchive/Makefile:  DEPENDS:=+zlib +liblzma +libbz2 +libexpat
./libs/libnpupnp/Makefile:  DEPENDS:=+libstdcpp +libexpat +libcurl +libmicrohttpd
./libs/neon/Makefile:  DEPENDS:=+libopenssl +libexpat +zlib
./libs/vips/Makefile:  DEPENDS:=+glib2 +libexif +libexpat +libjpeg +libpng +giflib +libxml2
./net/lftp/Makefile:  DEPENDS:=+libncurses +libopenssl +libreadline +libstdcpp +libexpat +zlib
./net/unbound/Makefile:  DEPENDS+= +unbound-daemon +libexpat
./sound/mpd/Makefile:            +AUDIO_SUPPORT:alsa-lib +libexpat +libflac +libid3tag +libfaad2 +libopus
./sound/owntone/Makefile:DEPENDS:=+libgpg-error +libgcrypt +libgdbm +zlib +libexpat +libunistring \
./utils/dbus/Makefile: DEPENDS:= +libexpat +libdbus
./utils/fontconfig/Makefile:  DEPENDS:=+libpthread +libexpat +libfreetype
./utils/mstflint/Makefile:    +libcurl +libexpat +liblzma +libopenssl \
```

Are there any breaking changes since 2.6.3?

hartwork, 2025-04-04:
> Looks like there are quite a few packages that depend on it directly, so it might not be trivial to test:
> ```console
> # grep -r +libexpat .
> ./lang/luaexpat/Makefile:  DEPENDS:=+lua +libexpat
> ./lang/perl-xml-parser/Makefile:  DEPENDS:=perl +libexpat +perlbase-dynaloader +perlbase-essential
> ./libs/apr-util/Makefile:  DEPENDS:=+libapr +libexpat +libuuid $(ICONV_DEPENDS)
> ./libs/avahi/Makefile:  DEPENDS:=+libavahi-dbus-support +libexpat +librt +libdaemon
> ./libs/avahi/Makefile:  DEPENDS:=+libavahi-nodbus-support +libexpat +librt +libdaemon
> ./libs/libarchive/Makefile:  DEPENDS:=+zlib +liblzma +libbz2 +libexpat
> ./libs/libnpupnp/Makefile:  DEPENDS:=+libstdcpp +libexpat +libcurl +libmicrohttpd
> ./libs/neon/Makefile:  DEPENDS:=+libopenssl +libexpat +zlib
> ./libs/vips/Makefile:  DEPENDS:=+glib2 +libexif +libexpat +libjpeg +libpng +giflib +libxml2
> ./net/lftp/Makefile:  DEPENDS:=+libncurses +libopenssl +libreadline +libstdcpp +libexpat +zlib
> ./net/unbound/Makefile:  DEPENDS+= +unbound-daemon +libexpat
> ./sound/mpd/Makefile:            +AUDIO_SUPPORT:alsa-lib +libexpat +libflac +libid3tag +libfaad2 +libopus
> ./sound/owntone/Makefile:DEPENDS:=+libgpg-error +libgcrypt +libgdbm +zlib +libexpat +libunistring \
> ./utils/dbus/Makefile: DEPENDS:= +libexpat +libdbus
> ./utils/fontconfig/Makefile:  DEPENDS:=+libpthread +libexpat +libfreetype
> ./utils/mstflint/Makefile:    +libcurl +libexpat +liblzma +libopenssl \
> ```

@GeorgeSapkin that's a short list in my world, my own list of know reverse dependencies is public at https://libexpat.github.io/doc/users/ .

> Are there any breaking changes since 2.6.3?

No, but [Hyrum's Law](https://www.hyrumslaw.com/) applies. There was an [arguable regression in 2.7.0](https://github.com/libexpat/libexpat/issues/980#issuecomment-2741172588) but we are talking 2.7.1 here with a related fix included.

There is a long list of distros that have updated to 2.7.1 by now at https://repology.org/project/expat/history so you'd be in good company.

GeorgeSapkin, 2025-04-04:
I'll build a few packages that I can run, see if something falls apart and make a PR.

closed 2025-04-06 by openwrt-bot