openwrt/packages #26263: [security] xmlrpc-c variant "internal" uses a (very old and) vulnerable bundle of libexpat

hartwork, 2025-04-04:
Hi! 👋 

xmlrpc-c variant "internal" seems to use a (very old and) vulnerable bundle of libexpat at… https://github.com/openwrt/packages/blob/74a66189f123bd1bca3171612327c5863761af26/libs/xmlrpc-c/Makefile#L42-L48 

I'm in touch with upstream on the topic, things are still moving. If disabling variant "internal" in OpenWrt is an option for a temporary fix, that's what I would recommend for the moment, personally.

Best

Sebastian

---

Related: https://bugs.gentoo.org/952113

hartwork, 2025-04-06:
Update: I have created and applied patch [`xmlrpc-c-1.59.03-use-system-expat.patch`](https://raw.githubusercontent.com/gentoo/gentoo/61b6130343a41b49da1ffe7376ab5d2077a37411/dev-libs/xmlrpc-c/files/xmlrpc-c-1.59.03-use-system-expat.patch) in Gentoo now that makes the code use system Expat rather than the vulnerable bundle to a point where `rm -R lib/expat/` in packaging becomes possible. The patch happens to target the very same version of xmlrpc-c that you have in OpenWrt, so applying that patch could be an alternative to disabling variant "internal" in OpenWrt.

PS: The related issue in Gentoo is https://bugs.gentoo.org/952113 .

brada4, 2025-04-06:
@thess @neheb yep, it looks yucky, there is no system-expat flavour ( which should be about same size as embedded expat minus vulnerability plus reusable)

thess, 2025-09-27:
ATM nothing seems to require xmlrpc-c-internal. I would go along with @hartwork and remove the `internal` variant. If no objections, I will push the removal.

PR #27553 (2025-09-30): xmlrpc-c: [Security] Remove obsolete "-internal" variant
Closes #26263

Remove myself as maintainer
commit c48ddeb5 (2025-09-30): xmlrpc-c: [Security] Remove obsolete "-internal" variant

Closes #26263
Remove myself as maintainer
closed 2025-09-30 by thess