openwrt/packages #26272: ipmitool 1.8.19: Change in handling Enterprise Numbers breaks reproducibility, offline builds, and increases size

osedlbauer, 2025-04-07:
With ipmitool version 1.8.19, the logic for handling IANA Enterprise Numbers (PEN) has changed. Previously, the numbers were compiled into the binary. However, upstream decided to fetch the list during the build process ([introduced here](https://github.com/ipmitool/ipmitool/commit/9d41136c9b7c7d392f1a3f3adeb6d7fe3bd3135e)).

This introduces several issues:

1. Non-reproducible builds: The downloaded list may change
2. Requires internet connection: Offline builds are no longer possible
3. Large file size: The downloaded list is around 4.5 MB, which is problematic for devices with limited storage

To solve this, I have prepared two PRs. One of them backports upstream commits that allow building ipmitool without downloading the enterprise numbers (#26270).

To still support vendor identification, I propose creating a separate package that provides a static copy of the enterprise numbers (#26271). This way, users can opt-in to install it if needed. This approach is similar to what we already do with hwdata (for lsusb, lspci, etc.).

But I believe it would make more sense to host it under the official [OpenWrt GitHub organization](https://github.com/openwrt), if accepted.

Would appreciate feedback.

osedlbauer, 2025-04-15:
Thank you @lynxis  for the review and @robimarko for merging my PRs.

I'm not sure about the hosting of the new package, as it's currently running under my organisation's repository.
Would it make sense to move it to the OpenWrt GitHub organisation?

closed 2025-05-21 by feckert

lynxis, 2025-05-26:
@robimarko having reproducible build is a good idea. I would prefer to have it in our repository.
I would be fine with having it in the openwrt space to have reproducilbe builds if the upstream repository doesn't want to keep it this way. Also offline builds are quite important.

@osedlbauer could you otherwise ask the upstream project to have an alternative to a live download?