openwrt/packages #26273: acme package: run-acme suggest using --force but does not support --force

bernd-wechner, 2025-04-07:
If I run `/usr/lib/acme/run-acme` and the Next renewal time is in the future, it suggests:

```
Add '--force' to force to renew.
```

but 

```
/usr/lib/acme/run-acme --force
```
just exits.

I traced it (note I add one instrumentation line at top) which improves the output:

```
# bash -x /usr/lib/acme/run-acme --force
+ PS4='+ ${BASH_SOURCE}:${LINENO}:${FUNCNAME[0]}: '
+ /usr/lib/acme/run-acme:12:: CHECK_CRON=--force
+ /usr/lib/acme/run-acme:13:: ACME=/usr/lib/acme/acme.sh
+ /usr/lib/acme/run-acme:14:: export CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt
+ /usr/lib/acme/run-acme:14:: CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt
+ /usr/lib/acme/run-acme:15:: export NO_TIMESTAMP=1
+ /usr/lib/acme/run-acme:15:: NO_TIMESTAMP=1
+ /usr/lib/acme/run-acme:17:: UHTTPD_LISTEN_HTTP=
+ /usr/lib/acme/run-acme:18:: STATE_DIR=/etc/acme
+ /usr/lib/acme/run-acme:19:: ACCOUNT_EMAIL=
+ /usr/lib/acme/run-acme:20:: DEBUG=0
+ /usr/lib/acme/run-acme:21:: NGINX_WEBSERVER=0
+ /usr/lib/acme/run-acme:22:: UPDATE_NGINX=0
+ /usr/lib/acme/run-acme:23:: UPDATE_UHTTPD=0
+ /usr/lib/acme/run-acme:24:: UPDATE_HAPROXY=0
+ /usr/lib/acme/run-acme:25:: NFT_HANDLE=
+ /usr/lib/acme/run-acme:26:: USER_CLEANUP=
+ /usr/lib/acme/run-acme:28:: . /lib/functions.sh
++ /lib/functions.sh:12:: N='
'
++ /lib/functions.sh:14:: _C=0
++ /lib/functions.sh:15:: NO_EXPORT=1
++ /lib/functions.sh:16:: LOAD_STATE=1
++ /lib/functions.sh:17:: LIST_SEP=' '
++ /lib/functions.sh:62:: reset_cb
++ /lib/functions.sh:438:: '[' -z '' ']'
++ /lib/functions.sh:438:: '[' -f /lib/config/uci.sh ']'
++ /lib/functions.sh:438:: . /lib/config/uci.sh
+++ /lib/config/uci.sh:20:: CONFIG_APPEND=
+ /usr/lib/acme/run-acme:374:: check_cron
+ /usr/lib/acme/run-acme:31:check_cron: '[' -f /etc/crontabs/root ']'
+ /usr/lib/acme/run-acme:31:check_cron: grep -q /etc/init.d/acme /etc/crontabs/root
+ /usr/lib/acme/run-acme:31:check_cron: return
+ /usr/lib/acme/run-acme:375:: '[' -n --force ']'
+ /usr/lib/acme/run-acme:375:: exit 0
```

which confirms this line:

https://github.com/openwrt/packages/blob/542930a91969394c6f3747a3c16f6592df58e933/net/acme/files/run.sh#L374C1-L374C31

is the last one because of this line:

https://github.com/openwrt/packages/blob/542930a91969394c6f3747a3c16f6592df58e933/net/acme/files/run.sh#L11

KsmoinO, 2025-05-10:
For me the force option suggested is a parameter for acme.sh script

BKPepe, 2025-05-20:
Hmm, another issue or feature request for acme. :) @stokito