openwrt/packages #26276: Toruble with `openvpn-mbedtls`

A1EF, 2025-04-07:
I use OpenVPN client on my router and had strange error with `openvpn-mbedtls`:

```
TLS_ERROR: read tls_read_plaintext error: SSL - Attempt to verify a certificate without an expected hostname. This is usually insecure.  In TLS clients, when a client authenticates a server through its certificate, the client normally checks three things: - the certificate chain must be valid; -
```

It strange because it was works perfect some time ago. I can manage this error only when I moved to `openvpn-openssl`. May be anyone know what's wrong with `openvpn-mbedtls`?

marjancinober, 2025-04-15:
Same error here. 

Here "OpenWrt 24.10._0_" "openvpn-mbedtls - 2.6.13-r1" server:
- works with  "OpenWrt 23.05.4" "openvpn-mbedtls - 2.5.8-5" clients ,   while it
- refuses to work with "OpenWrt 24.10._1_" "openvpn-mbedtls - 2.6.13-r1" client

The culprit might be the two lines in the log above the bad certificate claim, failed UDP read from openwrt server daemon? Or all permanently Unroutable control packets?

`Tue Apr 15 10:01:30 2025 daemon.err openvpn(homert)[18324]: TLS Error: Unroutable control packet received from [AF_INET]25.255.254.253:1194 (si=3 op=P_ACK_V1)`
`Tue Apr 15 10:01:34 2025 daemon.err openvpn(homert)[18324]: read UDPv4 [ECONNREFUSED]: Connection refused (fd=6,code=111)`
`Tue Apr 15 10:01:43 2025 daemon.err openvpn(homert)[18324]: TLS_ERROR: read tls_read_plaintext error: SSL - Attempt to verify a certificate without an expected hostname. This is usually insecure.  In TLS clients, when a client authenticates a server through its certificate, the client normally checks three things: - the certificate chain must be valid; -`

Firewall and UDP connectivity checked with `netcat -u` and works as expected + openvpn-openssl works in the same setup.

See: [log-openvpn-mbedtls.ERR.log](https://github.com/user-attachments/files/19756403/log-openvpn-mbedtls.ERR.log)

KsmoinO, 2025-04-15:
It also work on openwrt 24.10.0 with
```
OpenVPN 2.6.13 mipsel-openwrt-linux-gnu [SSL (mbed TLS)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
library versions: mbed TLS 3.6.2
```
More logs on openwrt 24.10.1
```
TLS_ERROR: read tls_read_plaintext error: SSL - Attempt to verify a certificate without an expected hostname. This is usually insecure.  In TLS clients, when a client authenticates a server through its certificate, the client normally checks three things: - the certificate chain must be valid; -
TLS Error: TLS object -> incoming plaintext read error
TLS Error: TLS handshake failed
Fatal TLS error (check_tls_errors_co), restarting

OpenVPN 2.6.13 mipsel-openwrt-linux-gnu [SSL (mbed TLS)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
library versions: mbed TLS 3.6.3
```

A1EF, 2025-04-15:
I suspect it cause due this Mbed-TLS [update](https://github.com/Mbed-TLS/mbedtls/releases/tag/mbedtls-3.6.3)
>In TLS clients, if mbedtls_ssl_set_hostname() has not been called,
mbedtls_ssl_handshake() now fails with
MBEDTLS_ERR_SSL_CERTIFICATE_VERIFICATION_WITHOUT_HOSTNAME
if certificate-based authentication of the server is attempted.
This is because authenticating a server without knowing what name
to expect is usually insecure. To restore the old behavior, either
call mbedtls_ssl_set_hostname() with NULL as the hostname, or
enable the new compile-time optio

marjancinober, 2025-04-16:
> It seems the issue has been [fixed](https://github.com/OpenVPN/openvpn/commit/4913628022f5d9d33bac656082a2ec279d8d1222) in `2.6.14` version.

This fix looks makes sense. Two questions come to mind:
1. Anybody knows when and how can we get OpenVPN 2.6.14 implemented on OpenWrt 24.10.x release? 
1. How to cross compile single package compatibly with the current public OpenWrt kernel binaries and other published OpenWrt packages - avoid own "unstable" firmware/package distribution website and keep using the public package distribution website?

KsmoinO, 2025-04-16:
[Makefile](https://github.com/openwrt/packages/blob/master/net/openvpn/Makefile) has already been updated. A new release for 24.10.1 should be available later.
If you want to cross compile it your self you can follow [this guide](https://openwrt.org/docs/guide-developer/toolchain/crosscompile).
As workaround you can use openvpn-openssl or openvpn-wolfssl for now.

KsmoinO, 2025-04-16:
OpenVPN 2.6.14 update has been merged 4 hours ago : https://github.com/openwrt/packages/pull/26261 it should be available in a few days

marjancinober, 2025-04-17:
> OpenVPN 2.6.14 update has been merged 4 hours ago : [#26261](https://github.com/openwrt/packages/pull/26261) it should be available in a few days

Great! 
Will that be in 24.10.1 I already use, or a new 24.10.2 - reflash needed (remote here)?

Temporarily using openwrt-openssl version, missing minimal *-mbedtls for weaker hardware.
Thanks for help guys.

KsmoinO, 2025-04-17:
It has already been published for some device. Example : https://downloads.openwrt.org/releases/packages-24.10/mips_24kc/packages/
The update is not yet available for mine but it should be published soon.

A1EF, 2025-04-17:
I get new version and problem has gone . So I close the issue.

closed 2025-04-17 by A1EF