openwrt/packages #26278: acme package: run-acme support whole run pre and post hooks

bernd-wechner, 2025-04-08:
The acme package supports specification of: `option user_cleanup` under any given cert config section. That is a post hook for each cert in `etc/config/acme`

It would be very useful if there were pre hooks as well, and pre and post hooks in the global context under `config acme`. 

I for example want to do something after a complete `run-acme` run after it's done everything it needs to do, not once after each cert. Specifically I need to it only if any certs were reissues, but hey I don't mind it's unconditional too (my needs involve copying the certs to other servers on the LAN which I do currently in a user_cleanup script but am forced to it after each cert  - a minor inefficiency but logical bugbear as I only need to do it after all certs are renewed).

Also, just s small vote for completeness (pre and post hooks, cert and global scope) for easy customisation.

leleb, 2025-04-13:
Yes, pre-hook would be nice because I need to disable geoip before issuing certificate (letsencrypt wants to "see" challenge")

KsmoinO, 2025-05-10:
hooks can be used from /etc/hotplug.d/acme/
Example:
/etc/hotplug.d/acme/00-uhttpd
```
case "$ACTION" in
  prepare)
    [ -d /www/.well-known ] && rm -r /www/.well-known
    [ $(readlink -f /www/.well-known) != "/var/run/acme/challenge/.well-known" ] && ln -sf /var/run/acme/challenge/.well-known /www/.well-known
    ;;
  issued|renewed)
    /etc/init.d/uhttpd reload
    ;;
  *)
    echo Unknown action $ACTION
    ;;
esac
```

KsmoinO, 2025-05-10:
Maybe not applicable because I do not have run-acme script
I use acme 4.0.0, acme-acmesh 3.0.7-r2 and acme-common 1.4.0

BKPepe, 2025-05-20:
Sorry @tohojo and @stokito. This issue is also related to acme packages.