openwrt/packages #26279: acme package: run-acme support lighttpd

bernd-wechner, 2025-04-08:
Currently, `run-acme` supports uhttpd and nginx, but lighttpd is what I use and belongs, rightly, in this set. It is a head on totally FOSS competitor with the freemium nginx. lighttpd needs to restarted after certs are updated, using `/etc/init.d/lighttpd restart` That is all. NO other configs need be supported as they are presumptive (as I find the uhttpd and nginx ones implemented - given the actual configuration of the server is more complex and I for example serve different certs for different domains - only because I use dns challenges and some not all domains are with the same registrar)

BKPepe, 2025-05-20:
I think, I should bring attention to @stokito as well. :)

bernd-wechner, 2025-05-21:
> > lighttpd 1.4.78 ([#26202](https://github.com/openwrt/packages/pull/26202)) includes a new feature (option disabled by default) to check and reload certificates and CRLs about once a minute, without needing to restart lighttpd.
> > https://www.lighttpd.net/2025/3/22/1.4.78/
> > https://wiki.lighttpd.net/Docs_SSL#Configuration

Interesting. Here's the setting:

> "ssl.refresh-certs" => "enable" periodically (every 64 secs) check for and reload updated files: TLS certificate and private key (default: disable)

It isn't clear on what check means. I imagine it's checking the time stamp on the cert file, low cost on 64s cycle.

64 is a puzzle, almost suggestive that it's intentionally out of sync with some other 60s cycles. Doubly interesting because it's not configurable and so the designer clearly thought this was a sweet frequency with no benefit to normal mugs like me modifying it ;-).

This issue remains valid. I see it as a call for recognising lighttpd's status in the game. I see three major players, of which two are generally on everyone's lips and supported:

1. Apache: old, bloated, literally named as "a patchy" server in deference to its evolutionary path. Exploded into popularity because it worked and was free.
2. Nginx: Redesigned from the ground up to be light on resources and fast and scalable. Freemium model. Really popular.
3. Lighttpd: Redesigned from the ground up to be light on resources and fast and scalable. FOSS model. 

Correct nay misconceptions there, because that's the story I run currently (and I wouldn't want it being wrong, but the story is flexible, reality less so ;-).

For that reason, I routinely champion it and request support for it.  Had to write this once:

https://github.com/bernd-wechner/django-lighttpd-middleware

Because neither lighttpd nor Django wanted to budge on the matter at the time (classic, not my problem from both sides -at the time).

gstrauss, 2025-05-21:
@stokito: incorrect (unless enabling the optional feature: `"ssl.refresh-certs" => "enable"`) and incorrect in your latest post, too:
#24217 lighttpd: trigger reload on ACME cert renewal (submitted by @stokito)
discussed a change to lighttpd.init which was later added to openwrt with the lighttpd1.4.77 package update in openwrt.

---

@bernd-wechner 
> Doubly interesting because it's not configurable and so the designer clearly thought this was a sweet frequency with no benefit to normal mugs like me modifying it ;-).

Maybe try describing your use case to the developers on the lighttpd support forum?
https://redmine.lighttpd.net/projects/lighttpd/boards/2

lighttpd runs on many platforms.  I could have used inotify on Linux instead of checking modification timestamps every 64 seconds, but that would add more code, and more code specific to Linux.  64 is a power of 2 and a quick bitmask CPU instruction on the cached time in seconds to determine whether or not to run periodic maintenance jobs inside lighttpd for each module.  Without further real-world input, 64 sec seemed to me like a reasonable initial value.  If you think it should be configurable, please describe why -- in the lighttpd forums, not here -- and include a reason why the extra knob is not just "because I can" but because there is a demonstrable need.

Regarding https://github.com/bernd-wechner/django-lighttpd-middleware there is https://www.rfc-editor.org/rfc/rfc3875.html.  Changing the way lighttpd handles SCRIPT_NAME and PATH_INFO might break existing applications, but you're welcome to change it for your application.  [lighttpd mod_magnet](https://wiki.lighttpd.net/mod_magnet) is one of the simpler mechanisms to manipulate these values once someone learns a little bit of Lua, or follows the examples on https://wiki.lighttpd.net/ModMagnetExamples