openwrt/packages #26280: hostapd: Incomplete implementation of the log level (semi-bug)

lars18th, 2025-04-08:
Hi,

In the current “hostapd” package there is a patch to adapt the tool's logging to Openwrt:
https://github.com/openwrt/openwrt/blob/main/package/network/services/hostapd/patches/410-limit_debug_messages.patch

But this patch is **incomplete**!

Here the problem:
- The current implementation sends _all_ tool messages to the syslog. This is true regardless of the logging level configured by the `logger_syslog_level` value.
- Therefore you will see a lot of System Log messages starting with `daemon.notice hostapd` (note the "notice" label) regardless the `logger_syslog_level=4` by default in the wireless configuration.
- Example of verbose messages (using Dawn):
```
Tue Apr  8 17:44:11 2025 daemon.notice hostapd: phy1-wl0: BEACON-RESP-RX xx:xx:xx:xx:xx:xx 164 04
Tue Apr  8 17:44:11 2025 daemon.notice hostapd: phy1-wl0: BEACON-REQ-TX-STATUS xx:xx:xx:xx:xx:xx 165 ack=1
Tue Apr  8 17:44:11 2025 daemon.notice hostapd: phy1-wl0: BEACON-RESP-RX xx:xx:xx:xx:xx:xx 0 00 [...]
Tue Apr  8 17:44:23 2025 daemon.notice hostapd: phy1-wl0: BEACON-REQ-TX-STATUS xx:xx:xx:xx:xx:xx 167 ack=1
Tue Apr  8 17:44:23 2025 daemon.notice hostapd: phy1-wl0: BEACON-RESP-RX xx:xx:xx:xx:xx:xx 0 00
Tue Apr  8 17:44:24 2025 daemon.notice hostapd: phy1-wl0: BEACON-REQ-TX-STATUS [02:fc:c5:24:a0:86](xx:xx:xx:xx:xx:xx) 166 ack=1
Tue Apr  8 17:44:24 2025 daemon.notice hostapd: phy1-wl0: BEACON-RESP-RX [02:fc:c5:24:a0:86](xx:xx:xx:xx:xx:xx) 166 00 [...]
Tue Apr  8 17:44:26 2025 daemon.notice hostapd: phy1-wl0: BEACON-REQ-TX-STATUS xx:xx:xx:xx:xx:xx 169 ack=1
Tue Apr  8 17:44:26 2025 daemon.notice hostapd: phy1-wl0: BEACON-RESP-RX xx:xx:xx:xx:xx:xx 0 00
```

More or less you'll see about 10~50 lines per minute. A lot of verbose logging!

---

The root cause is that the hostapd tool is sending the log using the `vsyslog()` standard C function, and passing to it the debug level:

https://github.com/Kamillaova/hostap/blob/298592db0eed2b293555a3abf040e5f5e8815b6f/src/utils/wpa_debug.c#L223

But the current implementation of Openwrt using `logd` does not internally filter the system log messages. Therefore, using regular mesh tools you're populating your System Log with hundreds of unnecessary entries.

One solution is to expand the current `410-limit_debug_messages.patch` to filter out the messages inside the functions `_wpa_printf()` and `_wpa_msg()`. Using the value of the `hapd->conf->logger_syslog_level` variable.

I hope someone will want to implement it and complete this patch. Without it so many users will continue to suffer from excessive `hostapd BEACON-*` messages in the System Log.

lars18th, 2025-04-22:
Hi @dangowrt ,

https://github.com/openwrt/packages/blob/d8272975468374232b72e85c70cf79c3699aca8b/net/hs20/Makefile#L18

I feel you're the mantainer of the `hostapd` package. So, please check this info. Thank you.

dwyeraidan, 2025-06-26:
My syslog is all about `BEACON-RESP-RX` these days.  Hard to troubleshoot anything else..

Would also love to see this patch completed.

lars18th, 2025-06-26:
Hi @dwyeraidan ,

This problem has been around for a long time. And nobody wants to fix it. And as I already said it's necessary to _add_ some code to hide these useless messages that make the LOG unusable.

I hope that some maintainer wants to solve this issue. 🙏

dwyeraidan, 2025-08-14:
> > Hi [@dwyeraidan](https://github.com/dwyeraidan) ,
> > 
> > This problem has been around for a long time. And nobody wants to fix it. And as I already said it's necessary to _add_ some code to hide these useless messages that make the LOG unusable.
> > 
> > I hope that some maintainer wants to solve this issue. 🙏
> 
> Well, why are you waiting for a fix which isn't coming anytime soon?
> 
> In the meantime, set the log_level to "ERROR" for the _**hostapd.phy0-ap0**_ and **_hostapd.phy1-ap0_** interfaces, the problem is solved. Once you do that, useless messages to syslog are gone!

I tried this too, but it had no effect.

dwyeraidan, 2025-08-14:
> What?  Where did you set the log_level?  I set it on mine and no longer see "BEACON-*" messages.

It was set in /etc/config/wireless for each interface. 
Log level 1 is error. 

uci set wireless.radio0.log_level='1'
uci set wireless.radio1.log_level='1'
uci commit wireless
wifi reload

dwyeraidan, 2025-08-14:
Thank you I will give this a try 👍🏻

RainGater, 2025-08-15:
> You could add this to `/etc/hotplug.d/net/` it will trigger for every wireless interfaces so don't need to specify individual interface anymore.
> ```
> [ "${DEVTYPE}" = "wlan" ] && {
>   [ "${ACTION}" = "add" ] && {
>     logger -t hostapd-loglevel "set ${INTERFACE} to error"
>     hostapd_cli -i "${INTERFACE}" log_level ERROR
>   }
> }
> ```

After adding your code, I rebooted the router and the script didn't execute.  Only on rc.local with a 10 second sleep works consistently.

agungrbudiman, 2025-08-15:
@RainGater try debugging with `logger -t hotplug $(env)` then toggle the wireless interface on/off to see which env variable available on your device. This will appear on the syslog:
`user.notice hotplug: ACTION=add HOTPLUG_TYPE=net DEVPATH=/.../phy0-ap1 DEVICENAME=phy0-ap1 SUBSYSTEM=net PATH= INTERFACE=phy0-ap1 PWD=/ DEVTYPE=wlan`

Mosney, 2025-10-23:
> It seems that even manually setting to ERROR level through hostapd_cli doesn’t prevent the log spam.

Update: looks ERROR-LEVEL less 3 `notice log` entries than INFO-LEVEL when I dis&reconnect my Android phone.

`<NOSHOW IN ERROR LEVEL>` Thu Oct 23 21:52:27 2025 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED [MAC]
Thu Oct 23 21:52:30 2025 daemon.info hostapd: phy1-ap0: STA [MAC] IEEE 802.11: authenticated
Thu Oct 23 21:52:30 2025 daemon.info hostapd: phy1-ap0: STA [MAC] IEEE 802.11: associated (aid 2)
`<NOSHOW IN ERROR LEVEL>` Thu Oct 23 21:52:30 2025 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED [MAC] auth_alg=open
Thu Oct 23 21:52:30 2025 daemon.info hostapd: phy1-ap0: STA [MAC] RADIUS: starting accounting session XXXXXX
Thu Oct 23 21:52:30 2025 daemon.info hostapd: phy1-ap0: STA [MAC] WPA: pairwise key handshake completed (RSN)
`<NOSHOW IN ERROR LEVEL>` Thu Oct 23 21:52:30 2025 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED [MAC]

lars18th, 2025-10-23:
Hi @Mosney , @RainGater and @agungrbudiman ,

> It seems that even manually setting to ERROR level through hostapd_cli doesn’t prevent the log spam.

As I've comment in the _first_ post that opens this issue...

- ### The root cause is that the hostapd tool is sending the log using the vsyslog() standard C function, and passing to it the debug level...

Therefore, any changes related to `log_level` cannot solve the problem!

It's _necessary_ to change the source code. And the way to do it it's to rewrite the ancient and incomplete: `410-limit_debug_messages.patch`.

Please, re-read the first post of the issue. And thank your for your interest. 😉

lars18th, 2025-12-30:
Hi to all!

I found the exact point where we need to change the source code so that the LOG is not filled with messages `BEACON-RESP-RX` & `BEACON-REQ-TX-STATUS` .

Here a simple patch:
```
diff --git a/src/ap/rrm.c b/src/ap/rrm.c
index 73771e4c5..8e64c581e 100644
--- a/src/ap/rrm.c
+++ b/src/ap/rrm.c
@@ -87,7 +87,7 @@ static void hostapd_handle_beacon_report(struct hostapd_data *hapd,
        report[0] = '\0';
        if (wpa_snprintf_hex(report, sizeof(report), pos, len) < 0)
                return;
-       wpa_msg(hapd->msg_ctx, MSG_INFO, BEACON_RESP_RX MACSTR " %u %02x %s",
+       wpa_msg(hapd->msg_ctx, MSG_DEBUG, BEACON_RESP_RX MACSTR " %u %02x %s",
                MAC2STR(addr), token, rep_mode, report);
 }

@@ -720,7 +720,7 @@ void hostapd_rrm_beacon_req_tx_status(struct hostapd_data *hapd,
 {
        if (len < 24 + 3)
                return;
-       wpa_msg(hapd->msg_ctx, MSG_INFO, BEACON_REQ_TX_STATUS MACSTR
+       wpa_msg(hapd->msg_ctx, MSG_DEBUG, BEACON_REQ_TX_STATUS MACSTR
                " %u ack=%d", MAC2STR(mgmt->da),
                mgmt->u.action.u.rrm.dialog_token, ok);
 }
```

What is doing this patch:

- In the original source code of the hostapd tool these two messages are in the _notice_ (aka INFO) level. However, when using Beacon Reporting many of these messages appear every few seconds. Therefore, it has sense to move them to the _debug_ level.
- The use of the command `hostapd_cli -i phy0-ap0 log_level ERROR` really works to remove these messages from the LOG. But a side effect is that you're already hidding other useful messages like `AP-STA-DISCONNECTED`, `AP-STA-DISCONNECTED` or `EAPOL-4WAY-HS-COMPLETED` that you need to discover troubles in your network.

So, I suggest to create a new one patch with the name `411-hide_beacon_messages.patch`.

Anyone can do it and merge with the `hostapd` package?

lars18th, 2025-12-30:
> [@lars18th](https://github.com/lars18th) you can do that as well.

Thank you @GeorgeSapkin . However the current signature enforcement generates a lot of troubles with the main OpenWRT branch. So I prefer if some other developer creates the PR.

lars18th, 2025-12-30:
In the meantime I recommend to add this file...

> `/etc/hotplug.d/net/99-hostapd-loglevel`
```shell
#!/bin/sh

if [ "$DEVTYPE" = wlan ]; then
        if [ "$ACTION" = add ]; then
                logger -t hostapd-loglevel "set ${INTERFACE} to WARNING"
                hostapd_cli -i "${INTERFACE}" log_level WARNING
        fi
fi
```

Perhaps someone would like to include it in the master repo. 😉