openwrt/packages #26283: owut upgrade fails with "Impossible package selection: luci version not as requested: 25.091.58779~2ff5aae vs. 25.094.58090~02dde50"

mikeb26, 2025-04-08:
I'm attempting to uprade my openwrt one with the latest firmware following instructions here: https://openwrt.org/docs/guide-user/installation/sysupgrade.owut

However, the owut upgrade step fails complaining about an impossible package selection. Log is included below:

```
root@openwrtone:~# owut upgrade
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         mediatek/filogic
Profile        openwrt_one
Package-arch   aarch64_cortex-a53
Version-from   24.10.0 r28427-6df0e3d02a (kernel 6.6.73)
Version-to     24.10.0 r28427-6df0e3d02a (kernel 6.6.73)
29 packages are out-of-date
Request hash:
  0dcbc465eda166c8ccf65e61cfa7453b4c1dcf750033fd09b4ffb1a25e272cee
--
Status:   Error: Impossible package selection: luci version not as requested: 25.091.58779~2ff5aae vs. 25.094.58090~02dde50
Progress:   0s total =   0s in queue +   0s in build

Build failed in   0s total =   0s in queue +   0s to build:
ASU server error =
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/rq/worker.py", line 1633, in perform_job
    return_value = job.perform()
                   ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1331, in perform
    self._result = self._execute()
                   ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1365, in _execute
    result = self.func(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/asu/build.py", line 252, in build
    report_error(job, err)
  File "/app/asu/util.py", line 308, in report_error
    raise RuntimeError(msg)
RuntimeError: Impossible package selection: luci version not as requested: 25.091.58779~2ff5aae vs. 25.094.58090~02dde50

ASU server stdout =
airoha-en8811h-firmware - 20241110-r1
attendedsysupgrade-common - 9
base-files - 1655~6df0e3d02a
busybox - 1.36.1-r2
ca-bundle - 20240203-r1
cgi-io - 2022.08.10~901b0f04-r21
ddns-scripts - 2.8.2-r64
ddns-scripts-services - 2.8.2-r64
dnsmasq - 2.90-r4
dropbear - 2024.86-r1
eip197-mini-firmware - 20241110-r1
firewall4 - 2024.12.18~18fc0ead-r1
fitblk - 2
fstools - 2024.07.14~408c2cc4-r1
fwtool - 2019.11.12~8f7fe925-r1
getrandom - 2024.04.26~85f10530-r1
hostapd-common - 2024.09.15~5ace39b0-r2
iw - 6.9-r1
iwinfo - 2024.10.20~b94f066e-r1
jansson - 2.14-r3
jshn - 2024.03.29~eb9bcb64-r1
jsonfilter - 2024.01.23~594cfa86-r1
kernel - 6.6.73~d649d775435da5a8637f7a955a80d331-r1
kmod-cfg80211 - 6.6.73.6.12.6-r1
kmod-crypto-acompress - 6.6.73-r1
kmod-crypto-aead - 6.6.73-r1
kmod-crypto-authenc - 6.6.73-r1
kmod-crypto-ccm - 6.6.73-r1
kmod-crypto-cmac - 6.6.73-r1
kmod-crypto-crc32c - 6.6.73-r1
kmod-crypto-ctr - 6.6.73-r1
[... 172 lines cut ...]
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_kmods https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/kmods/6.6.73-1-d649d775435da5a8637f7a955a80d331). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony). Skipping.

ERROR: Build failed with status 500
The above errors are often due to the upgrade server lagging behind the
build server, first suggestion is to wait a while and try again.
```

systemcrash, 2025-04-13:
should this go in the owut repo? Maybe just a refresh of the repo packages was necessary.

efahl, 2025-04-18:
> The above errors are often due to the upgrade server lagging behind the
> build server, first suggestion is to wait a while and try again.

All fixed now?