openwrt/packages #26301: ddns-scripts-dnspod and ddns-scripts-cloudflare can not autostart

openips, 2025-04-11:
build image with https://github.com/openwrt/packages/commit/15014f93e15bcb29956bf79cd397fea580bf6ea6
ddns-scripts is PKG_VERSION:=2.8.2 PKG_RELEASE:=68
reboot the router
 find ddns-scripts-dnspod and ddns-scripts-cloudflare can not autostart
@systemcrash

systemcrash, 2025-04-12:
Fixed in #26299

systemcrash, 2025-04-12:
Sorry for the noise - I was so busy when testing I forgot to reintroduce the backgrounding `&`.

openips, 2025-04-14:
PR https://github.com/openwrt/packages/pull/26299
test failed with arm a9

systemcrash, 2025-04-14:
Nothing to do with my PR. Just GitHub weather. 🤷‍♂️

openips, 2025-04-16:
use your pr https://github.com/openwrt/packages/pull/26299
reboot route and find ddns-scripts-dnspod and ddns-scripts-cloudflare can not autostart
but use
` /bin/sh /usr/lib/ddns/dynamic_dns_updater.sh -v 0 -S *** -- start&`
can autostart

openips, 2025-04-19:
pr https://github.com/openwrt/packages/pull/26299 test work very well

closed 2025-04-19 by openips