openwrt/packages #26302: Package `external-protocol` has unnamed dependency on package `ip-full`

brianredbeard, 2025-04-11:
The package `external-protocol` makes use of the `-json` flag of `ip`.  

https://github.com/openwrt/packages/blob/7383b4817b2482dab24b15c91311c83cd7fbdea9/net/external-protocol/files/external.sh#L44

This flag is not supported in the busybox `ip` binary, requiring the `ip-full` utility from the `iproute2` package:

https://github.com/openwrt/openwrt/blob/b68173500a1a6455a0faa3c4d54b2c62cf8593f1/package/network/utils/iproute2/Makefile#L46

When `ip-full` is not installed, `logread` is filled with busybox errors about the inability to parse the data, as can be seen below.

<details>
<summary>logread output</summary>
```
Fri Apr 11 20:36:06 2025 daemon.notice netifd: Interface 'container' is setting up now
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): BusyBox v1.36.1 (2025-03-09 16:28:50 UTC) multi-call binary.
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): Usage: ip [OPTIONS] address|route|link|neigh|rule [ARGS]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): OPTIONS := -f[amily] inet|inet6|link | -o[neline]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip route list|flush|add|del|change|append|replace|test ROUTE
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip link set IFACE [up|down] [arp on|off] [multicast on|off]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): 	[promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): 	[master IFACE | nomaster] [netns PID]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip rule [list] | add|del SELECTOR ACTION
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): Failed to parse message data
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): BusyBox v1.36.1 (2025-03-09 16:28:50 UTC) multi-call binary.
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): Usage: ip [OPTIONS] address|route|link|neigh|rule [ARGS]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): OPTIONS := -f[amily] inet|inet6|link | -o[neline]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip route list|flush|add|del|change|append|replace|test ROUTE
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip link set IFACE [up|down] [arp on|off] [multicast on|off]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): 	[promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): 	[master IFACE | nomaster] [netns PID]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): ip rule [list] | add|del SELECTOR ACTION
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): Failed to parse message data
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20794): interface br-container does not have ip address
Fri Apr 11 20:36:06 2025 daemon.notice netifd: Interface 'container' is now down
Fri Apr 11 20:36:06 2025 daemon.notice netifd: Interface 'container' is setting up now
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): BusyBox v1.36.1 (2025-03-09 16:28:50 UTC) multi-call binary.
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): Usage: ip [OPTIONS] address|route|link|neigh|rule [ARGS]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): OPTIONS := -f[amily] inet|inet6|link | -o[neline]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): ip route list|flush|add|del|change|append|replace|test ROUTE
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): ip link set IFACE [up|down] [arp on|off] [multicast on|off]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): 	[promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): 	[master IFACE | nomaster] [netns PID]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): ip rule [list] | add|del SELECTOR ACTION
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): Failed to parse message data
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): BusyBox v1.36.1 (2025-03-09 16:28:50 UTC) multi-call binary.
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): Usage: ip [OPTIONS] address|route|link|neigh|rule [ARGS]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): OPTIONS := -f[amily] inet|inet6|link | -o[neline]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836):
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): ip route list|flush|add|del|change|append|replace|test ROUTE
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): ip link set IFACE [up|down] [arp on|off] [multicast on|off]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): 	[promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
Fri Apr 11 20:36:06 2025 daemon.notice netifd: container (20836): 	[master IFACE | nomaster] [netns PID]
[... 15 lines cut ...]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): 	[promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): 	[master IFACE | nomaster] [netns PID]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): ip rule [list] | add|del SELECTOR ACTION
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): Failed to parse message data
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): BusyBox v1.36.1 (2025-03-09 16:28:50 UTC) multi-call binary.
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879):
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): Usage: ip [OPTIONS] address|route|link|neigh|rule [ARGS]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879):
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): OPTIONS := -f[amily] inet|inet6|link | -o[neline]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879):
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): ip route list|flush|add|del|change|append|replace|test ROUTE
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): ip link set IFACE [up|down] [arp on|off] [multicast on|off]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): 	[promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): 	[master IFACE | nomaster] [netns PID]
[... cut ...] ontainer (20879): ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): ip rule [list] | add|del SELECTOR ACTION
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): Failed to parse message data
Fri Apr 11 20:36:07 2025 daemon.notice netifd: container (20879): interface br-container does not have ip address
Fri Apr 11 20:36:07 2025 daemon.notice netifd: Interface 'container' is now down
```
</details>

cc @oskarirauta

GeorgeSapkin, 2025-04-13:
Which OpenWrt version are you on?

GeorgeSapkin, 2025-04-13:
Can you check if the PR addresses the issue for you? #26311

closed 2025-04-16 by BKPepe