openwrt/packages #26305: adblock: not blocking ads

brianjmurrell, 2025-04-12:
I have adblock installed and running on 24.10.0 (r28427-6df0e3d02a) but it's not blocking ads.  I am using unbound for my resolver on OpenWRT.

I can see that `/var/lib/unbound/adb_list.overall` is successfully created but I am not seeing anywhere in `/var/lib/unbound/unbound.conf` where that file is used.

How is adblock supposed to configure unbound to use the `adb_list.overall` file?

dibdot, 2025-04-12:
Please check the readme of unbound and adblock for that...or ask in the forum .

closed 2025-04-12 by dibdot

brianjmurrell, 2025-04-12:
There is nothing relevant to adblock suddenly not blocking ads in either readme.  It was working and without any configuration changes stopped working.

Now I've updated adblock from 4.2.6-r1 to 4.2.7-r1 and now it's working again, again with no other changes.  Clearly something fishy (read: buggy) was going on.