openwrt/packages #26313: adblock-fast: Memory availability check does not take available swap into account

bbergeron0, 2025-04-13:
I get the `Free RAM (<ram_available>) is not enough to process all enabled block-lists!` error message when starting adblock-fast, but my available RAM and swap combined should be more than enough for the task.

OpenWrt v24.10.0 (custom build +zram)  
adblock-fast 1.1.3

bbergeron0, 2025-04-13:
Here's a possible fix:

```diff
diff --git a/net/adblock-fast/files/etc/init.d/adblock-fast b/net/adblock-fast/files/etc/init.d/adblock-fast
index d10e4ae1e..ea38d0e77 100755
--- a/net/adblock-fast/files/etc/init.d/adblock-fast
+++ b/net/adblock-fast/files/etc/init.d/adblock-fast
@@ -362,9 +362,18 @@ is_running() {
        fi
 }
 ipset() { "$ipset" "$@" >/dev/null 2>&1; }
-get_ram_available() { ubus call system info | jsonfilter -e '@.memory.available'; }
-get_ram_free() { ubus call system info | jsonfilter -e '@.memory.free'; }
-get_ram_total() { ubus call system info | jsonfilter -e '@.memory.total'; }
+get_mem_available() {
+       local ram swap
+       ram="$( ubus call system info | jsonfilter -e '@.memory.available' )"
+       swap="$( ubus call system info | jsonfilter -e '@.swap.free' )"
+       echo "$((ram + swap))";
+}
+get_mem_total() {
+       local ram swap
+       ram="$( ubus call system info | jsonfilter -e '@.memory.total' )"
+       swap="$( ubus call system info | jsonfilter -e '@.swap.total' )"
+       echo "$((ram + swap))";
+}
 led_on(){ if [ -n "${1}" ] && [ -e "${1}/trigger" ]; then echo 'default-on' > "${1}/trigger" 2>&1; fi; }
 led_off(){ if [ -n "${1}" ] &&  [ -e "${1}/trigger" ]; then echo 'none' > "${1}/trigger" 2>&1; fi; }
 logger() { /usr/bin/logger -t "$packageName" "$@"; }
@@ -1305,7 +1314,7 @@ download_dnsmasq_file() {
        json set status 'statusDownloading'
 
        rm -f "$A_TMP" "$B_TMP" "$SED_TMP" "$outputFile" "$outputCache" "$outputGzip"
-       if [ "$(get_ram_available)" -lt 32 ]; then
+       if [ "$(get_mem_available)" -lt 32 ]; then
                output 3 'Low free memory, restarting resolver '
                if resolver 'quiet_restart'; then
                        output_okn
@@ -1360,7 +1369,7 @@ download_lists() {
                        [ -n "$size" ] && total_sizes=$((total_sizes+size))
                }
                local i free_mem total_sizes
-               free_mem="$(get_ram_available)"
+               free_mem="$(get_mem_available)"
                if [ -z "$free_mem" ]; then
                        json add warnning 'warningFreeRamCheckFail'
                        output "${_WARNING_}: $(get_text 'warningFreeRamCheckFail')!\n"
@@ -1384,7 +1393,7 @@ download_lists() {
        json set status 'statusDownloading'
 
        rm -f "$A_TMP" "$B_TMP" "$SED_TMP" "$outputFile" "$outputCache" "$outputGzip"
-       if [ "$(get_ram_total)" -lt 33554432 ]; then
+       if [ "$(get_mem_total)" -lt 33554432 ]; then
                output 3 'Low free memory, restarting resolver '
                if resolver 'quiet_restart'; then
                        output_okn
```

I included the swap in the available/total memory calculation and removed `get_ram_free` since it was unused. I haven't tested it yet, especially on a system without swap, but I will do that soon.

stangri, 2025-04-14:
Thanks, I'll try to test on swapless install and create PR for snapshots/24.10 within a week.

bbergeron0, 2025-04-14:
The script works on my router, and adblock-fast processed a 6 MB list with 7.5 MB of available RAM (+ 27 MB of free swap. The memory check estimates a `size * 2` requirement).

stangri, 2025-04-14:
@bbergeron0 are there both `'@.swap.free'` and `'@.swap.available'` entries on your system?

bbergeron0, 2025-04-14:
Yes, here's the output for `ubus call system info`:

```json
{
	"localtime": 1744651492,
	"uptime": 43688,
	"load": [
		0,
		0,
		0
	],
	"memory": {
		"total": 57630720,
		"free": 12374016,
		"shared": 3178496,
		"buffered": 0,
		"available": 5165056,
		"cached": 12402688
	},
	"root": {
		"total": 9088,
		"free": 8616,
		"used": 472,
		"avail": 8616
	},
	"tmp": {
		"total": 28140,
		"free": 24796,
		"used": 3344,
		"avail": 24796
	},
	"swap": {
		"total": 28307456,
		"free": 26472448
	}
}
```

Is it not there on swap-less systems?

closed 2025-04-16 by stangri