openwrt/packages #26314: RUST fails to extract because libdeflate-gzip -dc | tar -xf is being used, instead of just tar -zxf

Dnominated, 2025-04-13:
I draw your attention to the:

~/openwrt/staging_dir/host/bin/libdeflate-gzip -dc ~/openwrt/dl/rustc-1.85.0-src.tar.gz | tar -C ~/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.85.0-src/.. -xf -

command, below:

`mkdir -p ~/openwrt/dl
SHELL= flock ~/openwrt/tmp/.rustc-1.85.0-src.tar.gz.flock -c '       ~/openwrt/scripts/download.pl "~/openwrt/dl" "rustc-1.85.0-src.tar.gz" "2f4f3142ffb7c8402139cfa0796e24baaac8b9fd3f96b2deec3b94b4045c6a8a" "" "https://static.rust-lang.org/dist/"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://static.rust-lang.org/dist/rustc-1.85.0-src.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  565M  100  565M    0     0  10.8M      0  0:00:52  0:00:52 --:--:-- 10.9M
. ~/openwrt/include/shell.sh; ~/openwrt/staging_dir/host/bin/libdeflate-gzip -dc ~/openwrt/dl/rustc-1.85.0-src.tar.gz | tar -C ~/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.85.0-src/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors`
...
`make: *** [~/openwrt/include/toplevel.mk:233: world] Error 2`

I can
`curl https://static.rust-lang.org/dist/rustc-1.85.0-src.tar.gz --output rustc-1.85.0-src.tar.gz
mkdir rustctest
cd rustctest
tar -zxf ../rustc-1.85.0-src.tar.gz`

and it extracts normally.

Is there a way to have the build process use tar -xzf directly, instead of the Rube Goldberg method?

Xiaodx912, 2025-04-20:
Same issue here.
```
make[3]: Entering directory '/home/builder/openwrt/feeds/packages/lang/rust'
. /home/builder/openwrt/include/shell.sh; /home/builder/openwrt/staging_dir/host/bin/libdeflate-gzip -dc /home/builder/openwrt/dl/rustc-1.85.0-src.tar.gz | tar -C /home/builder/openwrt/build_dir/target-x86_64_musl/host/rustc-1.85.0-src/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [Makefile:111: /home/builder/openwrt/build_dir/target-x86_64_musl/host/rustc-1.85.0-src/.prepared06ecc841af17bcfa5a421b2aefaecaeb_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/home/builder/openwrt/feeds/packages/lang/rust'
```
System built in gzip works fine, but libdeflate-gzip will return 137 Killed.
```
root@ImageBuilder:/tmp/debug_build# /home/builder/openwrt/staging_dir/host/bin/libdeflate-gzip -dc /home/builder/openwrt/dl/rustc-1.85.0-src.tar.gz
Killed
root@ImageBuilder:/tmp/debug_build# echo $?
137
root@ImageBuilder:/tmp/debug_build# gzip -dc /home/builder/openwrt/dl/rustc-1.85.0-src.tar.gz|wc
65031178 303996063 3600653824
```

GeorgeSapkin, 2025-04-20:
@Xiaodx912 consider making a PR with your fixes.

BKPepe, 2025-05-23:
Let's bring attentation to rust maintainers here - @lu-zero

lu-zero, 2025-05-23:
It seems a problem specific of that implementation of gzip

Rondom, 2025-09-13:
What is your hypervisor and what is the configuration of your VM with regards to memory? My interest in this is mostly academic, because I assumed that provided that the memory is available, the balooning should be transparent to the guest and its applications. Maybe I am misinformed, but in any case I am curious about this...

I think there is not much that we can do from the OpenWrt. Using libdeflate-gzip has been a deliberate decision to speed up tarball extraction at the expense of higher RAM usage.

While the specific allocation pattern of allocating much at once is most likely the issue here, I think that building OpenWrt and packages using Rust and C++ possibly even with LTO will require some amount of RAM, so if you do not have 4 GB available, then other things will fail as well.

Xiaodx912, 2025-09-14:
@Rondom Thanks — quick update:
- I can no longer reproduce the OOM with 4/16 GiB ballooning.
- Since I recently fixed a host-side issue where the ZFS ARC limit was misconfigured, it’s very likely the earlier failure was due to insufficient host memory.

I’ll update my previous comment accordingly.

Dnominated, 2025-09-14:
> * it’s very likely the earlier failure was due to insufficient host memory.

Can you provide recommended settings?