openwrt/packages #26327: bind: Runs with "-4" even though I use IPv6 locally

rah2501, 2025-04-15:
@nmeyerhans

Hi,

I use IPv6 locally but don't use an Internet IPv6 route (or any IPv6 default route).  Unfortunately, my use case seems to have been intentionally disrupted by https://github.com/openwrt/packages/pull/20183 which disables bind's use of IPv6, in particular making it not listen on IPv6 interfaces.

Every time I upgrade my router, I wonder why DNS has stopped working on my network and then rediscover that I have to manually edit `/etc/init.d/named` on the router in order to replace `[ -z "$(ip -6 -o route show default)" ]` with `false` in the `no_ipv6` function.

Please could you either:
(1) not disable IPv6 by default (I'm not even sure why one would want to do that in 2025), or
(2) if you are going to disable IPv6, at least provide some easy way to prevent the disabling for those of us whose use case is disrupted.

Related: https://github.com/openwrt/packages/issues/20468

Thanks,

rah

nmeyerhans, 2025-04-22:
CC @pprindeville 

>  (1) not disable IPv6 by default (I'm not even sure why one would want to do that in 2025), or

IPv6 is not disabled by default.  It is only disabled if a specific check for IPv6 connectivity fails.

>  (2) if you are going to disable IPv6, at least provide some easy way to prevent the disabling for those of us whose use case is disrupted.

The mechanism to prevent this is to have working IPv6 connectivity. (It **is** 2025, as you note.)

I'm happy to consider options for how to make this handle your particular use case more gracefully. Maybe checking for non-empty output from `ip -6 addr show scope global` would be more accurate?

Alphix, 2025-07-28:
@nmeyerhans what happens if bind is run without ''-4'' unconditionally? I.e., does just blindly running ''bind'' without checking the ipv6 connectivity cause any problems?

nmeyerhans, 2025-07-28:
> [@nmeyerhans](https://github.com/nmeyerhans) what happens if bind is run without ''-4'' unconditionally? I.e., does just blindly running ''bind'' without checking the ipv6 connectivity cause any problems?

See the discussion in #20183 for the rationale. It was fixing a cosmetic issue more than anything else. AFAIK there were no functional concerns. A missing route to some destination will be a fast failure.

There are a couple of possibilities:
1. maybe newer named doesn't log these messages at all, I have not checked
2. maybe we could instead adjust logging to not emit them by default
3. I'm not entirely opposed to completely reverting the change. Though in most deployments I would expect a missing default route to be a reliable indicator that IPv6 is not in use, so it should generally be harmless.  OTOH, we have seen in this issue that there are some exceptions.

CC @pprindeville for comments, as he submitted the original PR to conditionally disable IPv6

Alphix, 2025-07-28:
> See the discussion in [#20183](https://github.com/openwrt/packages/pull/20183) for the rationale. It was fixing a cosmetic issue more than anything else. AFAIK there were no functional concerns. A [...]

I think this would be a better solution for those who are affected by the cosmetic issue:
https://bind9.readthedocs.io/en/v9.20.11/reference.html#namedconf-statement-query-source-v6

> There are a couple of possibilities:
>
> 1. maybe newer named doesn't log these messages at all, I have not checked
> 2. maybe we could instead adjust logging to not emit them by default
> 3. I'm not [...]

But it seems to me that it's not harmless if the ipv6 address isn't set up yet when bind starts (think `PPPoE` + `DS-Lite` and a reboot of the OpenWrt device, for example)...or in case someone configures IPv6 after installing `bind`...

pprindeville, 2025-08-03:
> I think this would be a better solution for those who are affected by the cosmetic issue: https://bind9.readthedocs.io/en/v9.20.11/reference.html#namedconf-statement-query-source-v6
>
>
> But it seems [...]

Someone might not want it listening to IPv6. They might not want it consuming resources needlessly. And yes, it’s unnecessary logging.

Alphix, 2025-08-03:
> Someone might not want it listening to IPv6.

Fair enough, then they can disable it. We're talking about a default in the init script here. A default which runs the risk of breaking IPv6 functionality.

IPv6 usage is at [50% worldwide](https://www.google.com/intl/en/ipv6/statistics.html). OpenWRT users probably skew towards a more advanced user base. OpenWRT users who use `bind` even more so. IOW, I would assume that > 50% of the userbase has IPv6. And that percentage is rising.

> They might not want it consuming resources needlessly.

Quick reality check (this is on a Debian system, not OpenWRT, I doubt it makes any difference):
```
$ ps ax | grep named | head -n1
   6016 ?        Ssl    0:00 /usr/sbin/named -f -u bind
$ ps -o vsz,rss -p  6016
   VSZ   RSS
609904 165864
$ systemctl stop named
$ vi /etc/default/named
$ systemctl start named
$ ps ax | grep named | head -n1
   7425 ?        Ssl    0:00 /usr/sbin/named -f -4 -u bind
$ ps -o vsz,rss -p  7425
   VSZ   RSS
609904 165352
$ echo $((165864 - 165352))
512
$ echo $((1000 * 512 / 165352))
3
```

A three promille change in memory usage. One extra `fd`. Is that worth breaking IPv6 functionality over?

> And yes, it’s unnecessary logging.

There's better ways of fixing that for the few users that are bothered by this cosmetic issue without breaking IPv6. I've already provided one example above.

pprindeville, 2025-08-03:
> IPv6 usage is at [50% worldwide](https://www.google.com/intl/en/ipv6/statistics.html). OpenWRT users probably skew towards a more advanced user base. OpenWRT users who use `bind` even more so. IOW, [...]

I'm talking about sockets.  Bind opens one socket per AF per core per interface.  It's a lot.  You can half that by not bringing up the AF if it's not provisioned.

Best practices in cyber security is that "anything not explicitly enabled is tacitly disabled".

If you want to add a bool to the UCI for IPv6 then that would be acceptable.

Alphix, 2025-08-03:
> I'm talking about sockets. Bind opens one socket per AF per core per interface. It's a lot. You can half that by not bringing up the AF if it's not provisioned.

Is it really "a lot"? How much? How much of what? RAM?

```
$ ls /proc/10230/fd | wc -l
354
$ ls -al /proc/10230/fd | grep socket | wc -l
168
<edit /etc/default/named, add -4 option>
$ systemctl restart named
$ ls /proc/11723/fd | wc -l
257
$ ls -al /proc/11723/fd | grep socket | wc -l
71
```

What's the impact of 38% extra fds? I'd say it's negligible on a router where you'd want to run `bind`. And we're still talking about breaking standard networking functionality...for the sake of saving some sockets.
 
> Best practices in cyber security is that "anything not explicitly enabled is tacitly disabled".

I fail to see the security implications. And so does the Debian maintainer (who is also working for upstream).

> If you want to add a bool to the UCI for IPv6 then that would be acceptable.

Adding a `dont_break_ipv6` bool seems pretty backwards.

pprindeville, 2025-08-03:
> Is it really "a lot"? How much? How much of what? RAM?
>
> ```
> $ ls /proc/10230/fd | wc -l
> 354
> $ ls -al /proc/10230/fd | grep socket | wc -l
> 168
> <edit /etc/default/named, add -4 option>
> $ systemctl [...]

It's more sockets than any other service on Openwrt as I've observed.  And the overhead of how polling works in the kernel is non-negligible scaling based on the number of sockets associated with an endpoint.

> > Best practices in cyber security is that "anything not explicitly enabled is tacitly disabled".
> 
> I fail to see the security implications. And so does the Debian maintainer (who is also working for upstream).

It was already pointed out by @nmeyerhans.  If at boot-time there's no IPv6 interface, but one later gets added by some form of tunneling (GRE, IPsec, PPPoX, etc) then this interface suddenly is open to Bind.  That's undesirable and it violates the principle of Least Astonishment.

> > If you want to add a bool to the UCI for IPv6 then that would be acceptable.
> 
> Adding a `dont_break_ipv6` bool seems pretty backwards.

So add `ipv4_only` and default it to `false`.  This isn't that hard.

rah2501, 2025-08-04:
> A default which runs the risk of breaking IPv6 functionality.

I raised this issue because as-is, the bind package breaks my network.  It isn't just a risk, breakage is actually occurring and has done since https://github.com/openwrt/packages/pull/20183 was merged.

pprindeville, 2025-08-04:
> > > What's the impact of 38% extra fds? I'd say it's negligible on a router where you'd want to run `bind`. And we're still talking about breaking standard networking functionality...for the sake of saving some sockets.
> > 
> > 
> > It's more sockets than any other service on Openwrt as I've observed. And the overhead of how polling works in the kernel is non-negligible scaling based on the number of sockets associated with an endpoint.
> 
> Non-negligible? So what is the impact? So far this is just hand-waving...

How is it handwaving?  I called out a very specific solution in my last message.

> > > > Best practices in cyber security is that "anything not explicitly enabled is tacitly disabled".
> > > 
> > > 
> > > I fail to see the security implications. And so does the Debian maintainer (who is also working for upstream).
> > 
> > 
> > It was already pointed out by [@nmeyerhans](https://github.com/nmeyerhans). If at boot-time there's no IPv6 interface, but one later gets added by some form of tunneling (GRE, IPsec, PPPoX, etc) then this interface suddenly is open to Bind. That's undesirable and it violates the principle of Least Astonishment.
> 
> And what if an IPv4 interface gets added later?

That's a moot point because clients use the loopback address locally.

> > > > If you want to add a bool to the UCI for IPv6 then that would be acceptable.
> > > 
> > > 
> > > Adding a `dont_break_ipv6` bool seems pretty backwards.
> > 
> > 
> > So add `ipv4_only` and default it to `false`. This isn't that hard.
> 
> I don't think I've claimed it's hard. But the equivalent of `ipv4_only` can already be achieved by the admin by editing `/etc/bind/named.conf`...is this really something that deserves a top-level UCI option?

No, you're claiming it's handwaving.

Security shouldn't be hard.  Most people don't know how to edit a `named.conf` file and they shouldn't have to.

Alphix, 2025-08-04:
> How is it handwaving? I called out a very specific solution in my last message.

It's handwaving because you're claiming "performance impact" with zero data to back up that claim. I say the difference in the number of sockets has no relevant performance impact, if you have any stats contradicting that, I'm all ears...

> That's a moot point because clients use the loopback address locally.

But `bind` listens to all IPv4 interfaces and all IPv6 interfaces out-of-the-box?

> No, you're claiming it's handwaving.
>
> Security shouldn't be hard. Most people don't know how to edit a `named.conf` file and they shouldn't have to.

If people don't know how to edit `named.conf`, it's rather questionable if they should install `bind`. But anyway, the defaults in this PR are safe. Upstream agrees. If you disagree, I'd expect some hard data.

pprindeville, 2025-08-15:
> 4. There is an issue with the current approach in the case where bind starts while IPv6 connectivity is still being established. In that case, bind may choose `-4` incorrectly, and there's currently no mechanism to reconcile the issue once IPv6 connectivity is established.

Even if it's not up on an interface, it will have still been provisioned, right?  Could we rely on that?

Alphix, 2025-08-15:
> Even if it's not up on an interface, it will have still been provisioned, right? Could we rely on that?

I don't think there is a heuristic that'll work. OpenWRT/bind may be running on a device which is not a router (e.g. a pure access point) and the RA announcing IPv6 connectivity could come in (or disappear) at any time, so it would require knowledge about the future.

PR #27471 (2025-09-19): bind: don't break IPv6 support
What started in #20183 as a attempt to clean up noise in the logfiles, turned out to be causing denial-of-service for dual-stack and especially IPv6-only environments.

Breaking core network functionality cannot possibly be less important than cosmetic issues, and those affected by log spam can avoid it via other means (e.g. "query-source-v6 none;" in named.conf).

There's no reliable heuristic for determining whether there's IPv6 connectivity at the time bind is started which will catch any and all corner cases, as discussed in #26327.

So, remove this logic for [... cut]