openwrt/packages #26339: Yggdrasil package wants too much space for installation

iav, 2025-04-17:
My router OpenWrt 24.10.1 (r28597-0425664679)
Trying to install yggdrasil via web gui, I see
yggdrasil	0.5.9-r1	5.81 MiB

got installation error
```

Collected errors:
 * verify_pkg_installable: Only have 5996kb available on filesystem /overlay, pkg yggdrasil needs 17040
 * opkg_install_cmd: Cannot install package yggdrasil.
```
not so much routers have 17 mb disk space. Isn't it too much? Possible, this requirement can be shrinked?
Maybe, with actual yggdrasil version?

systemcrash, 2025-04-18:
isn't 5.81 the compressed package size?

iav, 2025-04-18:
Oh, you right. Only 2 bnary files of yggdrasil are about 16 MB. Forgot about compression.
Sorry!

so huge...

wfleurant, 2025-04-19:
I think the build flags are as optimized as it gets as far as the last review I don't know three years ago?

wfleurant, 2025-04-19:
it's also worth mentioning the x64 binary are less in size versus arm at least they should be

iav, 2025-04-19:
Latest Yggdrasil released about 5 month ago, and it's v0.5.12.

iav, 2025-04-19:
I try to integrate yggprasil and yggdrasil-luci into system image — now openwrt website allows it.
but result system shows no working yggdrasil: no yggdrasil tab in web interface. And from the router ssh console I can to run yggdrasil and yggdrasilctl binaries and see version, but don't know how to run it with uci configuration correctly.

GeorgeSapkin, 2025-04-19:
> yggdrasil-luci

Do you mean `luci-proto-yggdrasil`?

iav, 2025-04-19:
Yes, sure

GeorgeSapkin, 2025-04-19:
You select a protocol when creating a new interface.

iav, 2025-04-19:
Sorry, currently I have no access to router I talking about. I will have a way to check everything on device from apr 22.
I'l post a screenshot of gui and paste from command line when obtain access to device.

iav, 2025-04-24:
Now I got access to my router.

I try to remove already exist from early versions `ygg` interface and add again.

Now I see — I have problems because I have old yggdrasil config in /etc/config/yggdrasil, seem it prevail over new config I set via luci.
Now I delete `/etc/config/yggdrasil`, and seems fresh added ygg interface works.

closed 2025-04-24 by iav