openwrt/packages #26354: dnscrypt-proxy2 package update

user10072023github, 2025-04-20:
Hi all, wondering if we can get https://github.com/openwrt/packages/tree/master/net/dnscrypt-proxy2 updated. Thanks

user10072023github, 2025-04-27:
?

justops1337, 2025-05-03:
PR [26452](https://github.com/openwrt/packages/pull/26452) has been created.

user10072023github, 2025-05-03:
Exciting news, thanks so much @justops1337 really appreciate you

user10072023github, 2025-05-31:
Hi all, what still needs to be done to get this later version of the package? I just tried the OpenWRT firmware selector tool and it pulls the old 2.1.5 version...sorry for not knowing how all this stuff works...

user10072023github, 2025-06-10:
Someone...?

erredi, 2026-01-09:
> > Can we re-open this issue since the new [v2.1.15](https://github.com/DNSCrypt/dnscrypt-proxy/releases/tag/2.1.15) is out there? I think that, after the great work of [@niklasthorild](https://github.com/niklasthorild) building the previous version, just update the source code and the new package will build successfully.
> 
> 2.1.15 is implemented in the SNAPSHOT builds. Is it a backport to 24.10 you're looking for?

I didn't realize it was available in the SNAPSHOT builds, yes I was referring to version 24.10 that currently the package has remained at version 2.1.14. Will it be updated automatically to the v2.1.15?