openwrt/packages #26356: mwan3 or One of its Dependencies Breaks netbird

cousinit99, 2025-04-20:
This applies to 24.10.0 or 24.10.1.  The issue appears to be the netbird or mwan3 packages, or one of their respective dependencies, and it appears to have surfaced within the past two months.

Normal/expected condition is netbird functional and connected:

> root@OpenWrt:~~# netbird up
> Connected
> root@OpenWrt:~# netbird status -d
> Peers detail:
>  xxxxxxxxxxx.netbird.cloud:
>   NetBird IP: xxxxxxxxxxxxxxxx
>   Public key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
>   Status: Disconnected
>   -- detail --
>   Connection type:
>   ICE candidate (Local/Remote): -/-
>   ICE candidate endpoints (Local/Remote): -/-
>   Relay server address:
>   Last connection update: 1 minute, 10 seconds ago
>   Last WireGuard handshake: -
>   Transfer status (received/sent) 0 B/0 B
>   Quantum resistance: false
>   Networks: -
>   Latency: 0s
> 
>  xxxxxxxxxxx.netbird.cloud:
>   NetBird IP: xxxxxxxxxxxxxx
>   Public key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
>   Status: Disconnected
>   -- detail --
>   Connection type:
>   ICE candidate (Local/Remote): -/-
>   ICE candidate endpoints (Local/Remote): -/-
>   Relay server address:
>   Last connection update: -
>   Last WireGuard handshake: -
>   Transfer status (received/sent) 0 B/0 B
>   Quantum resistance: false
>   Networks: -
>   Latency: 0s
> 
>  xxxxxxxxx.netbird.cloud:
>   NetBird IP: xxxxxxxxxxxxx
>   Public key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
>   Status: Disconnected
>   -- detail --
>   Connection type:
>   ICE candidate (Local/Remote): -/-
>   ICE candidate endpoints (Local/Remote): -/-
>   Relay server address:
>   Last connection update: -
>   Last WireGuard handshake: -
>   Transfer status (received/sent) 0 B/0 B
>   Quantum resistance: false
>   Networks: -
>   Latency: 0s
> 
>  xxxxxxxxxxxxx.netbird.cloud:
>   NetBird IP: xxxxxxxxxxxxxxx
>   Public key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
>   Status: Disconnected
>   -- detail --
>   Connection type:
>   ICE candidate (Local/Remote): -/-
>   ICE candidate endpoints (Local/Remote): -/-
>   Relay server address:
>   Last connection update: -
>   Last WireGuard handshake: -
>   Transfer status (received/sent) 0 B/0 B
>   Quantum resistance: false
>   Networks: -
>   Latency: 0s
> 
> Events:
>   [WARNING] DNS (0f7f8c6a-7ad4-4fd4-9ad7-bd84affd0030)
>     Message: The host dns manager does not support match domains
>     Time: 1 minute, 10 seconds ago
>     Metadata: manager: file
>   [INFO] SYSTEM (53908175-00e8-4e9b-82d6-ad1bc3eabd08)
>     Message: Network map updated
>     Time: 1 minute, 10 seconds ago
> OS: linux/arm64
> Daemon version: 0.39.2
> CLI version: 0.39.2
> **Management: Connected to https://api.netbird.io:443
> Signal: Connected to https://signal.netbird.io:443**
> Relays:
>   **[stun:stun.netbird.io:5555] is Available
>   [turns:turn.netbird.io:443?transport=tcp] is Available
>   [rels://streamline-us-sjo1-0.relay.netbird.io:443] is Available**
> Nameservers:
> FQDN: xxxxxxxxxx.netbird.cloud
> NetBird IP: xxxxxxxxxxxxxxxx
> Interface type: Kernel
> Quantum resistance: false
> Networks: -
> Forwarding rules: 0
> Peers count: 0/4 Connected

Immediately after installing mwan3 2.11.16-r5 + dependencies in an environment where netbird 0.39.2 has already installed, configured, and confirmed to be functioning normally, `netbird status -d` returns the following:

> root@OpenWrt:~# netbird status -d
> Peers detail:
>  xxxxxxxxxxxxxx.netbird.cloud:
>   NetBird IP: xxxxxxxxxxxxxxxx
>   Public key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
>   Status: Disconnected
>   -- detail --
>   Connection type:
>   ICE candidate (Local/Remote): -/-
>   ICE candidate endpoints (Local/Remote): -/-
>   Relay server address:
>   Last connection update: 8 minutes, 38 seconds ago
>   Last WireGuard handshake: -
>   Transfer status (received/sent) 0 B/0 B
>   Quantum resistance: false
>   Networks: -
>   Latency: 0s
> 
>  xxxxxxxxxx.netbird.cloud:
>   NetBird IP: xxxxxxxxxxxx
>   Public key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
>   Status: Disconnected
>   -- detail --
>   Connection type:
>   ICE candidate (Local/Remote): -/-
>   ICE candidate endpoints (Local/Remote): -/-
>   Relay server address:
>   Last connection update: -
>   Last WireGuard handshake: -
>   Transfer status (received/sent) 0 B/0 B
>   Quantum resistance: false
>   Networks: -
>   Latency: 0s
> 
>  xxxxxxxxxxx.netbird.cloud:
>   NetBird IP: xxxxxxxxxxxxxx
>   Public key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
>   Status: Disconnected
>   -- detail --
>   Connection type:
>   ICE candidate (Local/Remote): -/-
>   ICE candidate endpoints (Local/Remote): -/-
>   Relay server address:
>   Last connection update: -
>   Last WireGuard handshake: -
>   Transfer status (received/sent) 0 B/0 B
>   Quantum resistance: false
>   Networks: -
>   Latency: 0s
> 
> Events:
>   [WARNING] DNS (d8d8486a-c5f6-40d9-84fd-b5c0458f7592)
>     Message: The host dns manager does not support match domains
>     Time: 8 minutes, 38 seconds ago
>     Metadata: manager: file
>   [INFO] SYSTEM (e356582c-8d5d-4ac6-a456-ab7fc578ff00)
>     Message: Network map updated
>     Time: 8 minutes, 38 seconds ago
> OS: linux/arm64
> Daemon version: 0.39.2
> CLI version: 0.39.2
> **Management: Disconnected, reason: rpc error: code = Unavailable desc = keepalive ping failed to receive ACK within timeout
> Signal: Disconnected, reason: rpc error: code = Unavailable desc = keepalive ping failed to receive ACK within timeout
> Relays:
>   [stun:stun.netbird.io:5555] is Unavailable, reason: dial: failed to listen: d.Dialer.DialContext: dial udp 209.94.63.37:5555: connect: invalid argument
>   [turns:turn.netbird.io:443?transport=tcp] is Unavailable, reason: dial: d.Dialer.DialContext: dial tcp 209.94.61.253:443: connect: invalid argument
>   [rels://relay.netbird.io:443] is Unavailable, reason: relay connection is not established**
> Nameservers:
> FQDN: xxxxxxxxx.netbird.cloud
> NetBird IP: xxxxxxxxxxxxxxxxx
> Interface type: Kernel
> Quantum resistance: false
> Networks: -
> Forwarding rules: 0
> Peers count: 0/3 Connected

Cycling the interface returns the following:

> root@OpenWrt:~~# netbird down
> Disconnected
> root@OpenWrt:~# netbird up
> Error: login backoff cycle failed: rpc error: code = Unknown desc = getting device authorization flow info failed with error: context deadline exceeded

The mwan3 dependency versions from 1-2 months ago where the issue was not present are as follows:

> ip6tables-zz-legacy - 1.8.10-r1
> ipset - 7.21-r1
> iptables-mod-conntrack-extra - 1.8.10-r1
> iptables-mod-ipopt - 1.8.10-r1
> iptables-zz-legacy - 1.8.10-r1
> ip-tiny - 6.11.0-r1
> kmod-ip6tables - 6.6.73-r1
> kmod-ipt-conntrack - 6.6.73-r1
> kmod-ipt-conntrack-extra - 6.6.73-r1
> kmod-ipt-core - 6.6.73-r1
> kmod-ipt-ipopt - 6.6.73-r1
> kmod-ipt-ipset - 6.6.73-r1
> kmod-nf-conncount - 6.6.73-r1
> kmod-nf-conntrack - 6.6.73-r1
> kmod-nf-ipt - 6.6.73-r1
> kmod-nf-ipt6 - 6.6.73-r1
> libip4tc2 - 1.8.10-r1
> libip6tc2 - 1.8.10-r1
> libipset13 - 7.21-r1
> libiptext0 - 1.8.10-r1
> libxtables12 - 1.8.10-r1
> mwan3 - 2.11.16-r5
> xtables-legacy - 1.8.10-r1

The current mwan3 dependency versions where the issue is present are as follows (differences highlighted):

> ip6tables-zz-legacy (1.8.10-r1)
> ipset (7.21-r1)
> iptables-mod-conntrack-extra (1.8.10-r1)
> iptables-mod-ipopt (1.8.10-r1)
> iptables-zz-legacy (1.8.10-r1)
> ip-tiny (6.11.0-r1)
> **kmod-ip6tables (6.6.86-r1)
> kmod-ipt-conntrack (6.6.86-r1)
> kmod-ipt-conntrack-extra (6.6.86-r1)
> kmod-ipt-core (6.6.86-r1)
> kmod-ipt-ipopt (6.6.86-r1)
> kmod-ipt-ipset (6.6.86-r1)
> kmod-nf-conncount (6.6.86-r1)
> kmod-nf-ipt (6.6.86-r1)
> kmod-nf-ipt6 (6.6.86-r1)**
> libip4tc2 (1.8.10-r1)
> libip6tc2 (1.8.10-r1)
> libipset13 (7.21-r1)
> libiptext0 (1.8.10-r1)
> libiptext6-0 (1.8.10-r1)
> libxtables12 (1.8.10-r1)
> mwan3 (2.11.16-r5)
> xtables-legacy (1.8.10-r1)

feckert, 2025-04-22:
If I see this correctly, then only the version from the kernel was update from `6.6.73` to `6.6.86`.

I don't think that has anything to do with it. Have you ever tried the mwan3 without nebird? Did it work with the old and the new version?

cousinit99, 2025-04-22:
mwan3 was installed and functioning alongside netbird before I upgraded to 24.10.1, correct.  This was with netbird 0.36.7 and the same mwan3 binary version.   Now, netbird won't connect after mwan3 is installed.  Should at least be reproducible, so no need to take my word for it.

wehagy, 2025-04-23:
Someone mentioned this issue to me a few months ago. For more context, see: https://github.com/netbirdio/netbird/issues/3363#issuecomment-2675491800.

cousinit99, 2025-04-24:
https://github.com/openwrt/packages/pull/26002#issuecomment-2670209028

cousinit99, 2025-04-24:
> Someone mentioned this issue to me a few months ago. For more context, see: [netbirdio/netbird#3363 (comment)](https://github.com/netbirdio/netbird/issues/3363#issuecomment-2675491800).

Thanks, I wonder how I managed to get them running concurrently before the upgrade...  Feel free to close as duplicate if that seems appropriate.  I don't see an actual issue opened for it, so I'll leave as-is for now.

wehagy, 2025-04-24:
> Thanks, I wonder how I managed to get them running concurrently before the upgrade... Feel free to close as duplicate if that seems appropriate. I don't see an actual issue opened for it, so I'll [...]

I think it’s best to leave the issue open if it still exists, I also opened an issue at https://github.com/netbirdio/netbird/issues/3736 in the upstream.

hashtagsandro, 2026-11-01:
Still exists on OpenWrt 25.12.6 tested all yesterday before see this issue .... OMG