openwrt/packages #26428: Strongswan build error

vochong, 2025-05-01:
```
make[9]: Entering directory '/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/src/libcharon/plugins/uci'
/bin/bash ../../../../libtool  --tag=CC   --mode=compile ccache aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../../..  -I../../../../src/libstrongswan -I../../../../src/libcharon -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/usr/include -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/include -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/include/fortify    -rdynamic -Wno-format -Wno-format-security -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-pointer-sign -Wno-sign-compare -Wno-type-limits -Wno-unused-parameter -O3 -pipe -mcpu=cortex-a53+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14=strongswan-5.9.14 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -include /home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/config.h -c -o uci_plugin.lo uci_plugin.c
OpenWrt-libtool: compile:  ccache aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../../.. -I../../../../src/libstrongswan -I../../../../src/libcharon -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/usr/include -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/include -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/include/fortify -rdynamic -Wno-format -Wno-format-security -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-pointer-sign -Wno-sign-compare -Wno-type-limits -Wno-unused-parameter -O3 -pipe -mcpu=cortex-a53+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14=strongswan-5.9.14 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -include /home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/config.h -c uci_plugin.c  -fPIC -DPIC -o .libs/uci_plugin.o
/bin/bash ../../../../libtool  --tag=CC   --mode=compile ccache aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../../..  -I../../../../src/libstrongswan -I../../../../src/libcharon -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/usr/include -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/include -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/include/fortify    -rdynamic -Wno-format -Wno-format-security -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-pointer-sign -Wno-sign-compare -Wno-type-limits -Wno-unused-parameter -O3 -pipe -mcpu=cortex-a53+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14=strongswan-5.9.14 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -include /home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/config.h -c -o uci_parser.lo uci_parser.c
OpenWrt-libtool: compile:  ccache aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../../.. -I../../../../src/libstrongswan -I../../../../src/libcharon -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/usr/include -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/include -I/home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/include/fortify -rdynamic -Wno-format -Wno-format-security -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-pointer-sign -Wno-sign-compare -Wno-type-limits -Wno-unused-parameter -O3 -pipe -mcpu=cortex-a53+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14=strongswan-5.9.14 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -include /home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/config.h -c uci_parser.c  -fPIC -DPIC -o .libs/uci_parser.o
uci_parser.c: In function 'section_enumerator_enumerate':
uci_parser.c:76:21: error: implicit declaration of function 'uci_lookup'; did you mean 'uci_lookup_ptr'? [-Wimplicit-function-declaration]
   76 |                 if (uci_lookup(this->ctx, &element, this->package,
      |                     ^~~~~~~~~~
      |                     uci_lookup_ptr
make[9]: *** [Makefile:607: uci_parser.lo] Error 1
make[9]: Leaving directory '/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/src/libcharon/plugins/uci'
make[8]: *** [Makefile:2139: all-recursive] Error 1
make[8]: Leaving directory '/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/src/libcharon'
make[7]: *** [Makefile:1423: all] Error 2
make[7]: Leaving directory '/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/src/libcharon'
make[6]: *** [Makefile:543: all-recursive] Error 1
make[6]: Leaving directory '/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/src'
make[5]: *** [Makefile:605: all-recursive] Error 1
make[5]: Leaving directory '/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14'
make[4]: *** [Makefile:516: all] Error 2
make[4]: Leaving directory '/home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14'
make[3]: *** [Makefile:656: /home/user/mnt/sdb1/.owrt/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.14/.built] Error 2
make[3]: Leaving directory '/home/user/mnt/sdb1/.owrt/qosmio-nss/feeds/packages/net/strongswan'
time: package/feeds/packages/strongswan/compile#116.74#77.53#185.92
    ERROR: package/feeds/packages/strongswan failed to build.
make[2]: *** [package/Makefile:189: package/feeds/packages/strongswan/compile] Error 1
make[2]: Leaving directory '/home/user/mnt/sdb1/.owrt/qosmio-nss'
make[1]: *** [package/Makefile:183: /home/user/mnt/sdb1/.owrt/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/mnt/sdb1/.owrt/qosmio-nss'
make: *** [/home/user/mnt/sdb1/.owrt/qosmio-nss/include/toplevel.mk:233: world] Error 2
```

tiagogaspar8, 2025-05-04:
Facing the same issue, tagging @pprindeville

pprindeville, 2025-05-04:
There's an untested PR for this: #25353.

Not sure if it should be `uci_loopup_next()` or `uci_lookup_ptr()`.  Hoping that @jow- can chime it.

tiagogaspar8, 2025-05-04:
> There's an untested PR for this: https://github.com/openwrt/packages/pull/25353.

You're right, this fixes it for me, can we get it merged?

pprindeville, 2025-05-04:
Just waiting on @neheb to address a couple of review comments before that happens.