openwrt/packages #26442: Add DHCP clients to promtheus-exporter

mancubus77, 2025-05-02:
**Problem Statement**
Prometheus exporter doesn't report clients, therefore all grafana graphs have mac addresses instead of hostname 

**Solution**
Add prometheus exporter to parse `/var/dhcp.leases` and report clients for example

```
dhcp_client{mac="B8:2D:28:AF:DC:33",hostname="mygoodhost",ip="192.168.1.115"} 1
```

Aladex, 2025-05-18:
`local function scrape()
  local metric_dhcp_leases = metric("dhcp_leases_total", "gauge")
  local metric_dhcp_lease_info = metric("dhcp_lease_info", "gauge")
  local leases_file = "/tmp/dhcp.leases"
  local leases_count = 0
  local file = io.open(leases_file, "r")
  if file then
    for line in file:lines() do
      local expiry, mac, ip, hostname, client_id = line:match("^(%d+)%s+([%x:]+)%s+([%d%.]+)%s+([^%s]*)%s+([^%s]*)$")
      if ip and mac then
        leases_count = leases_count + 1
        local labels = {
          ip = ip,
          mac = string.upper(mac),
          hostname = (hostname == "*" or hostname == "") and "unknown" or hostname,
          client_id = (client_id == "*" or client_id == "") and "unknown" or client_id
        }
        metric_dhcp_lease_info(labels, 1)
      end
    end
    file:close()
  end
  metric_dhcp_leases({}, leases_count)
end
return { scrape = scrape }
`
Just add this script in /usr/lib/lua/prometheus-collectors/
And `chmod +x /usr/lib/lua/prometheus-collectors/dhcp_leases.lua`
Thats it

mancubus77, 2025-05-18:
Thanks @Aladex 
Yes, I do have the code as well and intended to raise PR, but didn't managed to pass through checks :-D

closed 2025-05-18 by mancubus77