openwrt/packages #26453: xtables-addons: fails to build with kernel 6.12 on x86_64

jmarcet, 2025-05-03:
```
# CC [M]  /home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions/rtsp/nf_conntrack_rtsp.o
  ccache x86_64-openwrt-linux-musl-gcc -Wp,-MMD,/home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions/rtsp/.nf_conntrack_rtsp.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -O3 -fivopts -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -mno-avx2 -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fmodulo-sched -fmodulo-sched-allow-regmoves -fno-allow-store-data-races -fstack-protector-strong -fomit-frame-pointer -ftrivial-auto-var-init=zero -fno-stack-clash-protection -fzero-call-used-regs=used-gpr -falign-functions=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -fno-builtin-wcslen -fno-builtin-wcslen -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-main -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-overflow -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wextra -Wunused -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-truncation -Wno-override-init -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -Wno-unused-parameter -march=alderlake -ffile-prefix-map=/home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves  -DMODULE  -DKBUILD_BASENAME='"nf_conntrack_rtsp"' -DKBUILD_MODNAME='"nf_conntrack_rtsp"' -D__KBUILD_MODNAME=kmod_nf_conntrack_rtsp -c -o /home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions/rtsp/nf_conntrack_rtsp.o /home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions/rtsp/nf_conntrack_rtsp.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --hacks=skylake --orc --retpoline --rethunk --static-call --uaccess --prefix=16   --module /home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions/rtsp/nf_conntrack_rtsp.o
../openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions/rtsp/nf_conntrack_rtsp.c: In function 'init':
../openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions/rtsp/nf_conntrack_rtsp.c:738:17: error: implicit declaration of function 'strlcpy'; did you mean 'strncpy'? [-Wimplicit-function-declaration]
  738 |                 strlcpy(hlpr->name, tmpname, sizeof(hlpr->name));
      |                 ^~~~~~~
      |                 strncpy
make[10]: *** [scripts/Makefile.build:229: /home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions/rtsp/nf_conntrack_rtsp.o] Error 1
make[9]: *** [scripts/Makefile.build:478: /home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions/rtsp] Error 2
make[8]: *** [/home/jmarcet/src/openwrt/linux/Makefile:1947: /home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions] Error 2
make[7]: *** [Makefile:224: __sub-make] Error 2
make[7]: Leaving directory '/home/jmarcet/src/openwrt/linux'
make[6]: *** [Makefile:457: modules] Error 2
make[6]: Leaving directory '/mnt/caches/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/extensions'
make[5]: *** [Makefile:619: all-recursive] Error 1
make[5]: Leaving directory '/mnt/caches/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27'
make[4]: *** [Makefile:391: all] Error 2
make[4]: Leaving directory '/mnt/caches/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27'
make[3]: *** [Makefile:203: /home/jmarcet/src/openwrt/openwrt-master-x64/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.27/.built] Error 2
make[3]: Leaving directory '/home/jmarcet/src/openwrt/openwrt-master-x64/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#0.67#0.10#0.82
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:189: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/home/jmarcet/src/openwrt/openwrt-master-x64'
make[1]: *** [package/Makefile:183: /home/jmarcet/src/openwrt/openwrt-master-x64/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jmarcet/src/openwrt/openwrt-master-x64'
make: *** [/home/jmarcet/src/openwrt/openwrt-master-x64/include/toplevel.mk:233: world] Error 2
```

neheb, 2025-05-04:
https://github.com/maru-sama/rtsp-linux/commit/4b67fe8f5a380f25038fc831f3ec23aa11ce5dca

sbwml, 2025-05-20:
p1

```diff
--- a/extensions/rtsp/nf_conntrack_rtsp.c
+++ b/extensions/rtsp/nf_conntrack_rtsp.c
@@ -735,8 +735,12 @@ init(void)
 		}
 
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(3,6,0)
+#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 8, 0)
 		strlcpy(hlpr->name, tmpname, sizeof(hlpr->name));
 #else
+		strscpy(hlpr->name, tmpname, sizeof(hlpr->name));
+#endif
+#else
 		hlpr->name = tmpname;
 #endif
 		pr_debug("port #%d: %d\n", i, ports[i]);
```

p2
```diff
--- a/extensions/xt_ipp2p.c
+++ b/extensions/xt_ipp2p.c
@@ -3,7 +3,11 @@
 #include <linux/netfilter_ipv4/ip_tables.h>
 #include <net/tcp.h>
 #include <net/udp.h>
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 12, 0)
+#include <linux/unaligned.h>
+#else
 #include <asm/unaligned.h>
+#endif
 #include "xt_ipp2p.h"
 #include "compat_xtables.h"
 
```

ptpt52, 2025-05-22:
please fix

backslashxx, 2025-05-22:
ye that first patch is needed, also

CONFIG_TEXTSEARCH_BM=y

on kernel config

namiltd, 2025-05-22:
Patch for this is already created, but still not accepted: https://github.com/openwrt/packages/pull/26368
Please test it.

backslashxx, 2025-05-22:
very nice, thanks for your work

BKPepe, 2025-09-24:
Fixed by https://github.com/openwrt/packages/commit/b093eea601195184ea1364c4678c35f96269f7f5