openwrt/packages #26459: mwan3: Use fewer bits for MMX_MASK

pprindeville, 2025-05-04:
The default for `MMX_MASK` in `/lib/mwan3/common.sh` is `0x3f00` if `mmx_mask` isn't otherwise defined in the `globals` section.

That's 6 bits out of a sixteen bit field.

Can we get by with fewer bits?  Users writing their own custom rules might want to use some of the marking bits themselves.

feckert, 2025-05-05:
The mwan3 is designed so that it can monitor a maximum of 60 interfaces with the default `MMX_MASK` setting.
```
Mon May  5 11:26:15 2025 user.debug mwan3-init[22733]: Using firewall mask 0x3F00
Mon May  5 11:26:15 2025 user.debug mwan3-init[22733]: Max interface count is 60
```
If you want fewer interfaces, you can also change the UCI configuration option  `mmx_mask` value to `0x0340`.
So only 4 bits are used. But then only 4 interfaces can be tracked.
```
Mon May  5 11:40:52 2025 user.debug mwan3-init[30296]: Using firewall mask 0x0340
Mon May  5 11:40:52 2025 user.debug mwan3-init[30296]: Max interface count is 4
```

It has always been the case that a maximum of 60 interfaces could be tracked here. This used to be static and could not be changed in the past. With the UCI configuration option `mmx_mask` it was possible to change the bit usage and the interface count.

BKPepe, 2025-05-23:
There is also discussion in mailing list:
https://lists.openwrt.org/pipermail/openwrt-devel/2025-May/043967.html

pprindeville, 2025-05-31:
> [@pprindeville](https://github.com/pprindeville) I hope this explanation is enough for you, so I'm closing the issue.

Just need to know what bits can safely be used without risk of conflict.

feckert, 2025-06-02:
@feckert This depends on the MMX_MASK value. You can use all other bits that are **not** set in the MMX_MASK. So in default configuration of mwan3 you can use all other bits that are **not** covert in  den MMX_MASK `0x3F00`