openwrt/packages #26470: package docker missing

congnvp, 2025-05-07:
I've installed openwrt 24.10.1 in x86 machine. But today I cannot see any docker package exist in repository (that luci-app-dockerman required). How can I fix?

drajabr, 2025-05-11:
This is so weird, on the first few days from 24 release, I updated my cluster from 23 to 24 clean without issue, installed docker fine and moved on.

TODAY, I built a new node, but all instances inside id returns me package not found.

```
root@dev:~# opkg install docker dockerd docker-compose luci-app-dockerman
Unknown package 'docker'.
Unknown package 'dockerd'.
Unknown package 'docker-compose'.
Unknown package 'luci-app-dockerman'.
Collected errors:
 * opkg_install_cmd: Cannot install package docker.
 * opkg_install_cmd: Cannot install package dockerd.
 * opkg_install_cmd: Cannot install package docker-compose.
 * pkg_hash_check_unresolved: cannot find dependency docker for luci-app-dockerman
 * pkg_hash_check_unresolved: cannot find dependency dockerd for luci-app-dockerman
 * pkg_hash_check_unresolved: cannot find dependency docker-compose for luci-app-dockerman
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-dockerman found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-dockerman.
```

Apparently, no docker, docker-compose, or dockerd package could be found in the [repo](https://mirror-03.infra.openwrt.org/releases/24.10.1/packages/x86_64/)

drajabr, 2025-05-11:
Idk what is the issue, but here's what I found:

Apparently this "archive" mirror has all docker packages:
https://archive.openwrt.org/releases/24.10.1/packages/x86_64/packages/

BUT other ones don't, for instance, the default opkg feed file:
https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages 

Which from my location leads to 
https://mirror-03.infra.openwrt.org/releases/24.10.1/packages/x86_64/packages/

HOWEVER, even in the archive repo, the packages index doesn't include the reference to the docker packages, so its useless to switch to archive in the feed, as those packages still won't show up for opkg.

This is apparently an issue and hopefully it'll get fixed soon. as even manually downloading packages and installing them is quite cumbersome due to the dependencies and errors like this:

```
opkg install /tmp/upload.ipk
Unknown package 'containerd'.
Errors
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency runc for containerd
 * pkg_hash_fetch_best_installation_candidate: Packages for containerd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package containerd.
```

GeorgeSapkin, 2025-05-11:
There are other packages that seem to affected as well, like lua5.4.

drajabr, 2025-05-11:
@congnvp  I see docker packages just appeared actually. 

But yeah @GeorgeSapkin still missing `lua5.4`. Also `dysk`, `bottom`, and `arp-whisper` packages.

The issue is gone for me, but I'm really interested to know what happened.

stangri, 2025-09-15:
4 months later, with updated packages list on a fresh 24.10 install on x86_64:

```
# opkg install docker-compose luci-app-dockerman
Unknown package 'docker-compose'.
Unknown package 'luci-app-dockerman'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency docker for docker-compose
 * pkg_hash_fetch_best_installation_candidate: Packages for docker-compose found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package docker-compose.
 * pkg_hash_check_unresolved: cannot find dependency docker for luci-app-dockerman
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-dockerman found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-dockerman.
```

@jow- was `docker` a meta package which got removed for some reason? The `dockerd` installs fine. Image Builder fails with the same messages with those two packages are being added.