openwrt/packages #26471: Snapshot package Buildbot worker out of space

kuanyili, 2025-05-08:
[ffb-dock-01](https://buildbot.openwrt.org/main/packages/#/workers/24) has no space left, generating lots of errors.

https://buildbot.openwrt.org/main/packages/#/workers

GeorgeSapkin, 2025-05-08:
Maybe this should be reported here instead: https://github.com/openwrt/docker/issues

ynezz, 2025-05-10:
Fixed, thanks for the report!

closed 2025-05-10 by ynezz

kuanyili, 2025-05-10:
It's also happening on other Buildbot workers. Is something triggering this?

ynezz, 2025-05-12:
> It's also happening on other Buildbot workers. Is something triggering this?

I've fixed that as well, thanks.

@hnyman noted following https://lists.openwrt.org/pipermail/openwrt-devel/2025-May/043984.html 

> Could there be a need for some kind of larger (cache) cleanup, due to the recent toolchain changes impacting SDKs?  (gcc 14, gdb 16.2)

Could be a combination of multiple events, indeed, being more visible on build workers with limited storage. I tried to look into this, but the I/O throughput of the storage is unusable for examination of 70GB build directory, so I gave up.

GeorgeSapkin, 2025-05-12:
Are the same workers used in CI? There's are still multiple jobs failing with core dumped.