openwrt/packages #26485: acme-common 1.4.0 package: bad symlink

KsmoinO, 2025-05-10:
Requesting certificates with webroot method and default settings failed during challenge

`/www/.well-known/acme-challenge` point to `/var/run/acme/challenge/` instead of `/var/run/acme/challenge/.well-known/acme-challenge`

Maybe `/www/.well-known` should point directly to `/var/run/acme/challenge/.well-known/`?

KsmoinO, 2025-05-10:
As workaround I created `/etc/hotplug.d/acme/00-uhttpd`
```
case "$ACTION" in
  prepare)
    [ -d /www/.well-known ] && rm -r /www/.well-known
    [ $(readlink -f /www/.well-known) != "/var/run/acme/challenge/.well-known" ] && ln -sf /var/run/acme/challenge/.well-known /www/.well-known
    ;;
  issued|renewed)
    /etc/init.d/uhttpd reload
    ;;
  *)
    echo Unknown action $ACTION
    ;;
esac
```

BKPepe, 2025-05-20:
@KsmoinO Which OpenWrt version are you using?

KsmoinO, 2025-05-20:
I use OpenWrt  24.10.1
Only acme-common v1.4.0 is available for now.

tohojo, 2025-05-20:
https://github.com/openwrt/packages/pull/26542

closed 2025-05-21 by BKPepe

KsmoinO, 2025-05-22:
I confirm it's fixed on 1.4.3
Thanks