openwrt/packages #26493: owut fails on reboot (rpi4)

K2IE, 2025-05-11:
Running owut upgrade on 24.10.1, it found a handful of packages to upgrade.  It built the image, but a a zcat error was when the script was dealing with the partitions.  The script went to reboot and the system failed to come up.

This is on an Rpi CM4 based system, bcm2711 architecture.

efahl, 2025-05-12:
Can you post the errors and some context?  What produced the zcat error?

efahl, 2025-05-12:
I'm mostly curious as to what phase of the upgrade the errors occurred.  Seems like the build was fine, downloaded and verified ok, but then sysupgrade had some internal failures that caused the issue?  Were you able to gain access with the partitions mangled (or whatever), or did you just bail out and reimage from scratch?

K2IE, 2025-05-12:
> efahl

I could gain access by mounting on another system and the filesystem
was there.  I suspect something went wrong that affected the boot
partition.  But then I reimaged.

The error was toward the end of the process.

[1] (openwrt/packages#26493)
    https://github.com/openwrt/packages/issues/26493#issuecomment-2874356353
[2] unsubscribe
    https://github.com/

BKPepe, 2025-05-20:
Unfortunately, I need to close this issue. No logs were provided.

closed 2025-05-20 by BKPepe

efahl, 2025-05-20:
@K2IE If you see this again, please open an issue on https://github.com/efahl/owut/issues