openwrt/packages #26516: dnsproxy: failed to sufficiently increase receive buffer size

Mile-Lile, 2025-05-15:
@1715173329 
on the latest SNAPSHOT build from [openwrt firmware selector](https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=mediatek%2Ffilogic&id=xiaomi_mi-router-ax3000t-ubootmod) on AX3000T with dnsproxy version v0.75.4 I get this message:

 `failed to sufficiently increase receive buffer size (was: 208 kiB, wanted: 7168 kiB, got: 416 kiB). See https://github.com/quic-go/quic-go/wiki/UDP-Buffer-Sizes for details.`

From time to time I have these messages in my logs I suspect has to do with UDP buffer:

` AX3000T dnsproxy ERROR exchange failed prefix=dnsproxy upstream=https://**doh3**.dns.nextdns.io:443/351234/AX3000T question=";geomobileservices-pa.googleapis.com.\tIN\t A" duration=94.352593ms err="requesting https://doh3.dns.nextdns.io:443/351234/AX3000T: **Get_0rtt** \"https://doh3.dns.nextdns.io:443/351234/AX3000T?dns=AAABAAABAAAAAAABFGdlb21vYmlsZXNlcnZpY2VzLXBhCmdvb2dsZWFwaXMDY29tAAABAAEAACkE0AAAgAAAFv3pAAZEHH_GV5wACAAIAAEgAMCoAn4\": **use of closed network connection"**

AX3000T dnsproxy ERROR exchange failed prefix=dnsproxy upstream=**quic:**//AX3000T-351234.dns.nextdns.io:853 question=";static.whatsapp.net.\tIN\t A" duration=15.569509ms err="reading response from quic://AX3000T-351234.dns.nextdns.io:853: **EOF**"`

1715173329, 2025-05-16:
Try to apply this sysctl conf https://github.com/openwrt/packages/blob/master/net/adguardhome/files/adguardhome.sysctl

Mile-Lile, 2025-05-16:
ah now I get it. It was working until I flashed new owrt firmware....

to make it persistent changes, these directives below should be added to `/etc/sysctl.conf` not to `/etc/sysctl.d/*` as that directory is not backed-up by default and will not survive a reimage

`net.core.rmem_max=7500000`
`net.core.wmem_max=7500000`

and apply it with 
`sysctl -p`

and everything could be checked with:

`cat /proc/sys/net/core/rmem_default`
`cat /proc/sys/net/core/rmem_max`
`cat /proc/sys/net/core/wmem_default`
`cat /proc/sys/net/core/wmem_max`

1715173329, 2025-05-17:
bd4e40fb04751ac112a6d3ea59b1b4aab08e1ef8

closed 2025-05-17 by 1715173329

Mile-Lile, 2025-06-13:
@1715173329 
can we get 0.75.6 in openwrt updated
[dnsproxy](https://github.com/AdguardTeam/dnsproxy)