openwrt/packages #26525: strongswan mediatek/filogic (aarch64_cortex-a53)  build error

nikolainv225, 2025-05-17:
Hello.

I can't install strongswan (strongswan-related) packages from the apk repository (no packages).
Error log - https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/strongswan/

Thank you in advance.

OpenWrt version
r29487-044c7d55b3

OpenWrt release
SNAPSHOT

OpenWrt target/subtarget
mediatek/filogic

Device
Cudy WR3000H v1

Image kind
Official downloaded image

Steps to reproduce
apk add --simulate strongswan

brada4, 2025-05-17:
Maintainer @pprindeville  last commit by @lowjoel

lowjoel, 2025-05-18:
If we're talking about ```
uci_parser.c: In function 'section_enumerator_enumerate':
uci_parser.c:76:21: error: implicit declaration of function 'uci_lookup'; did you mean 'uci_lookup_ptr'? [-Wimplicit-function-declaration]
```

It does not seem like it has anything to do with my changes, however when I did a quick look at the uci source code it hasn't changed anything in the past few months (years even). It'll need someone with more information about the uci plugin (maybe @Thermi)

robimarko, 2025-05-19:
Fixed via: https://github.com/openwrt/packages/commit/a8c89a0c77b71c1699c57b7c6d70b034a271d678

closed 2025-05-19 by robimarko