openwrt/packages #26535: dockerd: adjust shutdown timeout to allow graceful shutdown of containers

vincejv, 2025-05-19:
Maintainer: @G-M0N3Y-2503 

By default `procd` has a shutdown timeout of 5 seconds, `procd` initially sends a `SIGTERM` kill signal to allow docker service to shutdown gracefully, after waiting 5 seconds, if the service is still running, it will send a `SIGKILL` signal forcing the service is shutdown forcefully and immediately without consideration even if the container is still writing a a file to the disk.

This behavior often causes **severe data corruption and partial writes**, especially for Database services like mysql, postgres, influxdb, prometheus (tsdb), SQLite-based services, etc. This becomes a major issue especially when you're frequently restarting your device/computer (x86).

I'd like to propose increasing the `term_timeout` in `procd` for docker service to at least 5 minutes AND allowing the configuration of `shutdown-timeout` in `daemon.json` through UCI.

Proposed changes in `/etc/init.d/dockerd`

in `start_service()`
```sh
procd_set_param term_timeout 300  # e.g. 5 minutes to wait for shutdown
```

in `process_config()`
```sh
...
config_get shutdown_timeout globals shutdown_timeout "300"
...
json_add_int "shutdown-timeout" "${shutdown_timeout}"
```

vincejv, 2025-05-19:
Also to add in openwrt wiki, this `procd` behavior is also documented

> The default termination timeout value is 5 seconds. Services with expensive shutdown operations, such as database systems, should set the term_timeout parameter to sufficiently large value.

[image]

G-M0N3Y-2503, 2025-05-19:
This sounds like a good idea, I'd be happy to review a PR but I don't have openwrt compiling ATM so it may take a while to get around to this myself.