openwrt/packages #26552: lvm2: build fails due to find yielding a long command line

liovbt, 2025-05-21:
In utils/lvm2/Makefile, the following install command creates very long command lines, which can overflow the default limit of 4096 characters.
```
$(FIND) $(PKG_INSTALL_DIR)/usr/sbin/ -type l -exec $(CP) -a {} $(1)/sbin/ \;
```
I experienced build failures in my environment because the PKG_INSTALL_DIR is quite long. It would be more robust to execute individual cp commands per file.

BKPepe, 2025-05-21:
Is prplOS supported by us, @ynezz?

brada4, 2025-05-21:
Thats pre-23.05 package, ask prplos to help with their custom package tree. Or use OpenWrt to complain about this package set

ynezz, 2025-05-21:
@liovbt thanks for the report, can you try proposed fix https://github.com/openwrt/packages/pull/26554 ?