openwrt/packages #26569: libpfring: error: 'static' is not at beginning of declaration

BKPepe, 2025-05-23:
Maintainer: N/A
Env: OpenWrt daily snapshot, aarch64_cortex-a53

Log:
```
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:284:1: error: 'static' is not at beginning of declaration [-Werror=old-style-declaration]
  284 | const static ip_addr ip_zero = { IN6ADDR_ANY_INIT };
      | ^~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:467:7: error: no previous prototype for 'get_num_rx_queues' [-Werror=missing-prototypes]
  467 | u_int get_num_rx_queues(struct net_device *dev)
      |       ^~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:485:7: error: no previous prototype for 'lock_rss_queues' [-Werror=missing-prototypes]
  485 | u_int lock_rss_queues(struct net_device *dev)
      |       ^~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:744:14: error: no previous prototype for 'netns_lookup' [-Werror=missing-prototypes]
  744 | pf_ring_net *netns_lookup(struct net *net) {
      |              ^~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:762:14: error: no previous prototype for 'netns_add' [-Werror=missing-prototypes]
  762 | pf_ring_net *netns_add(struct net *net) {
      |              ^~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:1017:17: error: no previous prototype for 'pf_ring_device_ifindex_lookup' [-Werror=missing-prototypes]
 1017 | pf_ring_device *pf_ring_device_ifindex_lookup(struct net *net, int ifindex) {
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:1031:17: error: no previous prototype for 'pf_ring_device_name_lookup' [-Werror=missing-prototypes]
 1031 | pf_ring_device *pf_ring_device_name_lookup(struct net *net /* namespace */, char *name) {
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:3467:5: error: no previous prototype for 'check_perfect_rules' [-Werror=missing-prototypes]
 3467 | int check_perfect_rules(struct sk_buff *skb,
      |     ^~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:3541:5: error: no previous prototype for 'check_wildcard_rules' [-Werror=missing-prototypes]
 3541 | int check_wildcard_rules(struct sk_buff *skb,
      |     ^~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:3657:5: error: no previous prototype for 'bpf_filter_skb' [-Werror=missing-prototypes]
 3657 | int bpf_filter_skb(struct sk_buff *skb,
      |     ^~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:3711:5: error: no previous prototype for 'sample_packet' [-Werror=missing-prototypes]
 3711 | int sample_packet(struct pf_ring_socket *pfr) {
      |     ^~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:3731:11: error: no previous prototype for 'default_rehash_rss_func' [-Werror=missing-prototypes]
 3731 | u_int32_t default_rehash_rss_func(struct sk_buff *skb, struct pfring_pkthdr *hdr)
      |           ^~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:4477:6: error: no previous prototype for 'register_device_handler' [-Werror=missing-prototypes]
 4477 | void register_device_handler(void)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:4486:6: error: no previous prototype for 'unregister_device_handler' [-Werror=missing-prototypes]
 4486 | void unregister_device_handler(void)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:4724:6: error: no previous prototype for 'reserve_memory' [-Werror=missing-prototypes]
 4724 | void reserve_memory(unsigned long base, unsigned long mem_len)
      |      ^~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:4733:6: error: no previous prototype for 'unreserve_memory' [-Werror=missing-prototypes]
 4733 | void unreserve_memory(unsigned long base, unsigned long mem_len)
      |      ^~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:6038:14: error: no previous prototype for 'ring_poll' [-Werror=missing-prototypes]
 6038 | unsigned int ring_poll(struct file *file,
      |              ^~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:6130:23: error: no previous prototype for 'cluster_lookup' [-Werror=missing-prototypes]
 6130 | ring_cluster_element *cluster_lookup(u_int32_t cluster_id) {
      |                       ^~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:6149:5: error: no previous prototype for 'get_first_available_cluster_queue' [-Werror=missing-prototypes]
 6149 | int get_first_available_cluster_queue(ring_cluster_element *el)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:6162:24: error: no previous prototype for 'get_first_cluster_consumer' [-Werror=missing-prototypes]
 6162 | struct pf_ring_socket *get_first_cluster_consumer(ring_cluster_element *el)
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:6179:5: error: no previous prototype for 'add_sock_to_cluster_list' [-Werror=missing-prototypes]
 6179 | int add_sock_to_cluster_list(ring_cluster_element *el, struct sock *sk, u_int16_t consumer_id)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:6231:5: error: no previous prototype for 'remove_from_cluster_list' [-Werror=missing-prototypes]
 6231 | int remove_from_cluster_list(struct ring_cluster *cluster_ptr, struct sock *sock)
[... 4 lines cut ...]
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:8266:6: error: no previous prototype for 'pf_ring_zc_dev_register' [-Werror=missing-prototypes]
 8266 | void pf_ring_zc_dev_register(zc_dev_callbacks *callbacks,
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:8351:6: error: no previous prototype for 'pf_ring_zc_dev_unregister' [-Werror=missing-prototypes]
 8351 | void pf_ring_zc_dev_unregister(struct net_device *dev, u_int channel_id)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:8505:6: error: no previous prototype for 'remove_device_from_proc' [-Werror=missing-prototypes]
 8505 | void remove_device_from_proc(pf_ring_net *netns, pf_ring_device *dev_ptr) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.c:8540:6: error: no previous prototype for 'remove_device_from_ring_list' [-Werror=missing-prototypes]
[... cut ...] ]
 8637 | int add_device_to_ring_list(struct net_device *dev, int32_t dev_index)
      |     ^~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[6]: *** [scripts/Makefile.build:244: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/libpfring-8.6.1/kernel/pf_ring.o] Error 1
```
Log: https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/libpfring/compile.txt

# Solution:
- Remove this unmaintained package here

BKPepe, 2025-06-04:
Yes, it is kmod, but that kmod is in depends for the package, which you can not install, because of missing dependency. In the stable releases, it is there, so the issue is still valid. 😇

brada4, 2025-08-09:
Upgrade packaage to v9.0.x
Hopefully it is only about bumping version,maybe adjusting/removing a patch or two.