openwrt/packages #26572: pythong-event:  error: passing argument 1 of '__pyx_vtabptr_6gevent_14_gevent_cqueue_UnboundQueue->__pyx_base.put'

BKPepe, 2025-05-23:
Maintainer: @predators46
OpenWrt snapshot, aarch64_cortex-a53

Output:
```
cc1: note: someone does not honour COPTS correctly, passed 3 times
src/gevent/queue.c: In function '__pyx_pf_6gevent_14_gevent_cqueue_12UnboundQueue_2put':
src/gevent/queue.c:11895:83: error: passing argument 1 of '__pyx_vtabptr_6gevent_14_gevent_cqueue_UnboundQueue->__pyx_base.put' from incompatible pointer type [-Wincompatible-pointer-types]
11895 |   __pyx_t_1 = __pyx_vtabptr_6gevent_14_gevent_cqueue_UnboundQueue->__pyx_base.put(__pyx_v_self, __pyx_v_item, 1, &__pyx_t_2); if (unlikely(!__pyx_t_1)) __PYX_ERR(0, 432, __pyx_L1_error)
      |                                                                                   ^~~~~~~~~~~~
      |                                                                                   |
      |                                                                                   struct __pyx_obj_6gevent_14_gevent_cqueue_UnboundQueue *
src/gevent/queue.c:11895:83: note: expected 'struct __pyx_obj_6gevent_14_gevent_cqueue_Queue *' but argument is of type 'struct __pyx_obj_6gevent_14_gevent_cqueue_UnboundQueue *'
src/gevent/queue.c:11895:114: error: passing argument 4 of '__pyx_vtabptr_6gevent_14_gevent_cqueue_UnboundQueue->__pyx_base.put' from incompatible pointer type [-Wincompatible-pointer-types]
11895 |   __pyx_t_1 = __pyx_vtabptr_6gevent_14_gevent_cqueue_UnboundQueue->__pyx_base.put(__pyx_v_self, __pyx_v_item, 1, &__pyx_t_2); if (unlikely(!__pyx_t_1)) __PYX_ERR(0, 432, __pyx_L1_error)
      |                                                                                                                  ^~~~~~~~~~
      |                                                                                                                  |
      |                                                                                                                  struct __pyx_opt_args_6gevent_14_gevent_cqueue_12UnboundQueue_put *
src/gevent/queue.c:11895:114: note: expected 'struct __pyx_opt_args_6gevent_14_gevent_cqueue_5Queue_put *' but argument is of type 'struct __pyx_opt_args_6gevent_14_gevent_cqueue_12UnboundQueue_put *'
src/gevent/queue.c: At top level:
src/gevent/queue.c:7636:12: warning: '__pyx_pw_6gevent_14_gevent_cqueue_5Queue_25__nonzero__' defined but not used [-Wunused-function]
 7636 | static int __pyx_pw_6gevent_14_gevent_cqueue_5Queue_25__nonzero__(PyObject *__pyx_v_self) {
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
error: command '/builder/shared-workdir/build/sdk/staging_dir/toolchain-aarch64_cortex-a53_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc' failed with exit code 1

ERROR Backend subprocess exited when trying to invoke build_wheel
make[3]: *** [Makefile:52: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/pypi/gevent-23.9.1/.built] Error 1
time: package/feeds/packages/python-gevent/compile#83.41#10.66#90.09
```

Comes from https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/python-gevent/compile.txt

predators46, 2025-05-23:
https://github.com/gevent/gevent/issues/2031#issuecomment-2108204975

BKPepe, 2025-05-23:
Hmm, while reading issue, which is one year old and there is said that newer version of gevent will fix it. Then why dont update it here? 🤔 🤔 Is there something, which prevents to update it?

predators46, 2025-05-24:
https://github.com/gevent/gevent/blob/cf4e8c522995ed8c09f2fa2a3595c4c299ba7d8c/pyproject.toml#L21

predators46, 2025-05-24:
https://github.com/gevent/gevent/blob/cf4e8c522995ed8c09f2fa2a3595c4c299ba7d8c/pyproject.toml#L6

predators46, 2025-05-24:
https://github.com/gevent/gevent/blob/cf4e8c522995ed8c09f2fa2a3595c4c299ba7d8c/pyproject.toml#L23

predators46, 2025-05-24:
https://github.com/gevent/gevent/blob/cf4e8c522995ed8c09f2fa2a3595c4c299ba7d8c/pyproject.toml#L35